ImpulseAdventure/GUIslice #155: Example ex03_ard_btn_img compiler error

Afantor, 2019-07-11:
I have set  #define GSLC_SD_EN 1 in esp-tftespi-default-xpt2046_int.h file.
But when compiler the code ex03_ard_btn_img.ino is error.

Hardware of device
ESP32 development board
2.4" TFT touch_ILI9341 display(XPT2046)
sdcard connect esp32 SPI (pin23、19、18、5)

error message is：
 `C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:726:27: error: 'File' was not declared in this scope

 uint16_t gslc_DrvRead16SD(File &f) {

                           ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:726:27: note: suggested alternative:

In file included from C:\Program Files (x86)\Arduino\libraries\ESP32-TFT_eSPI/TFT_eSPI.h:120:0,

                 from C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:47:

C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\FS\src/FS.h:47:7: note:   'fs::File'

 class File : public Stream

       ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:726:33: error: 'f' was not declared in this scope

 uint16_t gslc_DrvRead16SD(File &f) {

                                 ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:726:36: error: expected ',' or ';' before '{' token

 uint16_t gslc_DrvRead16SD(File &f) {

                                    ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:733:27: error: 'File' was not declared in this scope

 uint32_t gslc_DrvRead32SD(File &f) {

                           ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:733:27: note: suggested alternative:

In file included from C:\Program Files (x86)\Arduino\libraries\ESP32-TFT_eSPI/TFT_eSPI.h:120:0,

                 from C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:47:

C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\FS\src/FS.h:47:7: note:   'fs::File'

 class File : public Stream

       ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:733:33: error: 'f' was not declared in this scope

 uint32_t gslc_DrvRead32SD(File &f) {

                                 ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:733:36: error: expected ',' or ';' before '{' token

 uint32_t gslc_DrvRead32SD(File &f) {

                                    ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp: In function 'void gslc_DrvDrawBmp24FromSD(gslc_tsGui*, const char*, uint16_t, uint16_t)':

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:744:3: error: 'File' was not declared in this scope

   File     bmpFile;

   ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:744:3: note: suggested alternative:

In file included from C:\Program Files (x86)\Arduino\libraries\ESP32-TFT_eSPI/TFT_eSPI.h:120:0,

                 from C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:47:

C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\FS\src/FS.h:47:7: note:   'fs::File'

 class File : public Stream

       ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:766:8: error: 'bmpFile' was not declared in this scope

   if ((bmpFile = SD.open(filename)) == 0) {

        ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:766:18: error: 'SD' was not declared in this scope

   if ((bmpFile = SD.open(filename)) == 0) {

                  ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:771:23: error: 'bmpFile' was not declared in this scope

   if(gslc_DrvRead16SD(bmpFile) == 0x4D42) { // BMP signature

                       ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:771:30: error: 'gslc_DrvRead16SD' cannot be used as a function

   if(gslc_DrvRead16SD(bmpFile) == 0x4D42) { // BMP signature

                              ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:772:50: error: 'gslc_DrvRead32SD' cannot be used as a function

     uint32_t nFileSize = gslc_DrvRead32SD(bmpFile);

                                                  ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:775:35: error: 'gslc_DrvRead32SD' cannot be used as a function

     (void)gslc_DrvRead32SD(bmpFile); // Read & ignore creator bytes

                                   ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:776:46: error: 'gslc_DrvRead32SD' cannot be used as a function

     bmpImageoffset = gslc_DrvRead32SD(bmpFile); // Start of image data

                                              ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:779:49: error: 'gslc_DrvRead32SD' cannot be used as a function

     uint32_t nHdrSize = gslc_DrvRead32SD(bmpFile);

                                                 ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:782:41: error: 'gslc_DrvRead32SD' cannot be used as a function

     bmpWidth  = gslc_DrvRead32SD(bmpFile);

                                         ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:783:41: error: 'gslc_DrvRead32SD' cannot be used as a function

     bmpHeight = gslc_DrvRead32SD(bmpFile);

                                         ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:784:32: error: 'gslc_DrvRead16SD' cannot be used as a function

     if(gslc_DrvRead16SD(bmpFile) == 1) { // # planes -- must be '1'

                                ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:785:42: error: 'gslc_DrvRead16SD' cannot be used as a function

       bmpDepth = gslc_DrvRead16SD(bmpFile); // bits per pixel

                                          ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:787:55: error: 'gslc_DrvRead32SD' cannot be used as a function

       if((bmpDepth == 24) && (gslc_DrvRead32SD(bmpFile) == 0)) { // 0 = uncompressed

                                                       ^

C:\Program Files (x86)\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:862:3: error: 'bmpFile' was not declared in this scope

   bmpFile.close();

   ^
`

Afantor, 2019-07-11:
and I want to know sdcard how to conncet esp32 pin?

ImpulseAdventure, 2019-07-11:
Hi -- thank you for reporting this. It seems that the GSLC_SD_EN=1 mode was missing in my TFT_eSPI regression tests. The test suite will be enhanced to include this mode as well.

I have now uploaded a fix for the `GSLC_SD_EN=1` mode to the latest GUIslice. Please update the library and confirm if the compilation error has gone away.

As for connecting up the SD card, it really depends on what exact display module you are using. Could you send a link or photo of the back of the 2.4" ILI9341 board you have?

We should start by confirming that you can run Bodmer's TFT_eSPI SD/SPIFFS examples first (without GUIslice).
- Can you run TFT_eSPI examples/generic/ ESP32_SDcard_jpeg?
- Can you run TFT_eSPI examples/generic/ TFT_SPIFFS_BMP?

thanks

Afantor, 2019-07-12:
Thank you for fix TFT_eSPI GSLC_SD_EN=1 compile issue.
when I run TFT_eSPI examples/generic/ ESP32_SDcard_jpeg, It is working, picture show from sdcard.
and ESP32 connect to sdcard by SPI(GPIO23、19、18、5).
Other，when I run TFT_eSPI examples/generic/ TFT_SPIFFS_BMP,  it just print "SPIFFS initialised.", not show any picture。

Finally, I run ex03_ard_btn_img.ino, compile success. But it is not show any picture, I'm sure "/exit_n24.bmp" and "/exit_g24.bmp" in the sdcard.

ImpulseAdventure, 2019-07-12:
Great - good to hear the compilation is OK, and thanks for running the TFT_eSPI tests.

I have now pushed some changes to the repository to support SD card drawing on the TFT_eSPI mode. Please download the latest version of GUIslice to try out **ex03**.

You will need to add the following line to your config file:
```c++
#define ADAGFX_PIN_SDCS 14 // Change to your SD card chip select
```
(With your `SPI(GPIO23、19、18、5)`, perhaps 5 was your chip select?)

I will be updating all of the TFT_eSPI example configs to include this line as well.

Hope that helps!

ImpulseAdventure, 2019-07-18:
Hi @Afantor - if you have a chance to re-test, I would be interested in hearing whether the changes worked for you (I can reopen the issue if there are any troubles). In the meantime I will assume that the updates have addressed TFT_eSPI mode SD card operation.

Thanks!

closed 2019-07-18 by ImpulseAdventure