ImpulseAdventure/GUIslice #156: gslc_ElemSetVisible() does not work

LowPowerLab, 2019-07-19:
It appears that calling `gslc_ElemSetVisible(&m_gui, _A_BUTTON_, false);` on a button, only disables the button, but does not hide it when the display is refreshed.

I found this comment in `XKeyPad.c`  but I'm not sure if this is specific to that component or it really refers to `gslc_ElemSetVisible` not yet working at all:
`  // FIXME: Need to fix dynamic visibility change`
`  // gslc_ElemSetVisible(pGui, pElemRefTmp, bVisible);`

**Platform:**
[MoteinoM0](https://lowpowerlab.com/guide/moteino/moteinom0/) (CortexM0+ SAMD21G18A 48Mhz)
ILI9341 2.4" + Touchscreen
`#define GUISLICE_VER "0.12.1"`
`#include "../configs/ard-adagfx-ili9341-xpt2046.h"`
Arduino 1.8.7

ImpulseAdventure, 2019-07-20:
Hi @LowPowerLab -- you're right! Thanks for reporting this and providing the helpful supporting details.

I have now checked in a fix to support hiding of elements via `ElemSetVisible()`.
Please try the latest GitHub master and let me know if the functionality works correctly for you now.

LowPowerLab, 2019-07-21:
That works, thanks for the quick resolution!

ImpulseAdventure, 2019-07-21:
Great - thank you for the followup

closed 2019-07-21 by ImpulseAdventure