ImpulseAdventure/GUIslice #161: gslc_PopupHide leaves Popup visible

pdupuy, 2019-08-03:
OS: Windows 7; GUIslice: 0.11.4 WIP-XGlowball (34b0f7e); Target: arduino

Calling gslc_PopupHide in the Popup's callback function does not cause it to be "invisible":

`gslc_PopupHide(&m_gui);
gslc_Update(&m_gui);
if ( gslc_ElemGetVisible(&m_gui, m_pElemKeypad) == true ) {
	Serial.println("Popup is visible");
}`

ImpulseAdventure, 2019-08-03:
Thanks for the report! I assume that your observation is that the element's **visible** flag is still set for the popup (keypad) control, even though the popup itself is no longer shown. In other words, `PopupHide()` does indeed hide the popup as expected, but that querying `ElemGetVisible()` indicates that the element is true.

If so, this behavior is actually by design, though I can see the potential for confusion!

Assuming that you find value in being able to query whether or not an element is truly visible on the screen, perhaps we can provide an additional API that takes into account both the element's status and the page's status.

## Details

Each element has a "visible" flag, and `ElemGetVisible()` queries the element's flag state. GUIslice supports up to 3 layers or pages in a **page stack** with the bottom layer typically for elements that appear on all screens, the middle layer is the active "page" and the top layer in the stack could be used for popups.

The way popups work is that we create a **popup page** that contains one or more elements. `PopupShow()` places this overlay/page onto the top of the page stack, so that we see the popup page elements on top of the current page. `PopupHide()` removes this page from the top of the page stack.

An element's "visible" status is independent of whether its associated page currently appears in the page stack or not. If an element is associated with a page that is not currently in the page stack (ie. in a popup layer), we don't force the element's visibility status to "hidden".

That said, I haven't given much thought to how users might want to use the knowledge of an element's overall visibility on the page (ie. taking into account both the element's own visible state plus the parent page's presence in the current page stack).

I still like the idea of keeping the element's visibility state independent of the page presence in the active page stack. However, on the basis of your question, I'm thinking that perhaps there could be value in adding an API (perhaps E`lemGetScreenVisible()`?) that only returns true if both the element's visibility flag is set and its associated page is currently visible. Would that be useful?

thx

pdupuy, 2019-08-03:
Any way to programmatically determine whether a Popup is being shown would satisfy my need (using an MVCish architecture). ElemGetOnScreen() sounds like a useful API for controlling a loosely coupled view. For now, if you could provide a snippet of code that does that, it would be most appreciated. Thanks Cal!

ImpulseAdventure, 2019-08-04:
Sure, here is a code snippet that should provide this functionality:

```c++
bool gslc_ElemGetOnScreen(gslc_tsGui* pGui,gslc_tsElemRef* pElemRef)
{
  int16_t nElemId = gslc_ElemGetId(pGui,pElemRef);
  for (int8_t nStackPage=0;nStackPage<GSLC_STACK__MAX;nStackPage++) {
    gslc_tsPage* pStackPage = pGui->apPageStack[nStackPage];
    if (!pStackPage) {
      continue;
    }
    gslc_tsCollect* pCollect = &pStackPage->sCollect;
    pElemRef = gslc_CollectFindElemById(pGui,pCollect,nElemId);
    if (pElemRef) {
      return gslc_ElemGetVisible(pGui,pElemRef);
    }
  }
  return false;
}
```

ImpulseAdventure, 2019-08-07:
I have now integrated `ElemGetOnScreen()` into the repo. If you notice any problems with the implementation, let me know and I'll reopen the issue. Thanks!

closed 2019-08-07 by ImpulseAdventure