ImpulseAdventure/GUIslice #163: Request: Support for ST7789v Display Driver

Bwanna, 2019-08-09:
Requesting adding support for ST7789v Display chipset using 4-wire interface.
I can beta test code on a KD024QVRMA041 running on ESP32.

ImpulseAdventure, 2019-08-10:
Hi there -- yes, I should be able to add support for the ST7789 displays.

There appear to be a number of driver libraries available for this display; could you let me know which one you are currently using?

For example:
- https://github.com/adafruit/Adafruit-ST7735-Library (Both ST7735 & ST7789)
- https://github.com/prenticedavid/ST7789_kbv
- https://github.com/PaulStoffregen/ST7789_t3 (Teensy-optimized)
- https://github.com/ananevilya/Arduino-ST7789-Library (derived from Adafruit's)

Note that I don't currently have an ST7789v display on-hand so I may be dependent upon your testing until I have ordered one.

thanks

Bwanna, 2019-08-10:
Hi,
My current project uses the [https://github.com/loboris/ESP32_TFT_library](url) driver.  This would be great for the ESP8266 and ESP32 MCUs.

I have used both of these: (good choice for Atmega mcus)
-  [https://github.com/ananevilya/Arduino-ST7789-Library](url)
- [https://github.com/ananevilya/Arduino-ST7789-Library](url)

No worries on testing one for the ESP32.  I'm new to using GUIslice, but will do my best

Note: With my project I want to setup changing variable values using either a Slider or a #-Pad.  Currently this is being done via serial terminal input.

Thank you!

ImpulseAdventure, 2019-08-10:
Thanks @Bwanna ... I am assuming that your project is currently using the **ESP-IDF** toolchain, is that correct?

## ESP-IDF
If so, please note that I have not yet tried compiling/porting GUIslice within ESP-IDF (I have mostly focused on addressing **Arduino IDE** and to some extent **PlatformIO**, plus make for **LINUX / RPi** targets).

There has been interest in the past in running GUIslice within ESP-IDF. I understand that some users have done this by integrating the [arduino_esp32](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) component and then adding GUIslice to the Arduino library folder, but I have not tested it myself yet.

## Arduino IDE
If your project was also able to target the Arduino IDE, then it should be a relatively simple matter to get support added for the display, given a suitable display library. I believe the ananevilya library has provided compatibility for ESP devices.

## External Controls
As for external controls, GUIslice has some basic support for external input control over the GUI (for non-touch displays), so I expect that it would be possible to adapt for an external numeric pad, for example.

Bwanna, 2019-08-10:
I'm still Arduino IDE for this project since it was originally built for Atmega MCUs and much easier to migrate over to ESP32.

For the controls. I'm looking to have TS controls to change settings in the programming.
The project has a few temp sensors and User changeable Alerts (Low, High).  The current project uses two physical buttons to step-through each sensor's Low and High Settings, and then cycle through the preferred number. (Think of basic digital clock setting functions).  I'd like to change this to a TS slider or TS numberpad, where I can direct input the new values.  At least those are my first ideas.

ImpulseAdventure, 2019-08-10:
Ok, that should make things easier! As you were using https://github.com/loboris/ESP32_TFT_library, I had (perhaps incorrectly) assumed that it only supported ESP-IDF and not the Arduino IDE. So, to confirm, I understand you are currently driving your display using the Arduino IDE with the ESP32_TFT library, and the same display library is also handling the touch input from the KD024QVRMA041?

Assuming the above, it should be straightforward to provide on-screen touch sliders for your controls to replace your physical buttons if we integrate support for ESP32_TFT.

thx

Bwanna, 2019-08-10:
CORRECTION: I made a mistake.  I'm using [https://github.com/Bodmer/TFT_eSPI](url) to driver the display.  (Very sorry about that mix up on my part!)  Your msg caused me to recheck my source!

For the TouchScreen part, I'm currently experimenting with the Adafruit_STMPE610 unit. 
[https://github.com/adafruit/Adafruit_STMPE610](url)  It's very easy to setup and provide the coordinates.

ImpulseAdventure, 2019-08-10:
Thanks for the update! In that case GUIslice may already support your display and driver setup, though some adjustments will be needed.

Please uncomment the following line in your GUislice_config.h:
- esp-tftespi-default-stmpe610.h

Instructions on configuring and testing are here:
- [Configure GUIslice for ESP8266 & ESP32](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-ESP8266-&-ESP32)
- Assuming Hardware SPI for touch, ADATOUCH_PIN_CS will need to match your Adafruit_STMPE610 config
- This process will also include the touch calibration for the STMPE610.

Bwanna, 2019-08-10:
Where would I check for the ADATOUCH_PIN_CS setting?  I've done everything and an error message : "- Init touch handler [STMPE610(SPI-HW)] FAIL"... I'm assuming it's because I haven't check the CS assignment.

Yes, I am using SPI HW.  If it helps, touchscreen is successful when running the test file from Adafruit_STMPE610 out of the box.

ImpulseAdventure, 2019-08-10:
We would set the ADATOUCH_PIN_CS to the same value you are currently using in the STMPE610 TouchTest.ino. In the Adafruit TouchTest example, you would find your pin setting (STMPE_CS) in the example’s constructor, eg:

Adafruit_STMPE610 touch = Adafruit_STMPE610(STMPE_CS);

Bwanna, 2019-08-11:
Still not sure what to do... and to which file to do it.  

I'm using I2C for the Adafruit_STMPE610.  Specifically, Option #1 in the TouchTest.ino which uses "Adafruit_STMPE610 touch = Adafruit_STMPE610();"

Bwanna, 2019-08-11:
OK. Got both the Display and Touch to initialize.  Recheck the esp-tftespi-default-stmpe610.h file and found the I2C switch that needed to be set.

ImpulseAdventure, 2019-08-11:
Glad to hear the **ST7789v** is already supported by the existing GUIslice drivers. I'll close this but please feel free to create a new issue if you ever have additional feature requests or encounter issues in developing your project. I will revisit the wiki documentation to ensure the above config details are covered properly too.

Thanks!

closed 2019-08-11 by ImpulseAdventure

Bwanna, 2019-08-11:
Thank you for walking me through this!  If this helps, what I realized I did not correctly configure was the "TFT_eSPI: Chip Select for Touch Device" setting in the "esp-tftespi-default-stmpe610.h" file.  Once I correctly set the I2C setting to my HW approach everything sang.

I'm looking forward to learning how your GUISlice works now  that all is connected!