ImpulseAdventure/GUIslice #164: Missing template: <FONT_INCLUDE>

mrbloozman, 2019-08-19:
I just tried builder 0.13 (linux version), upon code generation I receive an error message: builder.codegen.CodeGenException: Missing template: <FONT_INCLUDE>

GUIsliceBuilder/templates only contains c.t and ino.t along with the font spreadsheets, is there supposed to be more?

ImpulseAdventure, 2019-08-19:
Thank you @mrbloozman for noting this! I have recreated the same issue in my LINUX VM. I believe the issue is just a missing section in an internal LINUX template resource (`linux.t`); I'll let @Pconti31 take a look at it in more detail. thx

Pconti31, 2019-08-19:
The FONT_INCLUDE template isn't used when you target linux platform.
This template is only used by arduino.  The code was incorrectly trying to open
FONT_INCLUDE before testing for linux.
Fixed in 0.13.0.1 which is  available in the GUIslice-Builder-source repository releases.
Calvin will update the GUIslice API releases as soon as possible.

Paul--

mrbloozman, 2019-08-20:
Thanks @ImpulseAdventure and @Pconti31 for looking into this.  I've installed 0.13.0.1 and tried generating code again.  Now I'm getting a fatal crash (see attached log).

[20190819_215256.txt](https://github.com/ImpulseAdventure/GUIslice/files/3518769/20190819_215256.txt)

Let me know if I can provide more information.

Pconti31, 2019-08-20:
Thanks for your report.  However, to hunt down the problem I need a zip file of your project folder.
It contains the *.prj file we are trying to generate code for.  Without it I have no idea what is doing on.
Thanks.

Paul--

mrbloozman, 2019-08-21:
Ah, that's a good hint.  I found an older c file that was generated by a prior version of the builder, which I removed, then regenerated a fresh c file with no issues.  Thanks for these very useful tools.

closed 2019-08-21 by mrbloozman

ImpulseAdventure, 2019-08-22:
Thank you @mrbloozman for the followup. For future reference, Paul had identified the issue with the Builder's import of older Builder-generated projects; this should now be fixed in the Builder hotfix release `0.13.0.2`.