ImpulseAdventure/GUIslice #165: builder not working on Mac

kriodoxis, 2019-08-19:
main jar is not working and shows message:

Error: Could not find or load main class com.baeldung.fatjar.Application
Caused by: java.lang.ClassNotFoundException: com.baeldung.fatjar.Application
[image]

ImpulseAdventure, 2019-08-19:
Thanks Mario for the report. It is quite possible that we need to adjust a setting when creating the executable JAR for the Mac. @Pconti31 (the Builder author) may be able to offer some insight and/or identify a fix.

Pconti31, 2019-08-19:
Thanks for the info since neither Calvin or I have access to macs we rely on users to help us out.
The issue appears to be the jar's manifest.mf is incorrectly setting the startup class.
A quick work-around is to name the startup class in your command line as so:
cd to your GUIsliceBuilder directory
java -classpath builder-0.13.0.jar builder.Builder
should work for you until I check in a fix. You may need to use ./builder-0.13.0.jar
Paul--

Pconti31, 2019-08-19:
I created 0.13.0.1 which is  available in the GUIslice-Builder-source repository releases.
Calvin will update the GUIslice API releases as soon as possible.

Please let us know if this solves your problem.
Again without a Mac I have no way to test.

Paul--

ImpulseAdventure, 2019-08-24:
Hi @kriodoxis -- note that the updated Builder is available for download from the GUIslice [release notes](https://github.com/ImpulseAdventure/GUIslice/releases) page (v0.13.0.1 or later). If you can confirm whether it now loads correctly, that would be great. thanks!