ImpulseAdventure/GUIslice #169: build the example  "arduino/ex01_ard_basic" is error.

yuchao86, 2019-09-28:
lto1.exe: internal compiler error: in splice_child_die, at dwarf2out.c:4905

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.21.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
为开发板 Arduino/Genuino Uno 编译时出错。

ImpulseAdventure, 2019-09-28:
Hi -- I assume you are using an older version of the Arduino IDE version (1.8.6 - 1.8.9)? The LTO error you are seeing was a bug in the gcc compiler that appeared in older versions of the Arduino IDE.

Thankfully, Arduino fixed this problem in the latest version of the IDE. Please install **Arduino IDE 1.8.10** and update your Arduino AVR Boards in the Boards Manager.

For more details, please see the wiki page: [Note about Arduino IDE Versions - LTO issue](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#note-about-arduino-ide-versions---lto-issue)

Hope that helps!

ImpulseAdventure, 2019-10-02:
Closing as the arduino/gcc compiler was fixed in Arduino IDE 1.8.10.

@yuchao86 -- if you still see any issues after running the latest Arduino IDE, please let me know. Thanks!

closed 2019-10-02 by ImpulseAdventure