ImpulseAdventure/GUIslice #170: Text-Element Update Problem

feendrache, 2019-10-28:
### Describe the bug
I'm using the Text Element to display GPS-Koords and Date/Time.. Using this code:
` String gps_time = "";
    gps_time = String(gps.date.day())+("/")+String(gps.date.month())+"/"+String(gps.date.year())+" "+String(gps.time.hour())+":"+String(gps.time.minute())+":"+String(gps.time.second());
    Serial.println(gps_time);
    char carray[26] = "                        ";
    gps_time.toCharArray(carray, sizeof(carray));
    gslc_ElemSetTxtStr(&m_gui, m_pElemDateInfo, carray); `

it's working the text is updated, but i'm not able to clear the displaying text in the element.
So the new text (seconds as example) appear on top of the old text
The page is not updated properly even when i try to call it or use gslc_Update(&m_gui);  

Maybe i just need to get the trigger to force the update into my code but i wasn't able to determine what this would be.
Changing page and returning to main page does the charm but only shows me the problem not the solution, can you give me a quick tip?

### Device hardware
- *MCU*: ATmega2560
- *Display*: Ellegoo ILI9341 2.8" resistive
- *GUIslice config selected*:  `ard-shld-elegoo_28_res`

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* all works fine
- *Touch works correctly with standalone touch library examples without GUIslice:* yep

ImpulseAdventure, 2019-10-29:
Hi @feendrache -- thanks for reporting this.

There are a few different ways in which fonts can be rendered, depending on the display mode and color options. In your case, it sounds like the text is being updated without a background redraw.

Can you confirm a couple additional details for me:
- What library version are you using?
- What font have you selected?
- Please indicate any calls you have made to create & modify the appearance of the text element (`m_pElemDateInfo`). In particular, I'm interested in the foreground color, background color and and fill attribute, ie. `ElemCreateTxt()`, `ElemSetFillEn()`, `ElemSetTxtCol()`, etc.
- Alternately, please attach the sketch you have created.

thanks!

ImpulseAdventure, 2019-10-30:
Thanks for attaching the sketch!

The problem you are seeing is that the text output is wider than the created text element dimensions. GUIslice performs rendering updates within the boundaries of each element. In both the `KOORD` and `T_TIME` elements, the width was specified as 18 pixels.

Please try modifying your sketch (or Builder project) to use a field that is large enough to encompass your output. For example:
```c++
// E_ELEM_KOORD width changed from 18 to 150
pElemRef = gslc_ElemCreateTxt(&m_gui,E_ELEM_T_KOORD,E_PG_MAIN,(gslc_tsRect){90,40,150,10},
    (char*)m_sDisplayTextKoord,26,E_FONT_TXT5);

// E_ELEM_T_TIME width changed from 18 to 100
pElemRef = gslc_ElemCreateTxt(&m_gui,E_ELEM_T_TIME,E_PG_MAIN,(gslc_tsRect){90,60,100,10},
    (char*)m_sDisplayTextTime,26,E_FONT_TXT5);
```

To visualize the update region differences, you can enable a frame on the text field with:
`gslc_ElemSetFrameEn(&m_gui,pElemRef,true);`

Note that I would have expected that the Builder code generation would have defaulted to a wider dimension after you had entered an External Storage Size of 26, but if not, that could explain the config difference.

feendrache, 2019-10-31:
Ah thank you,
yeah that explains a lot and the enabling of the frame is magic for debugging.
You were right that this was the problem with the updating. 

I'm not quite sure if it was a fault of the builder code generation or if i changed the storage size afterward... 

THANK YOU for your help :)

closed 2019-10-31 by feendrache