ImpulseAdventure/GUIslice #171: Full redraw of XRingGauge hides active sectors

dbonnell, 2019-11-17:
### Describe the bug
The initial/full (non-incremental) draw, XRingGauge draws 360 degrees of inactive sectors, hiding the active sectors drawn first.

### Device hardware
- *MCU*: ATmega2560
- *Display with link*: mcufriend 2.4" shield (http://mcufriend.com/index.php?main_page=product_info&cPath=4&products_id=13)
- *GUIslice config selected*:  `ard-shld-mcufriend`

### Expected behavior
Active sectors are visible.

### Initialization messages
```
GUIslice version [0.13.0]:
- Init display handler [ADA_MCUFRIEND] OK
- No touch handler enabled
DBG: InvRgnScreen
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: PageRedrawGo() InvRgn: En=1 (0,0)-(319,239) PageRedraw=1
DBG: redraw inc=0 last=0 cur=93 segs 0..360

RingDraw: Val=93 ValLast=0 PosMin=0 PosMax=100 Q=72
RingDraw: ActiveG   start=0 end=-320 astart=0 arange=360
RingDraw:  Inactive start=360 end=-320
```

### Additional info
Issue resolved by reversing the drawing order in `gslc_ElemXRingGaugeDraw`, drawing inactive then active.

Also notice the "ActiveG" debug statement in the output above.  The XRingGauge was configured for flat active colour, so it shouldn't be using gradient drawing.  This happens because `gslc_ElemXRingGaugeSetColorInactive` is setting `bGradient=true` for some reason.  I've removed that and there doesn't appear to be any adverse effect.  The debug output is ow:
```RingDraw: Val=93 ValLast=0 PosMin=0 PosMax=100 Q=72
RingDraw:  Inactive start=360 end=-320
RingDraw: Active   start=0 end=-320 astart=0 arange=360
```

ImpulseAdventure, 2019-11-18:
Thank you for reporting this and the detailed debug info! I plan to test it this evening.

To ensure I have your case matched correctly, can you confirm your parameters for the ElemXRingGaugeSetAngleRange() call if present?

Thanks

dbonnell, 2019-11-18:
Sure, it is:
```gslc_ElemXRingGaugeSetAngleRange(&m_gui,pElemRef, 0, 360, false);```

That's the values shown in the `DBG: redraw inc=0 last=0 cur=93 segs 0..360` log entry.  (I un-commented that DBG statement and replaced the nAngEnd with (nAngStart+nAngRange), as the former variable was no longer defined.

ImpulseAdventure, 2019-11-19:
Yes, the `ElemXRingGaugeSetColorInactive()` was indeed incorrectly setting the gradient flag. This has been fixed in the latest release (removed the line as you identified).

As for the full redraw, you're right! There was a bug in the way the full redraw mode was calculated. The incremental redraw correctly avoided integer overflow, but the full redraw was missing a suitable integer typecast. This resulted in an invalid angular `end=-320` value, which caused the drawing to be done in the opposite direction and other side-effects. I have now fixed this issue and also revised the debug messaging accordingly.

Note that the `bClockwise` parameter in `XRingGaugeSetAngleRange()` is not currently supported (it looks like you attempted to set counter-clockwise mode in your config).

Please let me know if the latest version fixes some of these issues for you.
thanks!

dbonnell, 2019-11-22:
Works great, thanks for the quick turnaround!

P.S. The counter-clockwise direction setting was unintentional.  I must have accidentally changed that in GUIslice Builder

closed 2019-11-22 by dbonnell