ImpulseAdventure/GUIslice #172: XRingGauge does not draw correctly is quality is increased

dbonnell, 2019-11-17:
### Describe the bug
If the quality is increased using `gslc_ElemXRingGaugeSetQuality`, the gauge only partially renders.  For example, if you increase the quality from the default of 72 to 144, it only renders half a circle.  With quality of 180, only around 1/6th of a circle is used.

### Device hardware
- *MCU*: ATmega2560
- *Display with link*: mcufriend touch shield TFT 2.4"
- *GUIslice config selected*: `ard-shld-mucfriend`

### Expected behavior
Gauge renders full circle, at higher quality (e.g. quality of 180 => 180 sectors of 2 degrees each)

### Initialization messages
```
GUIslice version [0.13.0]:
- Init display handler [ADA_MCUFRIEND] OK
- No touch handler enabled
DBG: InvRgnScreen
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: PageRedrawGo() InvRgn: En=1 (0,0)-(319,239) PageRedraw=1
DBG: redraw inc=0 last=0 cur=93 segs 0..360

RingDraw: Val=93 ValLast=0 PosMin=0 PosMax=100 Q=180
RingDraw:  Inactive start=360 end=-320
RingDraw: Active   start=0 end=-320 astart=0 arange=360
```

### Additional info

ImpulseAdventure, 2019-11-19:
Hi David --

The side-effects of #171 may have caused additional issues, but as you have seen, the higher quality settings (ie. 120, 180, 360) currently produce incorrect rendering. Similar to #171, this was due to an integer range issue that overflows in the higher precision modes. I have now checked in a fix that should allow the higher quality modes to render correctly.

As you have noted, the quality setting supports a few discrete values (more specifically, the number of degrees that make up each sector of a gauge rendering). In order to ensure we don't consume precious FLASH memory pulling in the floating point libraries, the quality setting utilizes integer math, hence the discrete values. I have updated the documentation to list out the available setting values, ie: 360, 180, 120, 90, 72, 60, 45, 40, 36, etc.

Please give the latest version a try and let me know if it works better.

dbonnell, 2019-11-22:
Tested at Q=180, worked fine, looks much smoother.  Thank you for the quick turnaround.

closed 2019-11-22 by dbonnell