ImpulseAdventure/GUIslice #173: Waveshare_ILI9486 alignment with Flash strings

ImpulseAdventure, 2019-11-19:
Bug report filed by @pitrz migrated from https://github.com/ImpulseAdventure/Waveshare_ILI9486/issues/1#issuecomment-555199974:

Hi,

i've been playing around with the new version and have noticed a strange behavior with text position / alignment. Basically what is happening if i use the waveshare font as per instructions is that alignment works as it supposed to when the element is stored in RAM. If use the PROGMEM / flash api / _P versions of the element (Button) labels are not aligned correctly. Can you please give me a hint what could be off?

Also - the code is generated using the builder.

Thanks,
Petr

ImpulseAdventure, 2019-11-19:
Thanks @pitrz for the report. I have now added support in Waveshare_ILI9486 mode for FLASH-based strings. Please try the latest branch of GUIslice here: https://github.com/ImpulseAdventure/GUIslice/tree/WIP173-WaveshareFlash and let me know if that works.

pitrz, 2019-11-21:
Hi. Thanks. Seems to work great. The only thing that is a bit confusing now is that the larger waveshare fonts don't really come up in the sizing as previewed in the Builder. They are usually half the size than what's being shown in the Builder. But i don't use them anyways for now. :) Thanks! Awesome support!

ImpulseAdventure, 2019-11-23:
Glad to hear it, @pitrz !

As for the font size preview in the Builder, you should be able to add a more accurate preview of the Waveshare fonts to the Builder with the steps listed in https://github.com/ImpulseAdventure/Waveshare_ILI9486/issues/1#issuecomment-548242430

I'll close the ticket now as the font alignment sounds like it is working. If there are any other problems, please feel free to open a new issue ticket. thanks!

closed 2019-11-23 by ImpulseAdventure