ImpulseAdventure/GUIslice #174: ESP 32 Font Problem

feendrache, 2019-11-22:
### Describe the bug
- Okey, i changed my display from an elegoo to an spi one and need to rebuild my GUI since the old one won't work in the esp32 with the spi display
- My Problem now: only the BuiltinFonts are working, no displaying of text on the display when i select one of the other fonts

### Device hardware
- *MCU*: ESP32
- *Display with link*: 2.8'' SPi Display with touch, red one generic
- *GUIslice config selected*: esp-tftespi-default-xpt2046_int

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* yes
- *Touch works correctly with standalone touch library examples without GUIslice:* yes

### Expected behavior
Well i wanted to use other fonts.. 
took me some time to realise that this was the problem

### Initialization messages
no errors

ImpulseAdventure, 2019-11-23:
Hi @feendrache --

Can you confirm an example of a font that you are trying to use? (ie. is it included within the [TFT_eSPI/Fonts/GFXFF](https://github.com/Bodmer/TFT_eSPI/tree/master/Fonts/GFXFF) directory?)

If so, can you confirm that the GUIslice example **ex04** works using the TFT_eSPI freefonts? (we would like to test it first without the Builder)
- **`/examples/arduino/ex04_ard_ctrls.ino`**
- Uncomment the `#define USE_EXTRA_FONTS` line
- Does the "Quit" button now use the larger font? (it should use `FreeSansBold12pt7b`)

thanks

feendrache, 2019-11-23:
Yeah that'S working just fine

ImpulseAdventure, 2019-11-25:
If you are seeing `ex04` work fine with TFT_eSPI custom fonts then the problem is likely that the Builder is not set to TFT_eSPI output mode. The TFT_eSPI display library handles fonts differently from other libraries, so the Builder needs to be in a special mode to generate the correct font initialization.

Please check the setting in the Builder's **Edit -> Options -> Target Platform**; it should be set to `arduino TFT_eSPI`. Does this help?

[image]

I'm not sure if we documented this visibly so perhaps a wiki documentation update might be necessary.

ImpulseAdventure, 2019-11-30:
Closing as I expect that the above Builder option will address the original issue.

closed 2019-11-30 by ImpulseAdventure