ImpulseAdventure/GUIslice #175: React to any touch on the display

feendrache, 2019-11-23:
### Feature Request or Enhancement
*React to any touch

### Additional context
Maybe this is already possible but i wasn' able to find it:
I have the LED power on a separate pin so i can switch the display off after an amount of time and i want the display to get back online when there is a touch.
My idea was to have a timer, when there is no touch for this amount of time i want to deactivate the backlighting.
And turn it back on when there is touch and reset the timer till for the display off with every touch.

First i thought i could create a workaround with creating a second page with a bug button. 
Switching to this page after the amount if time and returning when this button is pressed... this is working, but the timer is only resetted when the second page is reached but my idea was that ANY touch would reset the timer... sooo... i still need the general touch reaction... 

Can you help me with this?

Pconti31, 2019-11-24:
I'm sure Calvin will help you out when he gets some time but meanwhile take a look at
closed issues:
64 adding touch hander class as new DRV_TOUCH to enable user specific touch
and 
59 Need for a touch handler driver that can be configured by the application?
Never saw ex11 so I can't help you there.
Paul--

ImpulseAdventure, 2019-11-24:
Thanks Paul -- you're right: a custom touch handler in #59 could potentially be used to set some flags that the user can query to see if a touch event occurred. Perhaps they could take an existing touch handler (maybe XPT2046) and add in some additional hooks (public APIs) to detect & notify the events, though other approaches could probably be implemented with this custom handler too.

An alternate solution could be done with a small enhancement to the core library. For example, we could add a `gslc_SetTouchTriggerCb()` that accepts a user function pointer. Internally it would set a flag in the GUI that gets cleared any time a touch event occurs. If the flag was cleared and a callback function was registered, then the user function would be triggered (a single time).

So, if the user noted the inactivity timer was exceeded, they would drive the GPIO to disable the display and call `SetTouchTriggerCb(&MyWakeFunction)`. In `MyWakeFunction()`, the user would drive the GPIO to enable the display and reset the timer.

Haven't done much thinking on it, but it could be a relatively low-cost enhancement that might potentially be useful in other user operations as well. Thoughts?

@feendrache -- your idea of swapping to a new page with a big, screen-sized button is an interesting one that I would not have thought of :) though as you point out, you really need any touch event to trigger your timer reset. Perhaps one of the two ideas that Paul and I mentioned above could provide you with a more seamless solution.

feendrache, 2019-11-24:
Yeah Idea with the `SetTouchTriggerCb` sounds great.
So far my version with the big-Button-Page is working and since i don't have much time for implementing something other (prototype testing is starting soon) my current solution is:
- There is a checkbox... once checked the timer is not used and the screen will stay on
- if not checked and the timer is up the page will swap and the display turns black
- if the big button is pressed, the page will swap and the display is lighted ... works as wanted :-)

ImpulseAdventure, 2019-12-03:
@feendrache -- FYI, I had added the `SetTouchTrigFunc()` capability in the branch [WIP175-TouchTrig](https://github.com/ImpulseAdventure/GUIslice/compare/WIP175-TouchTrig) (you can download the release from [here](https://github.com/ImpulseAdventure/GUIslice/tree/43145c22e5db5eb7f1138e4185a10a43bdb625b0) ). In it, you will find example `ex45_ard_touch_trig` that demonstrates a touch activity timeout.

closed 2019-12-24 by ImpulseAdventure