ImpulseAdventure/GUIslice #177: ER-TFTM070-5  eastring display with RA8875 controller and FT5206

gioreva, 2019-11-25:
I need to use
ER-TFTM070-5  eastring display with RA8875 controller and FT5206 capacitive touch 
buydisplay.com

I have working example for it, my micro is SAMD, arduino zero.
Can you help me to adjust example for use GUIslice ?

ImpulseAdventure, 2019-11-26:
Hi --

First, let's confirm that we can drive your display correctly with the non-touch config: 
- Uncomment `/configs/ard-adagfx-ra8875-notouch.h` in `/src/GUIslice_config.h`
- Please report if example **ex04** appears to render to the display properly

The touch controller on the EastRising display (FT5206) is not very common. I don't think any of the existing drivers in GUIslice support this touch controller module.

However, I believe the following touch library might work:
- [GitHub: sumotoy/FT5206](https://github.com/sumotoy/FT5206)
- If you can confirm that the "test_with_RA8875.ino" example works with your display, I can look into adding support for it into GUIslice.

thx

gioreva, 2019-11-26:
Nothing, after this change, old image remain on the TFT
TFT is 1024x600 and not #define DRV_DISP_ADAGFX_RA8875_INIT RA8875_800x480
But i think difference are more.
Can I send you the working examples you look at them?

ada-adagfx-ra8875-notouch.h
#define ADAGFX_PIN_CS       11    // Display chip select
#define ADAGFX_PIN_RST      12    // Display Reset Not connected
#define ADAGFX_PIN_SDCS     12    // SD card chip select (if GSLC_SD_EN=1)
#define GSLC_FEATURE_COMPOUND       1   // Compound elements (eg. XSelNum)
GUIslice_config.h
#include "../configs/ard-adagfx-ra8875-notouch.h"

ImpulseAdventure, 2019-11-26:
Ok thanks for trying out the default config.  As I don't have an **Arduino Zero** on hand to test with my **5" EastRising RA8875** display, we will need to see if you are able to run your hardware with the existing GUIslice-supported libraries. If they aren't compatible, then we can explore adding support for a new library.

In particular:
- Step 1) When you tried GUIslice with config `ada-adagfx-ra8875-notouch`, what was reported on the serial debug monitor? (you should have seen an initialization message -- did it report any errors?)
- Step 2) Please attach or link to the working example you are currently running with the display (I assume it was from EastRising directly?)
- Step 3) Try [Adafruit_RA8875](https://github.com/adafruit/Adafruit_RA8875): please see if the `buildtest.ino` example works. Note that you will need to change their `tft.begin(RA8875_480x272)` initialization to `tft.begin(RA8875_800x480)`. Does the graphics example work?
- Step 4) Try [sumotoy/RA8875](https://github.com/sumotoy/RA8875): please see if the `graphicAdaTest.ino` example works.

thanks!

gioreva, 2019-11-26:
I apologize, the controller RA8876 / 77
This working examples.
https://www.dropbox.com/s/sgcuqq2hn75apmc/TFT_7inch_adapter.zip?dl=0
EastRing Sources only for Arduno Due, Do not build for Zero
I have received it directly from RAIO, and modified according EastRing example.
There are two function, 
Slow:
lcdPutChar32x48
lcdPutString32x48
And Fast:
lcdPutChar32x48_block
lcdPutString32x48_block

I can try monitor debug and adafruit library, if you want, but I do not think you need, whereas the controller is different.

ImpulseAdventure, 2019-11-27:
Your attached examples were helpful in enabling me to understand the library/controller differences.

I presume your display might be the **EastRising 7" 1024x600 [ER-TFTM070-6](https://www.buydisplay.com/default/spi-7-inch-tft-lcd-dislay-module-1024x600-ra8876-optl-touch-screen-panel)**?

Unfortunately, it looks like it would be a fair amount of work for me to add support for the **RAIO RA8876** library (as attached) since it doesn't align with the common display APIs of other libraries. If it turned out that the **sumotoy/RA8875** library worked your display, I could consider adding support (as it is also more broadly applicable for users). Given that there are a number of [differences between RA8875 and RA8876](https://github.com/xlatb/ra8876/wiki/RA8875-vs-RA8876), I'm less optimistic that the RA8875 lib would work.

gioreva, 2019-11-27:
Using Adafruit RA8875, old image do not disapper from TFT, same your.
sumotoy not compile for SAMD
Comparing the code, all the registers are different.
I can try to do the mix of registers from my code to adafruit.
if xlatb/ra8876 can be helpfull for you ?

ImpulseAdventure, 2019-11-28:
Hi @gioreva -- I think you will probably find that [xlatb/ra8876](https://github.com/xlatb/ra8876) is the most promising library option. Thankfully, it does appear to implement a more conventional display library API. The first step would be for you to confirm that the library works on your device -- give the example `ra8876-demo` sketch a try and let me know if you have any success.

gioreva, 2019-11-28:
Waoooo, impressive speed xlatb / ra8876
Works well.
Except PWM light on that there is not.

I'm trying with [GEVINO Opto](https://shop.gevaelettronica.it/en/arduino-compatible/58-tft-7-1024x600-with-capacitive-touch.html) associated with ER-TFTM070-6
But I'm working on  [GEVINO TFT](https://shop.gevaelettronica.it/en/arduino-compatible/63-gevino-tft-7.html).

ImpulseAdventure, 2019-11-30:
@gioreva - glad to hear that the RA8876 library works on your display!

Assuming that you have an **FT5206** touch controller, does the [sumotoy/FT5206](https://github.com/sumotoy/FT5206) touch example (`test_with_RA8875.ino`) operate correctly on your capacitive overlay?

I have been working on a modification to GUIslice that adds support for the **RA8876** but I am not able to test it since I don't own one of these displays. As fonts are handled a bit differently and I can't check locally, I may need to exchange some test builds with you to try out for me.

It may be more convenient if you could email me at `guislice` (gmail.com) so that I can send you something to try. thanks!

ImpulseAdventure, 2019-12-09:
Just an update for the benefit of others: thanks to gioreva's remote test setup, the [WIP177-RA8876](https://github.com/ImpulseAdventure/GUIslice/tree/WIP177-RA8876) branch now supports the **RA8876 display driver** with the internal ROM font. The next step would be to add support for the external ROM font modules.

ImpulseAdventure, 2019-12-16:
Support for RA8876 has now been integrated into GUIslice, selected by the `DRV_DISP_ADAGFX_RA8876` mode. This driver mode utilizes the [xlatb/RA8876](https://github.com/xlatb/ra8876) library.

ImpulseAdventure, 2020-01-08:
For reference, touch support for the FT5206 touch controller has been posted in the [WIP177-FT5206](https://github.com/ImpulseAdventure/GUIslice/tree/WIP177-FT5206) branch.

ImpulseAdventure, 2020-02-09:
Now that the **EastRising 7" display** (ER-TFTM070-6) has finally arrived, I have been able to launch proper testing. Fixes have been made to the support for the FT5206 touch controller and merged into the master branch; the touch handler is working nicely now.

ImpulseAdventure, 2020-02-24:
Closing since the changes for the initial support of **RA8876** and **FT5206** have been integrated into the master. Testing was done on the **EastRising 7" ([ER-TFTM070-6](https://www.buydisplay.com/spi-7-inch-tft-lcd-dislay-module-1024x600-ra8876-optl-touch-screen-panel))** display.

Related config files were created for this board:
- ard-adagfx-ra8876-ft5206.h
- ard-adagfx-ra8876-notouch.h

Outstanding work is currently ongoing in an attempt to add support for external (custom) fonts in the Flash memory. A separate issue / branch will be opened to address that feature.

closed 2020-02-24 by ImpulseAdventure