ImpulseAdventure/GUIslice #181: diag_ard_touch_calib problem

arnold-b, 2019-12-19:
Hello, 

i´m using
ESP32, TFTESPI ,XPT2046 

if on calibration i can not select the first cross. 
 
On my own raw data it Shows 
Pressure = 900, x = 3845, y = 769

But slection is only accepted at
x = 3655, y = 932

how i can change

ImpulseAdventure, 2019-12-19:
Hi arnold --

Some touch displays have less sensitivity in the corners. You can adjust the position of the calibration target to be further away from the corner by modifying the `TARGET_MARGIN` line in the `diag_ard_touch_calib` sketch. Try increasing the margin from 10 to 50 pixels, for example:

`#define TARGET_MARGIN 50`

Does that help?

arnold-b, 2019-12-19:
Yes. this works. 
but if my touch would have Problem in one Corner why i still can see the Position in raw data if i print it out

ImpulseAdventure, 2020-01-08:
Hi @arnold-b -- sorry for the delay as somehow I overlooked this update.

Glad to hear the margin setting worked.

Regarding your second question: you had asked why you could still see the raw position reported in the corner:
- Were you seeing the position reported from GUIslice's touch diagnostic (eg. with `DBG_TOUCH` enabled) or was it from the XPT2046 library's touch test example?
- There are a few reasons why the raw touch may be detected but the calibration won't proceed from the target. Usually this is either caused by the touch pressure value not meeting the range defined in your config (`ADATOUCH_PRESS_MIN`, `ADATOUCH_PRESS_MAX` -- you could try reducing PRESS_MIN) or that the debounce filter is detecting noisy input.

If you were interested in getting a more specific explanation, please modify your config file to enable `DBG_TOUCH`, set the `TARGET_MARGIN` back to 10 and then let me know what debug messages are reported while you press and release in the first corner target.

Thanks!

ImpulseAdventure, 2020-05-01:
I believe the original issue was resolved a while back. If there are further updates to your second question, please feel free to update this or else create a new issue. Thanks

closed 2020-05-01 by ImpulseAdventure