ImpulseAdventure/GUIslice #182: Compiling ex03_bld_btn_img cause compiler error

arnold-b, 2019-12-19:
ESP32_DEV,
Hello, 
if i try to compile ex03 excample i get many Compiler template Errors. 

In file included from j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\memory:67:0,

                 from J:\PrjArduino\hardware\espressif\esp32\libraries\FS\src/FS.h:24,

                 from J:\PrjArduino\libraries\GUIslice-master\src\GUIslice_drv_tft_espi.cpp:96:

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_raw_storage_iter.h:67:3: error: template with C linkage

   template <class _OutputIterator, class _Tp>

   ^

In file included from j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\memory:77:0,

                 from J:\PrjArduino\hardware\espressif\esp32\libraries\FS\src/FS.h:24,

                 from J:\PrjArduino\libraries\GUIslice-master\src\GUIslice_drv_tft_espi.cpp:96:

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:43:3: error: template with C linkage

   template<typename _Tp, typename _Alloc, typename = __void_t<>>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:47:3: error: template with C linkage

   template<typename _Tp, typename _Alloc>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:54:3: error: template with C linkage

   template<typename _Tp, typename _Alloc>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:66:3: error: template with C linkage

   template<typename _Alloc>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:69:3: error: template with C linkage

   template<typename _Alloc>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:72:3: error: template with C linkage

   template<bool, typename _Tp, typename _Alloc, typename... _Args>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:75:3: error: template with C linkage

   template<typename _Tp, typename _Alloc, typename... _Args>

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\uses_allocator.h:83:3: error: template with C linkage

   template<typename _Tp, typename _Alloc, typename... _Args>

   ^

[... 2711 lines cut ...]

   ^

j:\prjarduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\backward\auto_ptr.h:319:3: error: template with C linkage

   template<typename _Tp, typename _Dp>

   ^

Mehrere Bibliotheken wurden für "FS.h" gefunden
 Benutzt: J:\PrjArduino\hardware\espressif\esp32\libraries\FS
Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: J:\PrjArduino\hardware\espressif\esp32\libraries\SD
 Nicht benutzt: C:\Program
Mehrere Bibliotheken wurden für "GUIslice.h" gefunden
 Benutzt: J:\PrjArduino\libraries\GUIslice-master
Mehrere Bibliotheken wurden für "SPI.h" gefunden
 Benutzt: J:\PrjArduino\hardware\espressif\esp32\libraries\SPI
Mehrere Bibliotheken wurden für "TFT_eSPI.h" gefunden
 Benutzt: J:\PrjArduino\libraries\TFT_eSPI
 Nicht benutzt: J:\PrjArduino\libraries\TFT_eSPI
Mehrere Bibliotheken wurden für "XPT2046_Touchscreen.h" gefunden
 Benutzt: J:\PrjArduino\libraries\XPT2046_Touchscreen
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

ImpulseAdventure, 2019-12-20:
Hi @arnold-b --

What version of GUIslice were you using? (what was the startup message shown on the serial monitor when you ran a sketch with GSLC_SD_EN=0?)

I am not seeing the same error during my tests, but I think I know what might be causing the problem for you. I have made an update to `src/GUIslice_drv_tft_espi.cpp` and posted it in the branch [WIP182-ESP32_SD](https://github.com/ImpulseAdventure/GUIslice/tree/WIP182-ESP32_SD) branch. Could you try downloading that version of GUIslice and retrying the compile?
- Make sure you move out your existing Arduino/libraries/GUIslice directory
- Download and install the ZIP file from https://github.com/ImpulseAdventure/GUIslice/archive/WIP182-ESP32_SD.zip

thanks

arnold-b, 2019-12-20:
Hello, 

thanks. I will try. 
Just now i used your current master

arnold-b, 2019-12-20:
Hello, 

if i set GSLC_SD_EN=1, I can compile now, but the screen will be blank. 
Even on ex05_and_pages. if i use  GSLC_SD_EN=0 the sceen shows the sample.

Serial terminal is empty on both test

arnold-b, 2019-12-20:
Now it works. ADAGFX_PIN_SDCS was wrong.
thanks

closed 2019-12-20 by arnold-b

ImpulseAdventure, 2019-12-21:
Thank you arnold-b for confirming that the TFT_eSPI SD compilation fix works. I have merged the changes now.