ImpulseAdventure/GUIslice #183: Error to change Grahic object

arnold-b, 2019-12-20:
[test.zip](https://github.com/ImpulseAdventure/GUIslice/files/3989284/test.zip)

Hello, 

i just made one sample and i try to Change the Colors. I can select the object, but i can not change it. Everytime another object will be change. 

Please try to change the Button with the label todo. 

Thanks

Pconti31, 2019-12-20:
This is a bug with the GUIsliceBuilder not GUIslice.
A quick look shows that you have somehow duplicated key and enum values for the todo and 
L6 buttons which is why you can't change todo.

Please repost the issue inside the github for the builder issues.
[https://github.com/ImpulseAdventure/GUIslice-Builder/issues](url)
and post a full zip of your test folder which will have the full history of your changes
which are included inside test\gui_backup sub-folder so I can hunt down how you got into this situation.  Or if you know the steps to create the problem you should post that also.
Please also post version number of Builder the latest version of which is 0.13.b009

In the meantime I'll also try and repair your project so you can get back to work.

Also, why do you have a blank main page with everything on a base page?
The implementation for a base page is that it acts as an overlay on top of the main page not in place of it.  For example, if you have a status bar you want on each page you place it on a base page.
Paul--

Pconti31, 2019-12-20:
Here is your repaired project.
[test.zip](https://github.com/ImpulseAdventure/GUIslice/files/3990146/test.zip)
Good luck.
Paul--

arnold-b, 2019-12-20:
Hello, 

thats the Folder. 

First i created L1 then i copied it 8 times and renamed it. 
then i created a few other Buttons and delete some again. 
I can not remember all steps playing around
[test.zip](https://github.com/ImpulseAdventure/GUIslice/files/3990440/test.zip)

arnold-b, 2019-12-20:
Is it posible to get the Information or the tool how i can fix it myself ?