ImpulseAdventure/GUIslice #185: Support for ILI9225

cianpdx, 2020-01-07:
### Feature Request or Enhancement
Support for ILI9225

### Additional context
There seem to be two libraries commonly used for ILI9225 which are based on but not exactly compliant with Adafruit_GFX, so the functions called by GUISlice don't have corresponding functions in the driver. 

This seems most applicable:

https://github.com/Nkawu/TFT_22_ILI9225

Is it possible to turn off all functions except draw_point and align with the ILI9225 library draw_pixel function to test?

ImpulseAdventure, 2020-01-08:
Hi @cianpdx - yes, I should be able to add support for the ILI9225. Will take a look at the library and provide an update later.

ImpulseAdventure, 2020-01-09:
I have now added preliminary support for ILI9225 and pushed it to the following branch: [WIP185-ILI9225](https://github.com/ImpulseAdventure/GUIslice/tree/WIP185-ILI9225).

Note that I don't have an ILI9225 to test with at the moment, so I am not able to confirm operation, though it appears to compile successfully in my setup.

The ILI9225 library's font support is more involved than in other libraries, so I would recommend that we see if the default (built-in) font works first before trying the GFX fonts.

Please uncomment the new config file `ard-adagfx-ili9225-notouch.h` in your `GUIslice_config.h`.
I would recommend that you begin by commenting out `USE_STRING_CLASS` in `TFT_22_ILI9225.h` before trying it with `USE_STRING_CLASS` enabled.

If it compiles, I would be interested in hearing if example `ex02_ard_btn_txt` displays anything on your TFT.

thanks

cianpdx, 2020-01-10:
Hello, compiles fine with your suggestion of commenting out STRINGS

here is output of ex02

Thank you

[image]

ImpulseAdventure, 2020-01-11:
Thank you @cianpdx for testing it out! I have now pushed another change to the WIP185-ILI9225 branch that should add support for orientation on your display. With that update, hopefully it will show the correct orientation (you can change this with GSLC_ROTATE in your config) -- if you could confirm later, that would be great.

I have ordered an ILI9225 display for testing but it may take some time for it to arrive. In the meantime I will try to get the text rendering going for you.

ImpulseAdventure, 2020-01-13:
Great... what you are seeing is not actually an offset; instead example ex02 was intended for a larger display (most of the examples are targeted for a 320x240 display).

I have created an example that demonstrates a number of controls for a smaller display that should be better matched for your hardware: ex40_ard_ctrls_small

I still need to address the text rendering on your ILI9225, however.

ImpulseAdventure, 2020-02-06:
Hi @cianpdx -- my apologies as somehow I overlooked this issue...

Unfortunately the ILI9225 I ordered still hasn't arrived for me to test the code, though I expect it may arrive soon.

That said, I took another look at the driver tonight and think I see why justified text may not have been rendering properly. My guess is that unjustified (top-left alignment) text would have worked.

I have now pushed an update to the [WIP185-ILI9225](https://github.com/ImpulseAdventure/GUIslice/tree/WIP185-ILI9225) branch. I'm not sure if you are still keen on this particular driver (as I saw you were also using the HX8357/ILI9486), but if so, I would be interested to see if the example above works any differently.

thanks & sorry for the delay

cianpdx, 2020-02-07:
no need to apologize, don't owe me anything. 

I will jump back in this project this weekend and report back. 

re. HX8357 - that is for another project, just by chance I discovered guislice and want to integrate it into both projects to replace my hand made fragile gauges...

ImpulseAdventure, 2020-02-08:
The ILI9225 display actually arrived today.... but sure enough with no postal packaging & nicely damaged!

Despite having nearly all the display broken and only 6 scan lines somewhat intact, I still managed to test the GUIslice text output with the branch posted above. Looks like it is working for the built-in terminal font OK. Hopefully things look better on your side :)

[image]

cianpdx, 2020-02-11:
Looks great !
[image]

closed 2020-02-12 by ImpulseAdventure