ImpulseAdventure/GUIslice #186: Issue: MCUFriend on Mega shows white screen. HX8357/ILI9487

cianpdx, 2020-01-10:
### Describe the bug
All White Screen when running EX02

### Device hardware
- *MCU*: ATmega2650
- *Display with link*: Link says ILI9486 but comments say HX8357 or MCUfriend works fine. It is 16 bit 
- *GUIslice config selected*: ard-shld-mcufriend.h

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
I configured MCU to use a Special:
#define USE_MEGA_16BIT_SHIELD     // 2.14sec Mega2560 Shield 
and graphictest_kbv runs fine. 

### Expected behavior
ex02 test to run 

### Initialization messages
GUIslice version [0.13.0.14]:

- Init display handler [ADA_MCUFRIEND] OK

- No touch handler enabled

### Additional info
*Add any other context about the problem here.*

ImpulseAdventure, 2020-01-12:
Hi —

During initialization, the MCUfriend library attempts to determine / read an identifier from the display (readID). I haven’t seen the output from your graphicstest example (where it reports the ID), but it is likely that it may report it as a write-only display (ID=0xD3D3). With a write-only display, the current graphicstest_kbv example attempts to try forcing ID=0x9481.

We can do the same in GUIslice by adding the following to your current config file:

#define DRV_DISP_ADAGFX_MCUFRIEND_FORCE 0x9481

cianpdx, 2020-01-13:
Looks great!

[image]

ImpulseAdventure, 2020-01-13:
Great -- looks like it is working well.

I have also now updated the config file template for the **MCUFRIEND** supported displays to include the `FORCE` parameter so that it may be more obvious for users of write-only displays.

I'll close this now, but feel free to followup if you have further concerns.

closed 2020-01-13 by ImpulseAdventure