ImpulseAdventure/GUIslice #187: Fonts from Adafruit_GFX crash display

rotus8, 2020-01-16:
### Describe the bug
A simple display with one text element, created with Builder, works fine with the builtin font, but crashes when any of the Adafruit_GFX fonts are selected.
### Device hardware
- *MCU*: Adafruit Feather Huzzah ES32
- *Display with link*: Adafruit HX8357 3.5" resistive (https://www.adafruit.com/product/3651)
- *GUIslice config selected*: `esp-tftexpe-defualt-stmpe610` modified for my configuration

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* ??? Yes, everything works, including complex multipage layouts, only crashes with non-builtin fonts.
- *Touch works correctly with standalone touch library examples without GUIslice:* Yes.

### Expected behavior
I expect a simple text string to show up on the display "Font Test". Screen is blank. If I add additional elements after the text box, they don't show up and any other code is not executed. If there are elements inserted before the text, it does appear. Apparently the display list crashes when it gets to the text.

### Initialization messages
GUIslice version [0.13.0]:
- Init display handler [TFT_eSPI] OK
- Init touch handler [STMPE610(SPI-HW)] OK

### Additional info
Using Arduino 1.8.10, Adafruit GFX library 1.7.4, GUIslice 0.13.0, Builder 0.13.b012, TFT_eSPI 1.4.20
Project and .ino files:
[fontTest.zip](https://github.com/ImpulseAdventure/GUIslice/files/4073100/fontTest.zip)

ImpulseAdventure, 2020-01-17:
Hi @rotus8 -- thank you for the detailed report and attaching an example!

The TFT_eSPI display library handles GFX fonts differently than other display libraries in that it supplies its own copy of the fonts instead of including `Adafruit_GFX.h`. This has meant that the Builder code generator needed to offer a "Target Platform" config option specific to TFT_eSPI output mode.

My guess is that you are probably generating code with the Builder configured for an `arduino` *Target Platform*. Since you are using the **TFT_eSPI** display driver, we would need to select `arduino TFT_eSPI` instead. I realize that this setting is very easy to overlook.

[image]

Let me know if that addresses the issue for you.

@Pconti31 -- although we have a brief mention of this in the Builder user guide "Introduction" section, I'm wondering if there might be another good place to add this note for users?

Another idea: perhaps we could have the Builder provide a compile-time error if they didn't set the correct platform? For example, something like the following:

```c++
<FONT_ADAFRUIT>
#if defined(DRV_DISP_TFT_ESPI)
  #error Builder config "Target Platform" should be "arduino TFT_eSPI"
#endif 
#include <Adafruit_GFX.h>
// Note that these files are located within the Adafruit-GFX library folder:
<STOP>

<FONT_TFT_ESPI>
#if !defined(DRV_DISP_TFT_ESPI)
  #error Builder config "Target Platform" should be "arduino"
#endif 
#include <TFT_eSPI.h>
<STOP>
```
Thoughts?

rotus8, 2020-01-17:
Thank you for the rapid and complete response.
        Also, thank you for the work you have put in on this, you have
        saved me a ton of work and allowed me to make a much nicer UI
        for my little home brew project.
    Yes, that was my problem. I agree that a compile
        time error would be really helpful.
    Ted
    
    On 1/16/2020 11:22 PM, Calvin Hass
      wrote:
    
    
      
      Hi @rotus8
        -- thank you for the detailed report and attaching an example!
      The TFT_eSPI display library handles GFX fonts differently than
        other display libraries in that it supplies its own copy of the
        fonts instead of including Adafruit_GFX.h. This
        has meant that the Builder code generator needed to offer a
        "Target Platform" config option specific to TFT_eSPI output
        mode.
      My guess is that you are probably generating code with the
        Builder configured for an arduino Target
          Platform. Since you are using the TFT_eSPI
        display driver, we would need to select arduino TFT_eSPI
        instead. I realize that this setting is very easy to overlook.
      
      Let me know if that addresses the issue for you.
      @Pconti31
        -- although we have a brief mention of this in the Builder user
        guide "Introduction" section, I'm wondering if there might be
        another good place to add this note for users?
      Another idea: perhaps we could have the Builder provide a
        compile-time error if they didn't set the correct platform? For
        example, something like the following:
      
        <FONT_ADAFRUIT>
#if defined(DRV_DISP_TFT_ESPI)
  #error Builder config "Target Platform" should be "arduino TFT_eSPI"
#endif 
#include <Adafruit_GFX.h>
// Note that these files are located within the Adafruit-GFX library folder:
<STOP>

<FONT_TFT_ESPI>
#if !defined(DRV_DISP_TFT_ESPI)
  #error Builder config "Target Platform" should be "arduino"
#endif 
#include <TFT_eSPI.h>
<STOP>
      
      Thoughts?
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ImpulseAdventure/GUIslice/issues/187?email_source=notifications\u0026email_token=ALMK4WEJXMRMLK6743EPKPTQ6FMCNA5CNFSM4KHYZZW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJGYYQQ#issuecomment-575507522",
"url": "https://github.com/ImpulseAdventure/GUIslice/issues/187?email_source=notifications\u0026email_token=ALMK4WEJXMRMLK6743EPKPTQ6FMCNA5CNFSM4KHYZZW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJGYYQQ#issuecomment-575507522",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Pconti31, 2020-01-17:
I'll add the compile error messages to Builder 0.13.b013 but with a small change:
`<FONT_ADAFRUIT>`
`#if defined(DRV_DISP_TFT_ESPI)`
 ` #error Builder config "Edit->Options->General->Target Platform" should be "arduino TFT_eSPI"`
`#endif` 
`#include <Adafruit_GFX.h>`
`// Note that these files are located within the Adafruit-GFX library folder:`
`<STOP>`
`<FONT_TFT_ESPI>`
`#if !defined(DRV_DISP_TFT_ESPI)`
 ` #error Builder config "Edit->Options->General->Target Platform" should be "arduino"`
`#endif` 
`#include <TFT_eSPI.h>`
`<STOP>`
By the way, Cal knows that the items in angle brackets '<>' are commands to the builder for template expansion and won't show up in your generated code. 
Paul--

ImpulseAdventure, 2020-01-18:
Great - thanks for the followup.

closed 2020-01-18 by ImpulseAdventure