ImpulseAdventure/GUIslice #189: ColorSwitch for BGR Displays

GoGIGA, 2020-01-21:
#184 # Feature Request or Enhancement
Thanks a lot, a really nice project!

Could you please add in GUIslice_drv_tft_espi.cpp an option for flipping from RGB TO BGR, because 
there are some some strange displays with BGR coding?

### Additional context
I have patched the function "gslc_DrvAdaptColorToRaw" in my local copy to:
uint16_t gslc_DrvAdaptColorToRaw(gslc_tsColor nCol)
{
  uint16_t nColRaw = 0;

  #ifdef BGR
      // RGB565
      nColRaw |= (((nCol.b & 0xF8) >> 3) << 11); // Mask: 1111 1000 0000 0000
      nColRaw |= (((nCol.g & 0xFC) >> 2) <<  5); // Mask: 0000 0111 1110 0000
      nColRaw |= (((nCol.r & 0xF8) >> 3) <<  0); // Mask: 0000 0000 0001 1111
  #else
      // RGB565
      nColRaw |= (((nCol.r & 0xF8) >> 3) << 11); // Mask: 1111 1000 0000 0000
      nColRaw |= (((nCol.g & 0xFC) >> 2) <<  5); // Mask: 0000 0111 1110 0000
      nColRaw |= (((nCol.b & 0xF8) >> 3) <<  0); // Mask: 0000 0000 0001 1111
  #endif

  return nColRaw;
}
And this works fine for text and lines.

ImpulseAdventure, 2020-01-22:
Hi @GoGIGA -- sure, we could add an option for this.
It looks like certain ILI9341 or ST7789 displays could potentially require this mode.

Would it be safe to assume that your TFT_eSPI `User_Setup` would configure the `TFT_RGB_ORDER` flag (ie. `TFT_BGR` or `TFT_RGB`)? If so, I could make GUIslice_drv_tft_espi look for this User_Setup setting (and default to `TFT_RGB` if unset).

GoGIGA, 2020-01-22:
Hallo ImpulseAdventure,
thank you for your response. I have patched in the lokal copy of GUIslice_drv_tft_espi.cpp the above code and inserted the "#define BGR" in User_setup.h of TFT_eSPI an it worked flawlessly. And yes it is an ILI9341-Display from China ...

ImpulseAdventure, 2020-01-23:
Thanks.... I can easily add a check for a new flag “BGR” but wondered if we should reuse the TFT_RGB_ORDER flag that might already exist in the User_Setup template.

Could you attach your original User_Setup file to this issue or let me know whether it might contain a line like the following:

#define TFT_RGB_ORDER TFT_BGR

Thanks!

GoGIGA, 2020-01-23:
It is your project - It was just my "temporary" selection for testing.
Thank a lot for your project again!

GoGIGA, 2020-01-23:
Sorry, I was too quick. I have not used a define of the kind: #define TFT_RGB_ORDER TFT_BGR.
Is there already a solution in TFT_eSPI for this type of displays?

GoGIGA, 2020-01-23:
I do a test without my #define BGR and with #define TFT_RGB_ORDER TFT_BGR in a minute.

GoGIGA, 2020-01-23:
Well, the simple test failed. I'm still a beginner, how do I have to change the preprocessor command in my patch if is use #define TFT_RGB_ORDER TFT_BGR for the selection? Thanks!

GoGIGA, 2020-01-23:
My workaround is now:
--> In "User_setup.h"  from TFT_eSPI 
    #define TFT_BGR
    #define TFT_RGB_ORDER TFT_BGR
--> Changed Patch in your "GUIslice_drv_tft_espi.cpp" to:
// Convert from RGB struct to native screen format
// TODO: Use 32bit return type?
uint16_t gslc_DrvAdaptColorToRaw(gslc_tsColor nCol)
{
  uint16_t nColRaw = 0;
  #if defined(TFT_RGB_ORDER) && defined(TFT_BGR)
      // RGB565
      nColRaw |= (((nCol.b & 0xF8) >> 3) << 11); // Mask: 1111 1000 0000 0000
      nColRaw |= (((nCol.g & 0xFC) >> 2) <<  5); // Mask: 0000 0111 1110 0000
      nColRaw |= (((nCol.r & 0xF8) >> 3) <<  0); // Mask: 0000 0000 0001 1111
  #else
      // RGB565
      nColRaw |= (((nCol.r & 0xF8) >> 3) << 11); // Mask: 1111 1000 0000 0000
      nColRaw |= (((nCol.g & 0xFC) >> 2) <<  5); // Mask: 0000 0111 1110 0000
      nColRaw |= (((nCol.b & 0xF8) >> 3) <<  0); // Mask: 0000 0000 0001 1111
  #endif

  return nColRaw;
}

ImpulseAdventure, 2020-01-24:
Hi -- thanks for sharing your workaround!

> I have not used a define of the kind: #define TFT_RGB_ORDER TFT_BGR.
Is there already a solution in TFT_eSPI for this type of displays?

From understanding, yes, the TFT_eSPI library driver already supports the MADCTL Blue/Red swap mode in the ILI9341 controller without needing any changes to GUIslice. This is why I was asking if you might already be setting `TFT_RGB_ORDER` within your TFT_eSPI `User_Setup`. So, for your particular TFT display, I believe you should probably set this option first:

1) If you **remove** all of your GUIslice workaround edits and just set the following in your TFT_eSPI User_Setup file, do the colors look right?
`#define TFT_RGB_ORDER TFT_BGR`

2) If that does not work, then I will add a mode to GUIslice to allow for the swap like you indicated in your workaround.

thanks!

GoGIGA, 2020-01-24:
Well, I am really sorry for having wasting your time ....
You are absolutely right, when the option is set (and not overwriten later - what was my fault), there is no need for my patch. Thank you for directing my to the right way!

closed 2020-01-24 by GoGIGA

ImpulseAdventure, 2020-01-24:
Glad to hear it worked! And no worries about wasting time :) The TFT_RGB_ORDER was a relatively recent feature Bodmer added to his TFT_eSPI library a couple months ago.