ImpulseAdventure/GUIslice #190: Can the spinner arrows be reversed?

spiff72, 2020-01-22:
### Feature Request or Enhancement
I am no expert in GUI design, but for some reason, I expected that the down (decrement) button on the spinner would be on the left, and the up (increment) spinner would be on the right.  Can this be added as an option in the spinner?

### Additional context
This is just a "nice to have" feature for me.  I just found this GUISlice and created a working GUI for a printer enclosure exhaust fan control in 1 day - completely solving my issue of figuring out how to debounce touchscreen buttons!  

Thanks for building this tool - it is amazing!

ImpulseAdventure, 2020-01-22:
Hi @spiff72 -- glad to hear the tool has been useful! Your request seems reasonable. I anticipate we might find people who prefer both placement orders :)

We could add an option for this; it would likely require an extra widget parameter setting in the Builder and an updated API. In the meantime, you should be able to change your existing **XSpinner** widget to work in the way you have requested with the following modification to your `/elem/XSpinner.c` file: (see the red highlights for changes)

[image]

Let me know if that works

spiff72, 2020-01-23:
Thanks!  I will give that a shot later.

spiff72, 2020-01-23:
Oops - wait a minute - what you show above is already exactly what my XSpinner.c file shows.  Are you saying that I need to swap the INC and DEC and the /030 and /031?

ImpulseAdventure, 2020-01-23:
Great! Thanks for confirming that it works. I will close this for now, but can reopen it if it looks like integrating the added flexibility into the Builder / APIs seems worthwhile.

closed 2020-01-23 by ImpulseAdventure