ImpulseAdventure/GUIslice #195: Report warning if SD card image requested w/o SD enabled

ImpulseAdventure, 2020-02-18:
### Describe the bug
- Currently, if the SD card feature has been disabled (ie. `GSLC_SD_EN=0`) and an attempt is made to display an image button (with an image sourced from an SD card), the image will not be rendered and no warning/error message is reported.

### Device hardware
- *MCU*: Arduino-compatible variants
- *GUIslice config selected*: `DRV_DISP_ADAGFX_*`

### Expected behavior
- An error message should be added into `DrvDrawImage()` to trap the case in `GSLC_IMGREF_SRC_SD` mode when `GSLC_SD_EN=0`.

Pconti31, 2020-02-19:
"No warning/error message is reported" is not quite TRUE.  Not that adding a runtime check isn't useful but as an FYI the Builder already outputs a compile time warning if users asked for images to be loaded from SD card and the #define GSLC_SD_EN is not set to 1.   
The person who had this problem had built their app with a very old version of the Builder that lacked this check.
Paul--

ImpulseAdventure, 2020-10-30:
Fixed to now report a runtime error if an API call is made to draw an image from SD while GSLC_SD_EN=0.

closed 2020-10-30 by ImpulseAdventure