ImpulseAdventure/GUIslice #199: XSpinner Element missing arrows if non built-in fonts used

Pconti31, 2020-02-25:
### Describe the bug
If user request non-built-in font for the XSpinner element, boxes are shown not the up/down arrows.

### Device hardware
All

### Expected behavior
Suggest adding up/down character setting as part of API so user can change it.
Say by using '+' and '-' characters or whatever.

This was reported to me by G. Evangelista as a Builder issue.
[image]

Pconti31, 2020-03-02:
Cal:
I have attached a possible solution by adding a new API call to XSpinner
and tested by modifying ex32_bld_spinner to make the optional new call.
The normal built-in arrows '\30' and '\31' are used if no makes this call.
`///`
`/// Set Up and Down characters for the Spinner element`
`/// - Called during redraw`
`///`
`/// \param[in]  pvGui:       Void ptr to GUI (typecast to gslc_tsGui*)`
`/// \param[in]  pElemRef:    ptr to ElementRef`
`/// \param[in]  cIncrement:  Character to use to indicate incrementing the spinner`
`/// \param[in]  cDecrement:  Character to use to indicate decrementing the spinner`
`///`
`/// \return true if success, false otherwise`
`///`
`bool gslc_ElemXSpinnerSetChars(void* pvGui,gslc_tsElemRef* pElemRef,uint8_t cIncr, uint8_t cDecr);`

I have attached my new version of XSpinner
Paul--

[XSpinner.zip](https://github.com/ImpulseAdventure/GUIslice/files/4277651/XSpinner.zip)

ImpulseAdventure, 2020-03-03:
Paul - that sounds like a good solution. If you end up creating a PR, I can merge it. Otherwise I’ll take care of it when I return on the weekend. Thanks!

ImpulseAdventure, 2020-03-03:
Fixed and merged

closed 2020-03-03 by ImpulseAdventure