ImpulseAdventure/GUIslice #201: Different result with and without touch versions of MCUFriend library

dduehren, 2020-03-01:
### Describe the bug
I am running on an Arduino Nano 33 IOT using a updated version of MCUFriend_kbv library.  I am trying to get GUIslice running too.  
I get different results depending on which version of the "driver" I select.
  //#include "../configs/ard-shld-mcufriend.h"
  #include "../configs/ard-shld-mcufriend_4wire.h"

The 4wire version just shows a light blue screen. No box and no text.
The other version shows a box and text.  See attached picture below.  But obviously no touch

### Device hardware
-Arduino Nano 33 IOT with Kuman 3.5" TFT 480x320 touch screen

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*
Seems irrelevant since at least one version works.
*Please confirm whether:*
 Display works with all MCUFRIEND examples and Kuman examples with "flaky" touch.

[image]

### Expected behavior
 The 4Wire library should work as well as the other.  And the box should look like your post of it.

### Initialization messages
 I don't know if I got this set up correctly as I'm not sure what the exact debug line should be.  I tried a #define DEGUG_ERR_2 but that didn't show anything.  

### Additional info
*Add any other context about the problem here.*

ImpulseAdventure, 2020-03-01:
Hi David —

Can you confirm:
- Adafruit_Touchscreen example (touchtestdemo.ino) works with your display. For reference: https://github.com/adafruit/Adafruit_TouchScreen/blob/master/examples/touchscreendemo/touchscreendemo.ino
- If the above works, then please indicate what  XP/XM/YP/YM settings were needed in your Adafruit touchtestdemo sketch.
- From there we can copy these same settings into your GUIslice ard-shld-mcufriend_4wire.h config
- At that point please proceed with the touch diagnostics described in the wiki here: https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-4-wire

Thanks

dduehren, 2020-03-01:
I’ll check that after I figure out a problem in my GuiSluce design - i must have hidden elements in it

David Duehren

Hi David —

Can you confirm:

Adafruit_Touchscreen example (touchtestdemo.ino) works with your display. For reference: https://github.com/adafruit/Adafruit_TouchScreen/blob/master/examples/touchscreendemo/touchscreendemo.ino
If the above works, then please indicate what XP/XM/YP/YM settings were needed in your Adafruit touchtestdemo sketch.
From there we can copy these same settings into your GUIslice ard-shld-mcufriend_4wire.h config
At that point please proceed with the touch diagnostics described in the wiki here: https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-4-wire
Thanks

dduehren, 2020-03-01:
The software runs but the values aren’t correct.  I suspect that there’s more to configuring the pins for the Nano 33 IOT and why using a separate controller works.   In the meantime I got my design to compile and download.  First screen is ok but missing a button image.  

I will probably wait until the STMPE610 is working.  Then I will need to integrate that in with the tFT display library

David Duehren

Hi David —

Can you confirm:

Adafruit_Touchscreen example (touchtestdemo.ino) works with your display. For reference: https://github.com/adafruit/Adafruit_TouchScreen/blob/master/examples/touchscreendemo/touchscreendemo.ino
If the above works, then please indicate what XP/XM/YP/YM settings were needed in your Adafruit touchtestdemo sketch.
From there we can copy these same settings into your GUIslice ard-shld-mcufriend_4wire.h config
At that point please proceed with the touch diagnostics described in the wiki here: https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-4-wire
Thanks

dduehren, 2020-03-02:
For whatever reason, the box and text do not show up when I run with the MCUFriend-4 wire library.  I am going to leave this alone until I get the STMPE610 working.  Then I’ll go through the config files more carefully. 

It looks to me that there are at least three config files that need to be checked: GUIslice_config,
..._config.ard, and the ard-shield-mcufriend 

It’s good enough for now that I compiled a GUIslice design and downloaded it to the Nano and the top menu was displayed.  Gives me hope for progress next week.

Regards

David Duehren

Hi David —

Can you confirm:

Adafruit_Touchscreen example (touchtestdemo.ino) works with your display. For reference: https://github.com/adafruit/Adafruit_TouchScreen/blob/master/examples/touchscreendemo/touchscreendemo.ino
If the above works, then please indicate what XP/XM/YP/YM settings were needed in your Adafruit touchtestdemo sketch.
From there we can copy these same settings into your GUIslice ard-shld-mcufriend_4wire.h config
At that point please proceed with the touch diagnostics described in the wiki here: https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-4-wire
Thanks

dduehren, 2020-03-03:
I was able to get sensible readings from the software only 4 wire touch sense on the Arduino Nano 33 IOT by making an empirical linear mapping that I think is ultimately due to the difference in supply voltage (5 vs 3.3).  

The question is where should something like that be put when working with your libraries.

I may yet go with the STMPE610 but the pin sharing makes it tricky.  So for now, I’ll use the software version.  

Thanks

David

Sent from my iPhone

David Duehren

dduehren, 2020-03-04:
Hi Calvin,

When I run your version of touch detect it stops sometime after displaying on the touch screen “Detecting Pins” “See Serial Output” and the “Broken Touchscreen”

David’s test, with your text still on the screen,
Runs to completion fine except for the result which says 
Diagnosing as:
YP, YM: (A2, D5) = 41
YP, YM: (A2, D6) = 41

Yes that’s right, no XM and two different pins for YM.

Meanwhile the phone call sketch works with my empirical scaling, but tftpaint does not.

I’m using the same pins A2, A3 and D0 and D9 as on the UNO - matching the shield.

Tomorrow I continue my investigation

David Duehren

The software runs but the values aren’t correct.  I suspect that there’s more to configuring the pins for the Nano 33 IOT and why using a separate controller works.   In the meantime I got my design to compile and download.  First screen is ok but missing a button image.  

I will probably wait until the STMPE610 is working.  Then I will need to integrate that in with the tFT display library

David Duehren

Hi David —

Can you confirm:

Adafruit_Touchscreen example (touchtestdemo.ino) works with your display. For reference: https://github.com/adafruit/Adafruit_TouchScreen/blob/master/examples/touchscreendemo/touchscreendemo.ino
If the above works, then please indicate what XP/XM/YP/YM settings were needed in your Adafruit touchtestdemo sketch.
From there we can copy these same settings into your GUIslice ard-shld-mcufriend_4wire.h config
At that point please proceed with the touch diagnostics described in the wiki here: https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-4-wire
Thanks

ImpulseAdventure, 2020-03-04:
Hi — it doesn’t look like your touchscreen is working correctly with the 4-wire touch library.

Were you running David’s mcufriend_kbv diagnose_Touchpins.ino sketch? If not, could you try that and confirm whether it reports a successful result?

Thanks

dduehren, 2020-03-04:
Calvin,

I have it working very well with tftpaint which uses the map function to scale the raw readings and the phoncall sketch using my empirical scaling.  I think they are similar.  

The question is where does something like a scaling factor go in GUISlice?  

As for the test results, have either of you done testing on a 3.3v platform?  

And where does scaling for different size screens go since the ADC range is fixed regardless of the screen size?  That might be the place to put it for GUIslice too.  

Now onto trying to get it running on the Nano

David Duehren

Hi — it doesn’t look like your touchscreen is working correctly with the 4-wire touch library.

Were you running David’s mcufriend_kbv diagnose_Touchpins.ino sketch? If not, could you try that and confirm whether it reports a successful result?

Thanks

dduehren, 2020-03-07:
Hi Calvin,

 

I’m now back to debugging GUISlice.  

 

It still works with ard-shield-mcufriend but not ard-shield-4wire.  I verified that it’s the interaction with the touch driver by changing the DRV_TOUCH definition in the 4wire file.  With Ex01 (basic test) it doesn’t draw a box if Touch is enabled but does if it isn’t.

 

I set debug to level 2 but that didn’t report anything.

 

I also added a section to the 4wire file to scale for the Nano 33 IOT that I found by trial and error with ttfpaint on the Nano

 

All of the examples that came with the board, including the two touch examples, both of which use the MCEFriend library, and the GLUE_DEMO_480x320, all work just fine now.  So it’s really something in the config files or some conflict in your software that is where the problem must be.  

 

So I’m looking for suggestions on how to debug this.  It’s the big block to my project right now.  I’ve attached the ard-shield-mcufriend-4wire for your review. (You can see how I added support for the nano).  I also looked at the GuiSlice_config_ard.h file though I don’t know if it matters.  I’ve attached that as well.  I haven’t changed any other files.

 

Thanks,

David Duehren

 

#ifndef _GUISLICE_CONFIG_ARD_H_
#define _GUISLICE_CONFIG_ARD_H_

// =============================================================================
// GUIslice library (example user configuration) for:
//   - CPU:     Arduino UNO / MEGA / etc
//   - Display: MCUFRIEND
//   - Touch:   Simple Analog (Resistive)
//   - Wiring:  Shield (pinout defined by mcufriend_kbv library)
//
//   - Example display:
//     - MCUFRIEND 2.4" TFT 4-wire Touch shield (UNO/MEGA)
//
// DIRECTIONS:
// - To use this example configuration, include in "GUIslice_config.h"
//
// WIRING:
// - As this config file is designed for a shield, no additional
//   wiring is required to support the GUI operation
//
// =============================================================================
// - Calvin Hass
// - https://github.com/ImpulseAdventure/GUIslice
// =============================================================================
//
// The MIT License
//
// Copyright 2016-2019 Calvin Hass
//
// Permission is hereby granted, free of charge, to any person obtaining a copy
// of this software and associated documentation files (the "Software"), to deal
// in the Software without restriction, including without limitation the rights
// to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
// copies of the Software, and to permit persons to whom the Software is
// furnished to do so, subject to the following conditions:
//
[... 845 lines cut ...]
  //#define DBG_FRAME_RATE    // Enable diagnostic frame rate reporting
  //#define DBG_DRAW_IMM      // Enable immediate rendering of drawing primitives
  //#define DBG_DRIVER        // Enable graphics driver debug reporting

// =============================================================================
// INTERNAL CONFIGURATION
// - Users should not need to modify the following
// =============================================================================

  // Display device string is only used in LINUX drivers
  #define GSLC_DEV_TOUCH ""   // No device path used

  // Define compatibility for non-AVR to call PROGMEM functions
  #if defined(__AVR__)
    #define GSLC_USE_PROGMEM 1
  #else
    #define GSLC_USE_PROGMEM 0
  #endif

// =============================================================================

#ifdef __cplusplus
}
#endif // __cplusplus
#endif // _GUISLICE_CONFIG_ARD_H_

ImpulseAdventure, 2020-03-08:
Hi David --

Sorry for the delay as I was out of town.

Thanks for taking the time to test & share a few of your experiments, however I would like to request a couple clarifications from the earlier messages. If you can try each of these specific tests, it should help narrow down your settings *before* we try GUIslice:

- Q1) Confirm "Adafruit_Touchscreen" library example (`touchscreendemo.ino`) works with your display. For reference: https://github.com/adafruit/Adafruit_TouchScreen/blob/master/examples/touchscreendemo/touchscreendemo.ino

- Q2) If the Adafruit_Touchscreen/touchscreendemo works, then please indicate what XP/XM/YP/YM settings were needed in your Adafruit touchscreendemo sketch. Did you have to modify the sketch?

- Q3) Please run the "MCUfriend_kbv" library example (`TouchScreen_Calibr_native.ino`) and confirm it works with your display. Did you have to modify the sketch? Please copy the output of this sketch at attach it to this issue.

Later, once we're running GUIslice, note that there should only be two config files to look at:
- `/src/GUIslice_config.h` (this one simply selects which custom config file to load, eg. `ard-shld-mcufriend_4wire.h`)
- `/configs/ard-shld-mcufriend_4wire.h` (this is the config file you may need to adjust)
- The `GUIslice_config_ard.h` is an old config file that is not used anymore (I intend to delete it to avoid confusion)

Thanks!

dduehren, 2020-03-08:
Hi Calvin,

 

I will get back to you, probably tomorrow.  I think I just bricked my Nano, or at least the USB interface.  I have a couple on order that should arrive on Monday sometime.

 

I did get some preliminary results using the Touchscreendemo that comes with the Touch library from Adafruit.  They were much different from the ones I got using the same library with the demos that came with the Touchscreen.  They use the Adafruit_GFX, the MCUFriend library and the Adafruit Touchscreen (aka simple analog) library.   The demo program was recording many false hits whereas there were none with the ttfpaint and phoncall sketches even though I configured them the same in terms of the Analog lines.  One possibility is that setting up graphics on the touch screen that happens in the two other sketches does not happen in the touch demo sketch. 

 

Here is the configuration that has been working for all of the other sketches that use touch.

CS = A3 = Yp

CD/RS = A2 = Xm

D8 = Ym

D9 = Xp

 

I also measured the X resistance to 287 ohms and found that 300 works well enough.

 

The rest of the pins are defined as they are for the UNO.  In these cases, X is the 320 direction and Y is the 480 direction.  

 

The 4 corners were touched for each of the sketches.  

Touchdemo  (1) X=875, y=905 z =83   (2) X=751, y=863 z = 95  (3) x=929, y-674  (4) x=890, y=880.  These make no sense.  

 

Phonecal (1)  X=43, Y=81  (mapped to 4,10) (2) X=238, y=75 (mapped 328, 2) (3) 237, 439 mapped 326, 466) and (4) 40, 433

 

Ttfpaint (1) 207, 805 z=460 translated 6,2  (2) 208, 323 (3) 871, 321 (4) 850, 807.

 

Once I have a working Nano again, I will dig into the Touchdemo variations.  They are just way off compared to the other two that I know work.  So we’ll see what gives,  hopefully tomorrow.  And if you look at the ard-shield-4wire file I sent yesterday, you’ll see that I’ve already put in the pins and the MIN/MAX info, I hope I’ve done it the right way.  

 

I don’t know if the issues I’m seeing with the TouchDemo are related to why Touch doesn’t work with GUISlice on the Nano or not.  A bit more investigation is required.  It maybe that some initialization of the other control lines to the LCD screen are in order.  It seems bad design to use the CS signal as one of the two analog inputs for touch, but that’s the way the Touchscreen is designed. 

 

Hopefully I’ll know more by end of day tomorrow.

 

Thanks,

 

David

dduehren, 2020-03-09:
Hi Calvin,

 

I have done all the tests that you requested, but the problem remains, as I suspected it would.  I wanted to see if it occurred on the UNO, but Ex02 is too big, even the version in the Arduino_min folder.  

 

On the Nano 33 IOT, I was able to run both the Adafruit demo (with modifications described below) and MCUFriend Touch-Shield-New. In both cases I got reasonable results and averaged the two to set the values in ard-shld-mcufriend_4wire as follows:

 

1.	Here is the section of the 4wire file. 
2.	  #define ADATOUCH_PIN_YP   A3
3.	  #define ADATOUCH_PIN_XM   A2
4.	  #define ADATOUCH_PIN_YM   9
5.	  #define ADATOUCH_PIN_XP   8
6.	  #if defined(__SAMD21G18A__) && defined(ARDUINO_SAMD_NANO_33_IOT)
7.	  #define ADATOUCH_Y_MIN 196
8.	  #define ADATOUCH_Y_MAX 880
9.	  #define ADATOUCH_X_MIN 334
10.	  #define ADATOUCH_X_MAX 806
11.	  #else
12.	  #define ADATOUCH_X_MIN    130
13.	  #define ADATOUCH_X_MAX    905
14.	  #define ADATOUCH_Y_MIN    75
15.	  #define ADATOUCH_Y_MAX    930
16.	  #endif
17.	  #define ADATOUCH_REMAP_YX 0    // Some touch controllers may swap X & Y cords

 

The  MCUFriend Touch-Shield-New results for the NANO_33_IOT (a 3.3 Volt part).

Top:  x=607, y=876

Left: x=347, y=578

Rgt: x=811, y=530

Bot: x=573, y=202

 

I also calibrated the UNO using this sketch

T: 83, 526

L: 528, 904

R: 496, 140

B: 935, 492

 

To get the Adafruit touch demo to work properly, I had to make several modifications.  Without these, the program reported many, many false positives whether running on a Nano or an UNO.  I was able to get it to behave properly by setting some of the other pins to HIGH values.   I have attached to code.

T: x=554, y=884

L: x=320, y=573

R: x=801, y=621

B: x=579, y=189

 

But even with these settings, when I run Ex02 with touch enabled, no box is drawn.  If it’s not enabled, - one line in 4wire, it does draw the box.  And I don’t know if this happens on UNOs because the code is too big to try.  So if you have some ideas on how to debug this, what file to add some report in, etc. I’m all ears.  It must be in the init function somewhere.  Does Touch init happen before or after the graphic init?  Maybe flipping them would be worthwhile.

 

Thanks,

 

David

 

// - https://github.com/ImpulseAdventure/GUIslice
// =============================================================================
//

  //   https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-Touch-Support
  // -----------------------------------------------------------------------------

  // . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .

  #define ADATOUCH_Y_MIN 196
  #define ADATOUCH_Y_MAX 880
  #define ADATOUCH_X_MIN 334
  #define ADATOUCH_X_MAX 806
  #else
  #define ADATOUCH_X_MIN    83
  #define ADATOUCH_X_MAX    935
  #define ADATOUCH_Y_MIN    140
  #define ADATOUCH_Y_MAX    905
  

  //   https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-Pressure

dduehren, 2020-03-10:
A few other thoughts.

 

I don’t know how many different processors you’ve ported your code to, but one area where I’ve seen trouble, besides pin definitions, etc. is with digitalPintoPort and related calls.  The SAMD processors use 32 bit pointers vs 8 for the UNO.  So there maybe some kind of compatibility problem there.  And like I said, I couldn’t reproduce it on the UNO because the code is too big.

 

Maybe a smaller sketch that does less but can run with and without Touch enabled on the UNO would give us some relevant info.

 

Thanks

 

David Duehren

ImpulseAdventure, 2020-03-10:
Hello David --

Your update above confirms that your display utilizes pin-sharing between the TFT and the touch overlay. The Adafruit_Touchscreen library has known compatibility issues with pin-sharing TFTs (as is typical with some mcufriend displays and your particular Kuman display), so this is very likely contributing to the problem you are seeing.

Using the mcufriend_kbv library on an UNO leaves very little Flash memory for any GUI library, so I would stick with the Nano 33 IOT.

I have a modified version of GUIslice that I would be interested in having you try out once your Nano is available again. If you can send an email to guislice@gmail.com, I'll send you a test version.

thanks!

ImpulseAdventure, 2020-04-28:
For reference, with the update to version 0.14.0.4, the primary issue involving touch interaction with pin sharing displays should be resolved.

closed 2020-04-28 by ImpulseAdventure