ImpulseAdventure/GUIslice #202: Text box background not restored when switching pages

rotus8, 2020-03-02:
### Describe the bug
When switching pages to a page with a text box, the text box background fill is not restored

### Device hardware
- *MCU*: ESP32
- *Display with link*: Adafruit 3.5" 480x320 TFT FeatherWing
https://learn.adafruit.com/adafruit-3-5-tft-featherwing/overview

- *GUIslice config selected*:  `esp-tftesps-default-stmpe610.h'

Test Case:
[textBoxTest1.zip](https://github.com/ImpulseAdventure/GUIslice/files/4272770/textBoxTest1.zip)

Load test case on hardware, display should come up with yellow text on a blue background and a LOG button. Pressing the LOG button takes you to a page with a text box. The text box should have a black background with yellow text. Pressing the TEXT button adds more text.

Pressing EXIT takes you back to the main screen. Pressing LOG again takes you to the log screen, but this time the text box has a blue background fill, with yellow/black text. Pressing TEXT adds a line of text, and restores the proper black background fill.

### Additional info
Arduino 1.8.12
GuiSlice 0.13.0.19

ImpulseAdventure, 2020-03-02:
Hi - Thanks for taking the time to create a testcase.

I am out of town at the moment so can’t run your test case, but perhaps you could try the following modification to your `/src/elem/XTextbox.c`  file line 464 (Latest version on GitHub) and let me know if it works differently for you?
From:

```c++
if (pBox->nRedrawRow == XTEXTBOX_REDRAW_ALL) {
```
To:
```c++
if ((pBox->nRedrawRow == XTEXTBOX_REDRAW_ALL) ||  (eRedraw == GSLC_REDRAW_FULL)) {
```

Pconti31, 2020-03-02:
Calvin and Ted: 
I tested Ted's sample with the code change Calvin recommended and it runs correctly.
I have attached the new version of XTextbox.c to save you the effort.
You just need to confirm things are working for you.

Paul--
[XTextbox.zip](https://github.com/ImpulseAdventure/GUIslice/files/4275892/XTextbox.zip)

rotus8, 2020-03-02:
Excellent. That fixed it.
Thanks!

Ted

ImpulseAdventure, 2020-03-08:
Fix has been merged.

closed 2020-03-08 by ImpulseAdventure