ImpulseAdventure/GUIslice #207: Set a slider position from ESP?

lukecorkill, 2020-03-14:
Hi
I've just started using GUIslice, and think it's great.  I have it running on an ESP8266 with ILI9341.  
I have been working through some of the examples and have created a few test screens with the Builder.

I can successfully **read** from a slider in the Arduino IDE, but can someone clarify if it's possible to **set** the value of a slider from the microcontroller?  

I'm developing a remote volume control for an amplifier, so need to update the slider on-screen if the main control is moved.  

I had a look through the examples and can see how to write text etc - but couldn't see an example of writing to a slider.

Can someone confirm if it's possible / share an example? 

Many thanks

Luke

Pconti31, 2020-03-14:
Yes, you can
```
///
/// Set a Slider element's current position
///
/// \param[in]  pGui:        Pointer to GUI
/// \param[in]  pGui:        Pointer to GUI
/// \param[in]  pElemRef:    Pointer to Element reference
/// \param[in]  nPos:        New position value
///
/// \return none
///
void gslc_ElemXSliderSetPos(gslc_tsGui* pGui,gslc_tsElemRef* pElemRef,int16_t nPos);
```
Although, Usually you would use a progress bar to display output.  
Se ex04.  The Slider being more to give input.
Paul--

lukecorkill, 2020-03-14:
That's brilliant - it worked first time.  

`gslc_ElemXSliderSetPos(pGui,m_pElemSlider1_2, nVal);
`
Many thanks.

closed 2020-03-14 by lukecorkill