ImpulseAdventure/GUIslice #208: Memory cost

Lucy2003, 2020-03-15:
### Describe the bug
I'm using esp8266.When I set up a base page as a taskbar,and setup a page (page1) as a window(just like windows),it works well.But when I open the window from base page,it will refresh the window.Then I open window from base page about 6 times,ESP8266 is restarted by watchdog...I think it's because the memory overflow.

On the base page,I just added a button and show page1 when press the button.Should I clean all the page before I show page1?

### Device hardware
- *MCU*: 2.8' tft display (ili9341) with xpt2046 touch
- *GUIslice config selected*:  `esp_tft_espi`

### Checklist to try first

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice: Yes.
- *Touch works correctly with standalone touch library examples without GUIslice:Yes

### Expected behavior
...

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*

### Additional info
*Add any other context about the problem here.*

Pconti31, 2020-03-15:
It's not much help but I ran your program on my Adafruit ESP32 Feather with Adafruit's Feather 3.5 TFT screen and once I commented out setting pin D2 it compiled and ran fine.
Any chance setting D2 Boot mode select TX1 is causing a problem?

FYI, Heap size was always reported as 318156 bytes.

Now GUIslice only uses global memory pre-allocated once at startup so it's not directly the problem
as far as I can see.  So no you don't need to cleanup pages.

Hopefully, Calvin can help you out. 

Paul--

ImpulseAdventure, 2020-03-15:
Hi @Lucy2003 -- good job on the interface you have created!

I ran your TestGUI on the following setup:
- **MCU**: ESP8266 (Arduino IDE "Node MCU 1.0 (ESP-12E Module)")
- **TFT**: Adafruit 2.8" ILI9341 Resistive Shield (hand-wired)
- **GUIslice setup**: `esp-tftespi-default-stmpe610.h`
- **TFT_eSPI setup**:
```c++
#define ILI9341_DRIVER
#define TFT_CS   PIN_D1
#define TFT_DC   PIN_D3
#define TFT_RST  -1
#define TOUCH_CS PIN_D0
...
```

With the above, I didn't observe any faults after swapping between various pages.
Can you confirm the steps you are using to recreate the problem? For example:
- Click Start, Machine, Window X
- Click Start, Machine, Window X
- Click Start, Machine, Window X
- ...

As Paul points out, GUIslice should not use any dynamic memory allocation, and we appear to have a good amount of memory left after compilation.

Thanks

Lucy2003, 2020-03-16:
I reconnected the wire and found the problem disappear.It seems that the problem causes by my wire.
### I'm sorry to disturb you!

ImpulseAdventure, 2020-03-16:
Great - thank you for the confirmation!

closed 2020-03-16 by ImpulseAdventure