ImpulseAdventure/GUIslice #210: Displaying variables

Joshi115, 2020-03-17:
[My Hardware:
-ELEGOO Mega2560 R3 [Amazon Link](https://www.amazon.de/gp/product/B01MA5BLQI/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)
-Generic 3.5 Inch Touch Display [Amazon Link](https://www.amazon.de/gp/product/B07KK4NV3V/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)
-My Skills: Newbie ]

Hi,
I'm not sure if I can ask questions here or if I should go to the Arduino Forum next time. 

**What do I want to do (at least for now / very basic for my program):**

I have a pH Meter and I would like to display the _always changing_ variable in my text box.
For that I have a working pH Meter Script:
`
#include <Arduino.h>
#include <Wire.h>
 
 
int phvalue2;
int phValue;
 
const float m = -0.1867;
const int adcPin = A0;
 
void setup() {
  Serial.begin(9600);
}
 
void loop(){
    phvalue();
}
 
 
void phvalue(){
 
  float Po = analogRead(adcPin) * 5.0 / 1024;
  float phValue = 7 - (2.5 - Po) / m;
  Serial.print("ph value = ");
  Serial.println(phValue);
  phValue = phvalue2;
 delay(1000);
   
}
`
In the end, I get my variable phValue or phvalue2. Both of them work and I get a reading for them through Serial Monitor.

Now I thought that I just have to copy it into my GUISlice Projekt.
So I created a Page with just a Text Box.
Then I used the code from an Example (Exp. 18) to get it running.

This is what I put together:
`
//<File !Start!>
// FILE: [ddddd.ino]
// Created by GUIslice Builder version: [0.13.b020]
//
// GUIslice Builder Generated File
//
// For the latest guides, updates and support view:
// https://github.com/ImpulseAdventure/GUIslice
//
//<File !End!>
//
// ARDUINO NOTES:
// - GUIslice_config.h must be edited to match the pinout connections
//   between the Arduino CPU and the display controller (see ADAGFX_PIN_*).
//
 
// ------------------------------------------------
// Headers to include
// ------------------------------------------------
#include "GUIslice.h"
#include "GUIslice_drv.h"
 
int phvalue2;
int phValue;
 
const float m = -0.1867;
const int adcPin = A7;
 
 
// Include any extended elements
//<Includes !Start!>
//<Includes !End!>
 
// ------------------------------------------------
// Headers and Defines for fonts
// Note that font files are located within the Adafruit-GFX library folder:
// ------------------------------------------------
//<Fonts !Start!>
#if defined(DRV_DISP_TFT_ESPI)
  #error Builder config "Edit->Options->General->Target Platform" should be "arduino TFT_eSPI"
#endif
#include <Adafruit_GFX.h>
// Note that these files are located within the Adafruit-GFX library folder:
#include "Fonts/FreeMono9pt7b.h"
//<Fonts !End!>
 
// ------------------------------------------------
// Defines for resources
// ------------------------------------------------
//<Resources !Start!>
//<Resources !End!>
 
// ------------------------------------------------
// Enumerations for pages, elements, fonts, images
// ------------------------------------------------
//<Enum !Start!>
enum {E_PG_MAIN};
enum {phbox};
// Must use separate enum for fonts with MAX_FONT at end to use gslc_FontSet.
enum {E_FONT_MONO9,MAX_FONT};
//<Enum !End!>
 
// ------------------------------------------------
// Instantiate the GUI
// ------------------------------------------------
 
// ------------------------------------------------
// Define the maximum number of elements and pages
// ------------------------------------------------
//<ElementDefines !Start!>
#define MAX_PAGE                1
 
#define MAX_ELEM_PG_MAIN 1 // # Elems total on page
#define MAX_ELEM_PG_MAIN_RAM MAX_ELEM_PG_MAIN // # Elems in RAM
//<ElementDefines !End!>
 
// ------------------------------------------------
// Create element storage
// ------------------------------------------------
gslc_tsGui                      m_gui;
gslc_tsDriver                   m_drv;
gslc_tsFont                     m_asFont[MAX_FONT];
gslc_tsPage                     m_asPage[MAX_PAGE];
 
//<GUI_Extra_Elements !Start!>
gslc_tsElem                     m_asPage1Elem[MAX_ELEM_PG_MAIN_RAM];
gslc_tsElemRef                  m_asPage1ElemRef[MAX_ELEM_PG_MAIN];
 
#define MAX_STR                 100
 
//<GUI_Extra_Elements !End!>
 
// ------------------------------------------------
// Program Globals
// ------------------------------------------------
 
// Save some element references for direct access
//<Save_References !Start!>
gslc_tsElemRef*  phvaluebox        = NULL;
//<Save_References !End!>
 
// Define debug message function
static int16_t DebugOut(char ch) { if (ch == (char)'\n') Serial.println(""); else Serial.write(ch); return 0; }
 
// ------------------------------------------------
// Callback Methods
// ------------------------------------------------
//<Button Callback !Start!>
//<Button Callback !End!>
//<Checkbox Callback !Start!>
//<Checkbox Callback !End!>
//<Keypad Callback !Start!>
//<Keypad Callback !End!>
//<Spinner Callback !Start!>
//<Spinner Callback !End!>
//<Listbox Callback !Start!>
//<Listbox Callback !End!>
//<Draw Callback !Start!>
//<Draw Callback !End!>
//<Slider Callback !Start!>
//<Slider Callback !End!>
//<Tick Callback !Start!>
//<Tick Callback !End!>
 
// ------------------------------------------------
// Create page elements
// ------------------------------------------------
bool InitGUI()
{
  gslc_tsElemRef* pElemRef = NULL;
 
//<InitGUI !Start!>
  gslc_PageAdd(&m_gui,E_PG_MAIN,m_asPage1Elem,MAX_ELEM_PG_MAIN_RAM,m_asPage1ElemRef,MAX_ELEM_PG_MAIN);
 
  // NOTE: The current page defaults to the first page added. Here we explicitly
  //       ensure that the main page is the correct page no matter the add order.
  gslc_SetPageCur(&m_gui,E_PG_MAIN);
 
  // Set Background to a flat color
  gslc_SetBkgndColor(&m_gui,GSLC_COL_WHITE);
 
  // -----------------------------------
  // PAGE: E_PG_MAIN
 
 
  // Create phbox text label
  pElemRef = gslc_ElemCreateTxt(&m_gui,phbox,E_PG_MAIN,(gslc_tsRect){214,129,44,25},
    (char*)"",0,E_FONT_MONO9);
  gslc_ElemSetCol(&m_gui,pElemRef,GSLC_COL_GRAY,GSLC_COL_BLACK,GSLC_COL_BLACK);
  phvaluebox = pElemRef;
//<InitGUI !End!>
 
  return true;
}
 
void setup()
{
  // ------------------------------------------------
  // Initialize
  // ------------------------------------------------
  Serial.begin(9600);
  // Wait for USB Serial
  //delay(1000);  // NOTE: Some devices require a delay after Serial.begin() before serial port can be used
 
  gslc_InitDebug(&DebugOut);
 
  if (!gslc_Init(&m_gui,&m_drv,m_asPage,MAX_PAGE,m_asFont,MAX_FONT)) { return; }
 
  // ------------------------------------------------
  // Load Fonts
  // ------------------------------------------------
//<Load_Fonts !Start!>
    if (!gslc_FontSet(&m_gui,E_FONT_MONO9,GSLC_FONTREF_PTR,&FreeMono9pt7b,1)) { return; }
//<Load_Fonts !End!>
 
  // ------------------------------------------------
  // Create graphic elements
  // ------------------------------------------------
  InitGUI();
 
//<Startup !Start!>
//<Startup !End!>
 
}
 
// -----------------------------------
// Main event loop
// -----------------------------------
void loop()
{
  float Po = analogRead(adcPin) * 5.0 / 1024;
  float phValue = 7 - (2.5 - Po) / m;
  phValue = phvalue2;
 
  Serial.print("ph value = ");
  Serial.println(phValue);
 
   delay(1000);
 
   char              acTxt[4];
 
  gslc_tsElemRef*  phvaluebox        = gslc_PageFindElemById(&m_gui,E_PG_MAIN,phbox);
   
 
  snprintf(acTxt,4,"%u",phvalue2);
  gslc_ElemSetTxtStr(&m_gui,phvaluebox,acTxt);
 
 
  // ------------------------------------------------
  // Update GUI Elements
  // ------------------------------------------------
 
  //TODO - Add update code for any text, gauges, or sliders
 
  // ------------------------------------------------
  // Periodically call GUIslice update function
  // ------------------------------------------------
  gslc_Update(&m_gui);
   
} 
`

When I'm playing around with
`  snprintf(acTxtNum,2,**"%u"**,phValue);
`  (with %u) then I usually get different results.

As a result, I get:

- Only the first number of my pH Value (My reading in Serial Monitor: 5.85 / Textbox displays 5)

**or**

- A bunch of long numbers which are kinda associated with my pH Value (that means first 2 numbers are correct but then 5 random numbers)

**or** 

- Just a bunch of random numbers or symbols

I think that I'm missing something here... Maybe with the snprintf string?
Maybe I'm defining the variables wrong? I really don't know.

Please keep your work up! It's great and I love it!

Joshua  

Sources and Websites I have already looked at:
[FAQ](https://github.com/ImpulseAdventure/GUIslice-Builder/releases/download/v0.13.b020/FAQ.pdf)
[User-Guide](https://github.com/ImpulseAdventure/GUIslice/wiki/User-Guide)
[Other Guide](https://impulseadventure.github.io/GUIslice/group____ElemUpd__.html#ga577779166e89c03abe61102bf8673071)
[sprintf Guide](https://arduinobasics.blogspot.com/2019/05/sprintf-function.html)

Pconti31, 2020-03-18:
First on an Arduino snprintf doesn't support floating point numbers
Second even if it did %u would be wrong, %f for floating point
say 5 long and two decimals would be "%5.2f". If it worked, but not supported
so you need to use dtostrf.
Now I'm doing this from memory so you might need to experiment with this code or google
but you want something like:
Note the extra include files, I don't remember which one you need but including both won't cause any harm.
```
#include "GUIslice.h"
#include "GUIslice_drv.h"

#include <stdio.h>
#include <stdlib.h>

  float Po = analogRead(adcPin) * 5.0 / 1024;
  float phValue = 7 - (2.5 - Po) / m;
  
   char              acTxt[4];
   gslc_tsElemRef*  phvaluebox        = gslc_PageFindElemById(&m_gui,E_PG_MAIN,phbox);

//   snprintf(acTxt,4,"%u",phvalue);
  dtostrf(phValue, 4, 0, acTxt );
gslc_ElemSetTxtStr(&m_gui,phvaluebox,acTxt);
```
Paul--

Joshi115, 2020-03-18:
Hi Paul,

Thank you very much! It works now without a charm! 
I'm so happy right now! Thank you again. I was working on this for hours and couldn't figure it out.

Another small question (I haven't taken a look at it yet):
I also want to add a Graph later.
So the values from my pH Meter on the x - Axis.
And on my y- Axis should be my values from another variable (simple counter variable / every time my pump goes on it adds 5 [ml] to my y Axis counter variable).

Is this possible? I haven't taken a look at it. 

Thank you again very much,

Joshua

Pconti31, 2020-03-18:
Yes, easy fast way and hardway.
Easy: using Builder drop graph element where you want it on display, size height and width,
set Maximum Points, say 100. and inside loop() call 
```
gslc_ElemXGraphAdd()
```
to add points. However, it's not very pretty.  Can't say I ever liked it. See ex11_bld_graph first.

Hardway, draw graph yourself, updating in a draw callback and using the tick callback to check your meter and trigger drawing callback.  See ex06_bld_callback.

If you run into trouble send an email to Support: [guislice@gmail.com](url)
or to me by going to my profile and using my public email address.

If you however run into a bug  or want to request enchancements please open a new issue.
The issues isn't really for C programming questions since it's not constructed as real forum it's just a flat list of bugs enhancement requests ,or questions of how to use the project.  THe issues list would get out of hand quickly if we ran it like the Arduino forum.

Also, Please close this issue if we are done. 
PS: I'm going to add your questions to my GUIsliceBuilder's FAQ.
Paul--

closed 2020-03-18 by Joshi115