ImpulseAdventure/GUIslice #211: Problems compiling for Nano 33 IoT

Channy00, 2020-04-05:
### Describe the bug
I created a small Test GUI and was able to succesfully upload it to the Arduino Uno and it was functioning perfectly fine. But whenever I try to compile it for Nano 33 IoT an Error message shows up. 

### Device hardware
- *MCU*: Arduino Uno and Nano 33 IoT
- *Display with link*:  Adafruit ILI9341 2.8" resistive (https://www.adafruit.com/product/1770)
- *GUIslice config selected*:  `ard-adagfx-ili9341-simple`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* Yes
- *Touch works correctly with standalone touch library examples without GUIslice:* Yes

### Expected behavior
Compile and Upload the code to the Nano 33 IoT

### Initialization messages
In file included from C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:69:0,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd.h:105,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/sam.h:540,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.1\cores\arduino/Arduino.h:48,

                 from C:\Users\name\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:5,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:55:

C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:226:0: warning: "LITTLE_ENDIAN" redefined

 #define LITTLE_ENDIAN          1

 

In file included from c:\users\name\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\types.h:67:0,

                 from c:\users\name\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\stdio.h:61,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice.h:43,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.h:46,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:43:

c:\users\name\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\machine\endian.h:17:0: note: this is the location of the previous definition

 #define LITTLE_ENDIAN _LITTLE_ENDIAN

 

In file included from C:\Users\name\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.1\cores\arduino/delay.h:23:0,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.1\cores\arduino/Arduino.h:81,

                 from C:\Users\name\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:5,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:55:

C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp: In function 'int gslc_TDrvGetPinMode(uint8_t)':

C:\Users\name\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.1\variants\nano_33_iot/variant.h:50:35: error: invalid conversion from 'PortGroup*' to 'uint8_t {aka unsigned char}' [-fpermissive]

 #define digitalPinToPort(P)      (&(PORT->Group[g_APinDescription[P].ulPort]))

                                  ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1590:31: note: in expansion of macro 'digitalPinToPort'

### Additional info
I used the newest version of GUISlice, Adafruit Touchscreen and the Arduino IDE.
I appologize if I made an obvious mistake, I am a rather "middle class" programmer and spend several hours trying to figure out where the problem is. I thank you for your help and efforts and keep up the great work!

ImpulseAdventure, 2020-04-05:
Hi @Channy00 — I haven’t done local testing on the Nano 33 IOT yet (though I know other users are working with it). However, I think we may just need to disable the pin sharing enhancement that is active by default in the “simple” touch driver.

To do this, try modifying line 940 of GUIslice_drv_adagfx.cpp to the following:

```c++
#if defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__) || defined(SAMD_NANO_33_IOT)
```

Does that make a difference for the compile?

Channy00, 2020-04-06:
Hi @ImpulseAdventure - Thank you for your quick response. I now tried it with the modification you said but it still won't compile. My GUIslice_drv_adagfx.cpp looks now somewhat like this:
```
   // Handle multi-line text:
    // If we just output a newline, Adafruit-GFX will automatically advance
    // the Y cursor but reset the X cursor to 0. Therefore we need to
    // readjust the X cursor to our aligned bounding box.
    if (ch == '\n') {
      #if defined(DRV_DISP_ADAGFX_RA8875)
      nCurPosY = m_disp.getCursorY();
      if (bInternal8875Font) {
        // TODO: Is getCursorY() supported in RA8875 mode?
        m_disp.textSetCursor(nTxtX, nCurPosY);
      }
      #elif defined(DRV_DISP_ADAGFX_ILI9341_DUE_MB)
      nTxtY += m_disp.getFontHeight();
      m_disp.cursorToXY(nTxtX,nTxtY);
      #else
      nCurPosY = m_disp.getCursorY();
      m_disp.setCursor(nTxtX,nCurPosY);
      #endif
    }
	#if defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__) || defined(SAMD_NANO_33_IOT)
    bFirstChar = false;

  } // while(1)

```

The error message which the compiler gives me looks like this:

```
In file included from C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:69:0,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd.h:105,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/sam.h:540,

                 from C:\Users\name\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.6\cores\arduino/Arduino.h:48,

                 from C:\Users\name\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:5,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:55:

C:\Users\name\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:226:0: warning: "LITTLE_ENDIAN" redefined

 #define LITTLE_ENDIAN          1

 

In file included from c:\users\name\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\types.h:67:0,

                 from c:\users\name\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\stdio.h:61,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice.h:43,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.h:46,

                 from C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:43:

c:\users\name\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\machine\endian.h:17:0: note: this is the location of the previous definition

 #define LITTLE_ENDIAN _LITTLE_ENDIAN

 

C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:36:0: error: unterminated #if

 #if defined(DRV_DISP_ADAGFX) || defined(DRV_DISP_ADAGFX_AS)

 
[... 80 lines cut ...]
 #define portModeRegister(port)   (&(port->DIR.reg))

                                         ^

C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1593:31: note: in expansion of macro 'portModeRegister'

     volatile uint8_t *nReg  = portModeRegister(nPort);

                               ^~~~~~~~~~~~~~~~

C:\Users\name\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.6\variants\nano_33_iot/variant.h:53:41: error: base operand of '->' is not a pointer

 #define portOutputRegister(port) (&(port->OUT.reg))

                                         ^

C:\Users\name\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1599:31: note: in expansion of macro 'portOutputRegister'

     volatile  uint8_t *nOut = portOutputRegister(nPort);

                               ^~~~~~~~~~~~~~~~~~

exit status 1
Error compiling for board Arduino NANO 33 IoT.
```

Pconti31, 2020-04-06:
Well if you modified line 940 no wonder you can't compile.
undo that change and remove that extra #endif you added and
change the correct line 1940 which currently looks like
```
  #if defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__)
    #undef FIX_4WIRE_PIN_STATE
  #endif
```
and change to:
```
  #if defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__)  || defined(SAMD_NANO_33_IOT)
    #undef FIX_4WIRE_PIN_STATE
  #endif
```
Now I have no idea if that will solve your problem or not but at least you won't be adding to them.
Paul--

Channy00, 2020-04-06:
Hi Paul - Thank you for your hint, I was wondering why that line was empty. I also looked up the line 1940 but there was something for the XPT2046 controller. The line which you mentioned is in my "GUIslice_drv_adagfx.cpp"  file on the line 1560. But changing it still didn't made any difference.

ImpulseAdventure, 2020-04-06:
Sorry for the confusion... this is what happens when you try to respond via a little screen on a phone :)

I was intending to recommend a change to line 1940, and adding in the compilation macro for the Arduino Nano 33 IOT to exclude the "4-wire pin mode fixes". It wasn't clear to me what macro was used by the Nano 33 IOT, and from a quick search of the net it looked like `SAMD_NANO_33_IOT` was necessary.

Upon testing it out on my computer now, I have determined that the macro should have been `ARDUINO_SAMD_NANO_33_IOT`.

Line 1940 should read:
```c++
#if defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__)  || defined(ARDUINO_SAMD_NANO_33_IOT)
```
The above appears to compile fine in my local testing within the Arduino IDE.

**NOTE**: I have made this change to the repository now so you can download the latest version there.

Hope that works better! :)

closed 2020-04-06 by Channy00