ImpulseAdventure/GUIslice #216: ard-shld-mcufriend_4wire fails to compile with Arduino Due

zacpod, 2020-04-17:
### Describe the bug
ard-shld-mcufriend_4wire fails to compile with Arduino Due.
ard-shld-mcufriend works, so I think it's a problem with the DRV_TOUCH_ADA_SIMPLE piece.

### Device hardware
- *MCU*: Arduino Due (SAMD, I think).
- *Display with link*: mcufriend 2.4"  
- *GUIslice config selected*: ard-shld-mcufriend_4wire

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* Yep.  Even the guislice stuff works as long as I don't enable touch.  
- *Touch works correctly with standalone touch library examples without GUIslice:* Yes.  Adafruit touch examples work, and some stuff I coded uses the mcufriend libraries work, too.

### Expected behavior
Example code compiles.

### Initialization messages
In file included from C:\Users\zac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:201:0,

                 from C:\Users\zac\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:5,

                 from C:\Users\zac\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:55:

C:\Users\zac\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp: In function 'int gslc_TDrvGetPinMode(uint8_t)':

C:\Users\zac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\variants\arduino_due_x/variant.h:63:65: error: invalid conversion from 'Pio*' to 'uint8_t {aka unsigned char}' [-fpermissive]

 #define digitalPinToPort(P)        ( g_APinDescription[P].pPort )

### Additional info
  // Pin connections from diag_ard_touch_detect:
  #define ADATOUCH_PIN_YP     A2
  #define ADATOUCH_PIN_XM     A3
  #define ADATOUCH_PIN_YM     8 
  #define ADATOUCH_PIN_XP     9 
(Those values work in the ada touch example, and in the mcufriend examples.)

ImpulseAdventure, 2020-04-18:
Hi @zacpod -- thanks for reporting this!

Yes, I had integrated some additional workarounds to improve stability of the Adafruit_Touchscreen library (selected by `FIX_4WIRE_PIN_STATE`), but these are not compatible with the Due. I have now pushed a fix to the repository that should enable you to proceed with compiling the configuration on the Due. Let me know if that helps.

zacpod, 2020-04-18:
Thanks for the quick response!
Sketch compiles now, but doesn't display anything.  Screen blinks at startup but just stays blank otherwise.
.
When I recompile using the non-touch mcufriend, I get a display (but no touch, obvs)
But when using configs/ard-shld-mcufriend_4wire.h I just get a blank screen.

Serial output looks basically the same except the touch handler.
GUIslice version [0.13.1.3]:
- Init display handler [ADA_MCUFRIEND] OK
- Init touch handler [SIMPLE(Analog)] OK
GUIslice version [0.13.1.3]:
- Init display handler [ADA_MCUFRIEND] OK
- No touch handler enabled

ImpulseAdventure, 2020-04-18:
Thanks for testing out the updated version.

It looks like your MCUFRIEND display uses pin-sharing between the TFT interface and the resistive touch overlay (I think you have A2=YP=LCD_RS & A3=XM=LCD_CS). These displays often have troubles using the default Adafruit_Touchscreen library as it was not intended to support pin-sharing displays (various pins are left in the incorrect state after performing a touch access).

This was precisely the purpose of the `FIX_4WIRE_PIN_STATE` workarounds I had implemented (but not yet for Due).

Can you confirm what display controller (eg. ILI9341, etc.) your MCUFRIEND display is using? Or what ID the mcufriend_kbv examples report during startup?

We may need to use an alternate touchscreen library for your display, or see if we can implement the same workarounds for the Due. These workarounds involve saving and restoring the pin states.

Note that we have had mixed successes trying to workaround pin-sharing with other users. In some cases we still ended up with some inconsistent operation, whereas for others it all works fine.

zacpod, 2020-04-18:
Yup, it for sure uses shared pins for the screen and touch sensor.  The XM and YP pins are both shared.
ID is 0x9341.

In the example code for the screen there's always a call to set the pinmode to output after every read of the touch sensor, if that helps.
```
        tp = ts.getPoint();
        pinMode(XM, OUTPUT);
        pinMode(YP, OUTPUT);
```

zacpod, 2020-04-22:
Nope, I couldn't find another getPoint() being called in the GUISlice code with the #defines I had in place.  Not sure why the screen goes blank after selecting an item.

I tried one of the other demos and it worked perfectly, so may just be something with the listbox control?  Which I don't actually need in my project.  So I think we're all good here.  

Feel free to close this - and thanks so much for your time!

ImpulseAdventure, 2020-04-27:
Hi @zacpod --

I revisited the Due (SAM3X8E) datasheet tonight and stepped through the Adafruit_Touchscreen code and think I discovered one of the potential causes of conflict between the mcufriend_kbv library and the Adafruit_Touchscreen library's operation (I have recreated the issue outside of GUIslice). In order to provide robust handling for certain touch scenarios, GUIslice uses both `getPoint()` as well as `pressure()` Adafruit APIs. Looking at it further, it looks like the Adafruit_Touchscreen library leaves the `YM` pin as an input after `pressure()`. This would interfere with the parallel 8080-8 MCUFRIEND display operation.

I have pushed a change to the GUIslice repository now that also restores this pin. In my local testing it looks like the Due runs much better with my pin-sharing MCUFRIEND (ILI9341) display (previously I could experience a white screen or visual gitches).

Please try the latest version of GUIslice (0.14.0.3+) and let me know if it works any better for you.

thanks!

zacpod, 2020-04-30:
Works perfectly now!  Thanks so much!

ImpulseAdventure, 2020-05-01:
Great - thank you for the confirmation

closed 2020-05-01 by ImpulseAdventure