ImpulseAdventure/GUIslice #218: ESP32 + ST7735S 1.8 (128x160)

shafr, 2020-04-27:
### Describe the bug
It seems like resolution is wrong when using display `ST7735 1.8"`.

### Device hardware
- *MCU*: ESP32
- *Display with link*: [ST7735S 1.8 with SD card](http://www.lcdwiki.com/1.8inch_Arduino_SPI_Module_ST7735S_SKU:MAR1801)
- *GUIslice config selected*: custom config that is based on arduino st7735 and esp32.

### Checklist to try first
*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* YES, TFT_eSPI examples work fine.
- *Touch works correctly with standalone touch library examples without GUIslice:* no Touch

### Expected behavior
Resolution is detected correctly.

### Additional info
I'm using [02_arduino example](https://raw.githubusercontent.com/ImpulseAdventure/GUIslice/master/examples/arduino_min/ex02_ardmin_btn_txt/ex02_ardmin_btn_txt.ino)

My configuration file is as follows:
```
#ifndef _GUISLICE_CONFIG_ARD_H_
#define _GUISLICE_CONFIG_ARD_H_

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

  #define DRV_DISP_TFT_ESPI         // bodmer/TFT_eSPI
  #define DRV_DISP_ADAGFX_ST7735    // Adafruit ST7735
  #define DRV_TOUCH_NONE            // No touch enabled

  #define TFT_WIDTH 128
  #define TFT_HEIGHT 160

  #define DRV_DISP_ADAGFX_ST7735_INIT   INITR_18REDTAB      // Red tab (1.8")

  #define GSLC_ROTATE     1

  #define DEBUG_ERR               2   // 1,2 to enable, 0 to disable

  #define GSLC_FEATURE_COMPOUND       0   // Compound elements (eg. XSelNum)
  #define GSLC_FEATURE_XTEXTBOX_EMBED 0   // XTextbox control with embedded color
  #define GSLC_FEATURE_INPUT          0   // Keyboard / GPIO input control

  #define GSLC_SD_EN    0

  #define GSLC_SPIFFS_EN               0 

  #define GSLC_TOUCH_MAX_EVT    1

  #define GSLC_SD_BUFFPIXEL   50

  #define GSLC_CLIP_EN 1

  #define GSLC_BMP_TRANS_EN     1               // 1 = enabled, 0 = disabled
  #define GSLC_BMP_TRANS_RGB    0xFF,0x00,0xFF  // RGB color (default: MAGENTA)

  #define GSLC_USE_FLOAT        0   // 1=Use floating pt library, 0=Fixed-point lookup tables

  #define GSLC_DEV_TOUCH ""
  #define GSLC_USE_PROGMEM      0

  #define GSLC_LOCAL_STR        0   // 1=Use local strings (in element array), 0=External
  #define GSLC_LOCAL_STR_LEN    30  // Max string length of text elements

  #define DBG_LOG           // Enable debugging log output
  #define DBG_TOUCH         // Enable debugging of touch-presses
  #define DBG_FRAME_RATE    // Enable diagnostic frame rate reporting
  #define DBG_DRAW_IMM      // Enable immediate rendering of drawing primitives
  #define DBG_DRIVER        // Enable graphics driver debug reporting

  // =============================================================================

#ifdef __cplusplus
}
#endif // __cplusplus
#endif // _GUISLICE_CONFIG_ARD_H_
```

Also TFT_eSPI is configured through `platformio.ini`:
```
build_flags =
  -DUSER_SETUP_LOADED
  -DST7735_DRIVER
  -DTFT_CS=5
  -DTFT_DC=19
  -DTFT_RST=-1
;   -DTFT_MISO=19
  -DTFT_MOSI=23
  -DTFT_SCLK=18
  -DLOAD_GLCD
  -DLOAD_FONT2
  -DLOAD_FONT4
  -DLOAD_FONT6
  -DLOAD_FONT7
  -DLOAD_FONT8
  -DSMOOTH_FONT
```

### LOG Output:
```
GUIslice version [0.14.0]:
- Init display handler [TFT_eSPI] OK
- No touch handler enabled
INFO: Added 1 elements to current page (max=2), ElemId=0 (FLASH)
INFO: Added 1 elements to current page (max=2), ElemId=1 (FLASH)
Update rate: 1 / sec
Update rate: 226 / sec
```

Image that is shown on the screen looks like this:
[image]

ImpulseAdventure, 2020-04-28:
Hi @shafr — thank you for providing good detail in your report. It looks like the GUI is displaying correctly. Note that the GUI does not perform any dynamic scaling. However, most examples were designed for the common display resolution of 320x240. For the reduced resolution ST7735 displays, we have a couple examples that will fit the display. For example: ex40_ard_ctrls_small.

Let me know if that works for your display. You can always adjust the dimensions and position of elements in the other examples to fit the smaller resolution if needed.

Thanks

shafr, 2020-04-28:
@ImpulseAdventure Thank you for update! Example that you have provided works well!

Just one note on ESP32 development, just not to create separate ticket:
* All examples require `#include "Arduino.h"` in their header in order for ESP32 to work. Maybe you can add it into documentation for ESP32.

closed 2020-04-28 by shafr