ImpulseAdventure/GUIslice #219: Single bit icons colour mauve

notascooby, 2020-04-30:
After upgrade to version 0.14.0 from 0.13.0 certain single colour bit maps are now mauve in colour.
There are many icons on my screen and not all have changed to mauve. The colour of the bitmaps are defined by 3 off 8 bit bytes (r, g, b). From the tests I have carried out so far it seems that any bit map that has the top bit of the 'r' byte set, that icon is displayed in mauve.
I know this is a little vague and I will carry on testing, but you may be able to point me in the right direction, knowing the changes that have occurred since version 0.13.0.
Not sure if this is anything to do with transparency since the colour is definitely a mauve but have tried setting GSLC_BMP_TRANS_EN to zero in both GUIslice_config_ard.h & esp-tftespi-default-xpt2046_int.h but this didn't help.
The icon colours are correct when reverting to 0.13.0 
Each of my icons are defined in the following format
const PROGMEM unsigned char image_data_sun_orange[]  = {
  0x00, 0x20, 0x00, 0x20, 0xff, 0xa5, 0x00, 0x00,   //32bit Width, 32bit Height, red, green, blue, 0x00
    0x00, 0x01, 0x80, 0x00, 
    etc
 }

### Device hardware
- *MCU*: ESP32
- *Display: ILI9486 RPI. Using current version of TFT_eSPI ver 2.2.0
- *GUIslice config selected*: esp-tftespi-default-xpt2046_int

Thanks in advance

ImpulseAdventure, 2020-05-01:
Thank you @notascooby for noticing this!
Yes, it does look like an issue slipped in to the color conversion while adding clipping support. Your observation regarding the leading bit was very helpful.

I have put together a quick patch and posted it in the following branch: [WIP219-FixColor](https://github.com/ImpulseAdventure/GUIslice/tree/WIP219-FixColor)
Please give this a try and let me know if it works any differently. Thanks

ImpulseAdventure, 2020-05-05:
Fix merged

closed 2020-05-05 by ImpulseAdventure

notascooby, 2020-05-20:
Sorry for late reply.
Yes, this issue is resolved