ImpulseAdventure/GUIslice #221: Wiki for User controls / More user friendly code

shafr, 2020-05-06:
### Feature Request or Enhancement
GUIslice is very advanced library for creating GUI for MCU's, but it is very harsh with newcomers, when they are facing C code and trying to figure out what to do (with devices that are not touch screen). Sadly some of us lack C/C++ programming experience and most of code that is done is just importing of libraries, instantiating them and using methods over those libraries in loop routine.

There are some different input devices:
* Button (Seen examples with EasyButton)
* Joystick
* Rotary Encoder with click function
* Wheel-button (Those, seen on Sony Ericsson back in a day).
* Accelerometer
* Gyroscope
* GPS
* Other sensors

Some of them don't have well implemented libraries with events / interrupts implemented so user have to do GPIO reading and handling in a loop or some other way. 

There few options that can help:

* Wiki page, with detailed explanation or simple example - how to assign GPIO state to a screen Event. Or if there would be a possibility `to assign variable` to GUI state - for cases where non-binary input device is used.
* Provide code example with minimal breadcrumbs (for example button and screen event declaration, and CbPinPoll method for event handling, etc).

shafr, 2020-05-06:
Something like here with steps:
https://github.com/igorantolic/ai-esp32-rotary-encoder#Details

ImpulseAdventure, 2020-05-07:
Hi @shafr -- those are some great suggestions.

Although we have the ex21_ard_input_pin example to demonstrate how to manipulate a GUI with external buttons, it is might not be generic enough for users to see how to integrate other input devices.

Let me look at some example peripheral libraries (perhaps a rotary encoder) or other GPIOs and see if I can create a simple example demonstrating how one can integrate. I would probably start with an new example, but may end up creating a wiki page (as you have indicated) if that looks more appropriate to handle different use-cases.

saikek, 2020-05-07:
Great!

One more question - I see that some documentation is in wiki, some in gh-pages branch. Is it all in Markdown or some of that is generated from other sources? (as in gh-pages).

It would be easier to migrate to Jekyll for example for better structure, reusability and layout.

MickTheMechanic, 2021-01-14:
How did you go with this? I am now introducing a rotary coder into my tire tool project. A simple example would be awesome.

ImpulseAdventure, 2021-01-17:
For reference, I posted an example of how one might do this with an encoder in #348 .

krdarrah, 2021-12-23:
Hello, I am also trying to get joystick control, but only see Focus Next and Previous actions available.  These work fine for left/right buttons, but how can up/down buttons be used?  Specifically, I'm trying to get these to work with the virtual keyboard... not user friendly to only have left/right navigation when typing.  Let me know if possible, thanks!