ImpulseAdventure/GUIslice #222: Special characters are not compatible with fonts driven by "TFT_eSPI"

guo652917087, 2020-05-07:
Hi,
     I use the "TFT_eSPI" driver and use the tools on "https://processing.org/" to generate the "TFT_eSPI" font. When the generated font is less than 127, it can work normally, but greater than 127, "Adafruit_GFX_Library" and "TFT_eSPI" display the font structure is different, or the display mode is different. How do I choose, I want to use the Chinese characters in the display part. At first I used the "Adafruit_GFX_Library" driver, but not the esp32 hardware spi driver, I had to use "TFT_eSPI".

ImpulseAdventure, 2020-05-07:
Hi @guo652917087 — I understand from Bodmer’s update in his repo that you may be able to use his “smooth” fonts in order to support sizes exceeding 127.

GUIslice should already be compatible with the TFT_eSPI smooth fonts. Please try example “ex19_ard_smooth“ and let me know if it works (note the directions in the sketch header comments).

Thanks

guo652917087, 2020-05-07:
Hi，
    @ImpulseAdventure ——Thank you very much for your reply. I am changing "TFT_eSPI" to c language because I am using the "esp-idf" branch of esp32 and I am not using the code of "arduino" I am also modifying your code "GUIslice" to the form supported by "esp-idf". After I verify, I will reply here.

guo652917087, 2020-05-07:
Hi,@ImpulseAdventure 
Hahaha, it works fine, thank you again.

closed 2020-05-07 by guo652917087