ImpulseAdventure/GUIslice #225: request to implement an element that is used on different pages

Avpi18, 2020-05-11:
the feature should make it possible to show the same element on different pages without further loading the RAM. 
e.g. Page_1 with Listbox_1 and Page_2 with Listbox_1 in the same place.

Pconti31, 2020-05-11:
@Avpi18 This is already possible.  In fact I recently saw an example where someone used a Listbox as
a menu structure going down the side of a page.  Anyway, just plop the control(s) down on a base page.  See ex24 where a base page is used to create tabs.
Paul--

Avpi18, 2020-05-11:
@Pconti31 Thanks for your answer. 
Maybe I did not express myself clearly and therefore I try again ;-) 
I have several pages whereof 2 are very similar. The only difference is the different number of buttons and other text. The problem is, if I insert a base_page, it will be the base. But I also have other pages which should not be linked to the base_page.
I have 2 pictures where you can see what i mean.

[image]
[image]

Pconti31, 2020-05-12:
@Avp18 If I were doing this kind of UI I would make Choose a Popup page.

Paul--

ImpulseAdventure, 2020-05-12:
Hi @Avpi18 & @Pconti31 --

Paul, thanks for taking the time to propose some potential workarounds.

The good news is that I think there is a much simpler way of achieving this.
After looking at the API further, I believe we can create an element "alias" that allows us to share items on multiple pages.

Please note: The following is a quick response to give you an idea of one solution, but I'll update this later with a clearer example for you to try out, assuming you are using the Builder.

I have tested this with a simple 3-page example based on ex05_ard_pages where I added an XListbox to the 2nd page (E_PG_EXTRA). I then added an element alias on the 3rd page (E_PG_EXTRA2). I wanted to provide a quick response now but then can followup with a clearer example later.

The simplest approach may be to do the following:
- Create the XListbox as normal on one of your pages (I'll assume it has created an element reference called `m_pElemListbox`).
- On the additional page (eg. `E_PG_EXTRA2`) that you would like to create an element alias, add the following:
```c++
pElemRef = gslc_ElemAdd(&m_gui,E_PG_EXTRA2,gslc_GetElemFromRef(&m_gui,m_pElemListbox),GSLC_ELEMREF_DEFAULT);
```
- Then we would need to bump up the element count on the page to account for the extra element:
```c++
#define MAX_ELEM_PG_EXTRA2      2+1                 // Increase the element count to accommodate the alias
```

The above will save you the cost of the extended element (eg. XListbox) and enable you to reuse all of its configuration, current selected state, etc. but still bear the memory cost of one element (~80B). If you want to save on the cost of the extra element as well, then I have an idea for a modification to this that may enable that as well.

thx

Avpi18, 2020-05-12:
Hello @Pconti31 

Thanks for your great feedback. I have also thought a little bit and then I got the idea with the pop up. 

i have already chosen the flash api= true. without this function my project would not be realizable. best thanks here also to Calvin Hass. 

Since ImpulseAdventure already suggested a slightly different method, I tried this one first and will continue with a pop up if the problem cannot be solved.

Hello @ImpulseAdventure 
Thanks also to you for your help. The proposal with:

`pElemRef = gslc_ElemAdd(&m_gui,E_PG_EXTRA2,gslc_GetElemFromRef(&m_gui,m_pElemListbox),GSLC_ELEMREF_DEFAULT);`

has worked. I also used the ListSlider as Ref.

`pElemRef = gslc_ElemAdd(&m_gui,E_PG_EXTRA2,gslc_GetElemFromRef(&m_gui,m_pListSlider),GSLC_ELEMREF_DEFAULT);`

The problem I have to solve now is that the listslider is not compatible with the function works:

`nVal = gslc_ElemXSliderGetPos(pGui,m_pListSlider);`
`gslc_ElemXListboxSetScrollPos(pGui,m_pElemListbox, nVal);`

, but only if I make a page break. More precisely explained: On Page_1 is my original ListSlider and on Page_2 is my copied/referenced element. If I now use the slider on Page_2, nothing happens. If I then leave Page_2 and go back to Page_2, it has adjusted the position. 

Do you have any idea how I can update the position of the ListSlider on Page_1 and Page_2 in real time? (With the original element, i.e. on Page_1, it updates when I use the ListSlider)

ImpulseAdventure, 2020-05-13:
Hi @Avpi18 — good to hear that progress has been made!

I haven’t had an opportunity to test it yet but my guess is that we’re just missing the linkage between the scrollbar position and the pair of listboxes (original and alias).

For example, let’s say you have saved the element reference for the aliased Listbox as m_pElemListbox2. Can you try revising your above line to the following:

```c++
nVal = gslc_ElemXSliderGetPos(pGui,m_pListSlider);
gslc_ElemXListboxSetScrollPos(pGui,m_pElemListbox, nVal);
gslc_ElemXListboxSetScrollPos(pGui,m_pElemListbox2, nVal);

```

 If you can attach your example sketch, it should make it easier to identify the necessary changes.

Thanks

Avpi18, 2020-05-14:
Hello @ImpulseAdventure 

Thanks for your help. It's very kind and I appreciate your help. 

Actually I wanted to avoid that I have to create a new name, like `m_pElemListbox2`. Apparently this does not work. But it doesn't matter. I created a name anyway, set the links and everything worked. 
Unfortunately I saw your message too late, when I already worked a night shift ;-)

But your detailed answer helped me anyway. I even think that you could include this example in your documentation, because there are surely more cases where you have to create an alias. 

The topic is closed for me now. Thanks to @Pconti31 and @ImpulseAdventure for your support.

ImpulseAdventure, 2020-05-14:
Hi @Avpi18 --

I had added the additional global just to make accessing the element alias elsewhere a little easier, but you're right -- it would probably be best to limit any extra edits.

It is easy to avoid this step... I have included an updated change list for you below.

Good suggestion on putting it in the wiki. I will create a section for tips and integrate this since I hadn't thought about "element aliases" until you asked the question :)

## Detailed list of Changes
The changes I needed to make to the sketch_GSLC.h:
- Increase MAX_ELEM_PG3 by 2 to account for Listbox and Slider (scrollbar)
- Add code to create the two aliases on E_PG3:
```c++
  pElemRef = gslc_ElemAdd(&m_gui,E_PG3,gslc_GetElemFromRef(&m_gui,m_pElemListbox1),GSLC_ELEMREF_DEFAULT);
  pElemRef = gslc_ElemAdd(&m_gui,E_PG3,gslc_GetElemFromRef(&m_gui,m_pListSlider1),GSLC_ELEMREF_DEFAULT);
```

The changes I needed to make to the sketch.ino:
- Link the scrollbar to the listbox (original and alias):
```c++
      gslc_ElemXListboxSetScrollPos(pGui, m_pElemListbox1, nVal);      
      gslc_ElemXListboxSetScrollPos(pGui, gslc_PageFindElemById(pGui,E_PG2,E_ELEM_LISTBOX1), nVal); 
````

Hope that helps!

closed 2020-05-16 by ImpulseAdventure

Avpi18, 2020-05-19:
Thank you for your answer. This helps a lot. I will implement this code and hope that it will work. the Project is progressing, thanks to you.

Avpi18, 2020-09-14:
hello @ImpulseAdventure 
As I have noticed, GUISlice is making good progress. Great work!
I have a question regarding an alias to a button. 

This is what i tried: (I also adjusted the number of Elements on the Page. Normally, with a button it doesn't generate a pElemRef, so i tried but failed as you will see)

` pElemRef =  gslc_ElemCreateBtnTxt_P(&m_gui,E_ELEM_BTN63,NEW2_PG,10,190,90,40,
    "Speichern",&m_asFont[E_FREE_SANS9],
    GSLC_COL_BLACK,((gslc_tsColor){110,110,110}),GSLC_COL_GRAY_LT1,((gslc_tsColor){110,110,110}),
    ((gslc_tsColor){110,110,110}),GSLC_ALIGN_MID_MID,true,true,&CbBtnCommon,NULL);`

`m_pElemSave0 = pElemRef`

And then I added:

`extern gslc_tsElemRef* m_pElemSave0;`
`extern gslc_tsElemRef* m_pElemSave1;   // alias from m_pElemSpeichern0`

`m_pElemSave1 = gslc_ElemAdd(&m_gui,NEW3,gslc_GetElemFromRef(&m_gui,m_pElemSave0),GSLC_ELEMREF_DEFAULT);`

How can I create an alias to a button? I'm afraid I'm not getting anywhere with the last description from this topic. 

I hope for your support and thank you already now. 
Thanks a lot 
Avpi18