ImpulseAdventure/GUIslice #227: Popup pages and advanced controls

Pconti31, 2020-05-17:
Recently I have been somewhat annoyed by the full page refresh that occurs when we use a popup page.  I think that we could improve this if we added properties of frame size to the popup pages.
Normally, it would be the full page but if set to a smaller size with x and y coordinates you could use your refresh code to handle just the frame area.   The builder could handle this for users.

The second thing is assigning popup page to keypad seems overkill and awkward. It's not a problem for the builder but still seems odd.  Now I accept blame for this having come up with popup page and the keypad but given the rule about not using malloc that was the best I could think of at the time.

However, as I think about implementing something like a ComboBox it seems to me that a more general solution might be possible.

I'm thinking of a new component that i'll refer to as a popup block.   Think box with mini page like qualities.   It would only be for compound elements hand crafted like we do now but with a new data structure that is a smaller page struct with xdata for storing all the components in it.  Much as we do now for compound elements.   Users or the Builder would alloc this new struct instead of say, gslc_tsXKeyPad_Num. Everything else stays the same.
The page struct would add a pointer to the current, if any, mini page and the refresh rules would be:
- popup block
- current popup page (so we could place popup blocks on popup page?)
- active page
- base page

This I think this would make creating a new control like, ComboBox much easier since once a user touched a combobox we could paint the popup block with its options.

This would also cleanup the keypad creation.   We could of course keep the current keypad for backward compat and simply add a frame to popup page to solve refreshes or create a new version using popup blocks so we don't need to create a page and assign it to the keypad.

rotus8, 2020-05-17:
Sounds like a good idea Paul. One thing I noticed
        is when a popup is active, the underlying page is not refreshed.
        I would be nice if it was updated; for example my application
        has a constantly running clock at the top of a page and when a
        popup is active, the clock is stopped, I would like to see it
        running. Is this possible?
      
    Ted
    
    On 5/17/2020 10:28 AM, Paul Conti
      wrote:
    
    
      
      Recently I have been somewhat annoyed by the full page refresh
        that occurs when we use a popup page. I think that we could
        improve this if we added properties of frame size to the popup
        pages.
        Normally, it would be the full page but if set to a smaller size
        with x and y coordinates you could use your refresh code to
        handle just the frame area. The builder could handle this for
        users.
      The second thing is assigning popup page to keypad seems
        overkill and awkward. It's not a problem for the builder but
        still seems odd. Now I accept blame for this having come up with
        popup page and the keypad but given the rule about not using
        malloc that was the best I could think of at the time.
      However, as I think about implementing something like a
        ComboBox it seems to me that a more general solution might be
        possible.
      I'm thinking of a new component that i'll refer to as a popup
        block. Think box with mini page like qualities. It would only be
        for compound elements hand crafted like we do now but with a new
        data structure that is a smaller page struct with xdata for
        storing all the components in it. Much as we do now for compound
        elements. Users or the Builder would alloc this new struct
        instead of say, gslc_tsXKeyPad_Num. Everything else stays the
        same.
        The page struct would add a pointer to the current, if any, mini
        page and the refresh rules would be:
      
        popup block
        current popup page (so we could place popup blocks on popup
          page?)
        active page
        base page
      
      This I think this would make creating a new control like,
        ComboBox much easier since once a user touched a combobox we
        could paint the popup block with its options.
      This would also cleanup the keypad creation. We could of course
        keep the current keypad for backward compat and simply add a
        frame to popup page to solve refreshes or create a new version
        using popup blocks so we don't need to create a page and assign
        it to the keypad.
      —
        thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ImpulseAdventure/GUIslice/issues/227",
"url": "https://github.com/ImpulseAdventure/GUIslice/issues/227",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

ImpulseAdventure, 2020-05-17:
@Pconti31 :  thanks for the idea. I'll take a look at what is involved and followup.

@rotus8 : regarding the underlying pages:
- By default, when `PopupShow()` is called, the redraws for underlying pages (base and current) are disabled. This was done to limit any "bleed-through" of obscured elements whilst avoiding the need to complicate the invalidation logic (determining what was visible, etc.). The complexity arises when elements on underlying pages are partially obscured by the overlapping popup region.
- You should be able to enable background redraws during a popup by adding a call to `gslc_SetStateState()`. For example: (for a modal popup)
```c++
gslc_PopupShow(&m_gui, E_PG_POPUP, true);
gslc_SetStackState(pGui, GSLC_STACK_CUR, false, true); // Enable redraw
gslc_SetStackState(pGui, GSLC_STACK_BASE, false, true) // Enable redraw
```

As your clock is not obscured by the popup, I believe the above should work.
Please let me know if it helps?

Pconti31, 2020-05-17:
I posted here to see if anyone else had any feedback or ideas.
If after you have thought more about it and it looks reasonable to you I'm happy to do a prototype.

Pconti31, 2020-05-17:
I put together a quick test using a counter instead of a clock.
Turns out that if you have your clock or counter on the base page you only need the
```
gslc_SetStackState(pGui, GSLC_STACK_BASE, false, true) // Enable redraw
```
attached is my test project
[modeless_dialog.zip](https://github.com/ImpulseAdventure/GUIslice/files/4640909/modeless_dialog.zip)

Paul--

Pconti31, 2020-05-23:
Going back to when a popup is active, the underlying page is not refreshed.
I've given this more thought and it seems to me that maybe a simplier solution would be if a user requests a modeless dialog with `gslc_PopupShow(&m_gui, E_PG_POPUP, false);` the api could
in addition to setting other layers clickable could also set
```
gslc_SetStackState(pGui, GSLC_STACK_CUR, false, true);
gslc_SetStackState(pGui, GSLC_STACK_BASE, false, true) 
```
IMHO, This would be more in the spirit of a modeless dialog.
Paul--

ImpulseAdventure, 2020-05-23:
Paul -- interestingly enough, when I was reviewing the `PopupShow()` defaults for the question from @rotus8, I was assuming that I would have set the background redraw to true for modeless dialog mode, but noticed instead that redraws were also disabled (hence the workaround above).

That said, I agree that the background redraws should be active on modeless dialogs. The downside here is that if a popup obscures other controls that continue to update in the background, there will be unnecessary redraw -- I *assume* this is probably a less common scenario.

BTW, for modeless dialogs I think you meant:
```c++
gslc_SetStackState(pGui, GSLC_STACK_CUR, true, true);
gslc_SetStackState(pGui, GSLC_STACK_BASE, true, true) 
```

I have now made this change and pushed it to the repo.

Thanks for reminding me of this detail!

Pconti31, 2020-05-26:
In doing a code review of GUIslice I found that the Popup page refresh problem was in fact already solved.   You already had the frame defined, not statically as I proposed but dynamically constructed as elements are added to the page. So much the better.  The driver setting
```
 pGui->bRedrawPartialEn = true;
```
makes the refresh work fine as long as the driver supports it.  The larger than I thought refreshes are caused by my using the examples as tests for popups, since most have progress bars and counters changing.  If you have a popup on a page where some elements are changing each cycle through gslc_Update() you of course need to update the larger rectangle they are inside.  Re-arrange these elements with the popup and the refresh looks quite smooth.

So now my issue is how to implement a combobox or any control that has a drop down box?
and enable touch over the new region?  More code review unless you have some helpful suggestions.

Pconti31, 2020-05-26:
So my first thoughts to implement a combobox would be a creating a compound element that contains a standard text button that has text storage assigned for the selection, and the second element a single column listbox that starts out hidden and with click disabled until someone touches the first button.  Only than we unhide the listbox and enable click on it.  User make a selection we pass it back to the text button for display and hide the listbox again and disable click.

One problem, of course, is refresh under and around the listbox without causing full page redraws.
Like we do with popups.

Pconti31, 2020-05-30:
Closed due to lack of activity
Paul--

closed 2020-05-30 by Pconti31