ImpulseAdventure/GUIslice #231: Box element dont execute CbBtnCommon

hellzakk, 2020-05-19:
from test01_GSLC.h
```
...
  bool CbBtnCommon(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY);
...
  // Create E_ELEM_BOX1 box
  pElemRef = gslc_ElemCreateBox(&m_gui,E_ELEM_BOX1,E_PG_MAIN,(gslc_tsRect){10,25,300,150});
  gslc_ElemSetTouchFunc(&m_gui, pElemRef, &CbBtnCommon);
  gslc_ElemSetCol(&m_gui,pElemRef,GSLC_COL_GRAY,GSLC_COL_BLACK,GSLC_COL_BLUE);
  E_ELEM_BOX_1 = pElemRef;
...

```

from test01.ino
```
bool CbBtnCommon(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  Serial.println("pushed");
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui,pElemRef);

  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BOX1:
        Serial.println("box pushed");
        break;

//<Button Enums !End!>
      default:
        break;
    }
  }
  return true;
}
```

Never print "box pushed" nor "pushed" and no "selected color" of the box element.
A Text Button work as expected instead.

ImpulseAdventure, 2020-05-20:
Hi!

By default the box elements have touch handling disabled. Could you try adding the following and b let me know if that makes a difference?

gslc_ElemSetClickEn(&m_gui, pElemRef, true);

Perhaps it may make sense for the gslc_ElemSetTouchFunc() API to also cause the ClickEn to also be set.

hellzakk, 2020-05-20:
Hi ImpulseAdventure, firstly i would like to thank you for your work.

This is the updated test01_GSLC.h code
```
// Create E_ELEM_BOX1 box
  pElemRef = gslc_ElemCreateBox(&m_gui,E_ELEM_BOX1,E_PG_MAIN,(gslc_tsRect){10,25,300,150});
  gslc_ElemSetTouchFunc(&m_gui, pElemRef, &CbBtnCommon);
  gslc_ElemSetClickEn(&m_gui, pElemRef, true); // added
  gslc_ElemSetCol(&m_gui,pElemRef,GSLC_COL_RED,GSLC_COL_BLACK,GSLC_COL_BLACK);
  m_pElemT1Box = pElemRef;
```

and i can confirm it works.

ImpulseAdventure, 2020-05-20:
Great.. thank you for the confirmation.

closed 2020-05-20 by ImpulseAdventure