ImpulseAdventure/GUIslice #233: Text string element

giordy72, 2020-05-22:
Hi,
I can't make text string works well.
I want to update a value on the screen once per second, so I wrote this inside my loop:
```
   _gslc_ElemSetTxtStr(&m_gui, m_pElemOutTxtStatus,acTxt);    
   snprintf(acTxt,MAX_STR,(char*)"RPM = %4u\n",speedRead);_
```
This works...but the string didn't update well mixing all the new updated strings together.

[image]

Suggestions?

ImpulseAdventure, 2020-05-22:
Hi @giordy72 -- it looks like you are doing everything correctly with the updates.

What the output above usually means is that the size of the text element wasn't large enough to contain all of the text (including the RPM value updates). To keep things efficient, the element dimensions dictate where redraws will be done.

Look for the dimensions in the `gslc_ElemCreateTxt()` call. In the example below, we would increase the 44 which defines the width: 
```c++
pElemRef = gslc_ElemCreateTxt(&m_gui,E_ELEM_TXT_COUNT,E_PAGE_MAIN,(gslc_tsRect){80,60,44,12},(char*)m_sDisplayText2,8,E_FONT_TXT5);
```

If you are using the Builder, then the text element size is normally estimated by the  "External Storage Size" setting and the current font. Please try increasing the width to see if that helps.

Ideally, if you could ZIP up your project folder, with any builder and generated sketch file and attach to this issue, it may help me see how you are creating the element.

thanks

ImpulseAdventure, 2020-05-31:
Closing based on the assumption that the text element needed to be larger to accommodate the text updates. @giordy72 if at some point you are able to confirm, that would be great. If the problem remains, I will reopen the issue for you. thanks.

closed 2020-05-31 by ImpulseAdventure

giordy72, 2020-06-04:
Hi all,
gslc_ElemCreateTxt with the right settings (310pxl wide now!) solves the issue.
Thank's a lot