ImpulseAdventure/GUIslice #235: DUE + Elegoo 2.8 TFT - Graphic & Touch don't work together

hellzakk, 2020-05-28:
Describe the bug :
Hi all, i previously configured guislice to work with this lcd and a MEGA 2560.
Now i've switched to an Arduino DUE clone but it isn't with the graphic and the touch function at the same time.

I have changed the XM XP YM YP to the correct value and now i can see and run with touch the TouchScreen_Calibr_native (MCUFRIEND_kvb).

In guislice if i set ard-shld-mcufriend.h i can correctly see the graphics (but with no touch support as expected).
If i set elegoo_28_res.h or ard-shld-mcufriend_4wire.h i can see from the serial monitor that the touch event is received but the screen remain white  (powered on) with no graphics.

Any hint?

Device hardware :
- MCU: Arduino DUE clone
- Display: Elegoo 2.8 TFT
- GUIslice config selected: elegoo_28_res.h - ard-shld-mcufriend.h - ard-shld-mcufriend_4wire.h

ImpulseAdventure, 2020-05-29:
Hi -- thanks for providing the detailed debug info!

What you are observing is very likely to be an incompatibility between your pin-sharing display (driven by mcufriend_kbv) and the Adafruit_Touchscreen driver, and not an issue with the GUI itself. Unfortunately, the Adafruit_Touchscreen driver was not designed for TFTs that pin-share between the display (8-bit parallel interface) and the resistive touch overlay.

To help confirm the behavior, could you try running the attached sketch (it uses **mcufriend_kbv** and **Adafruit_Touchscreen**, but does not incorporate GUIslice). Please try the following steps:

[graphictest_kbv-touch20.zip](https://github.com/ImpulseAdventure/GUIslice/files/4700155/graphictest_kbv-touch20.zip)

1) Modify the sketch to suit your pinout (if you could report what your `XP/YP/XM/YM` settings are, that would be helpful too).
2) Run the sketch with `CAL_TEST` uncommented and `CAL_FIX` commented. Does the graphics test proceed correctly?
3) If the above test (2) fails, perhaps with a white screen, please run the sketch again with `CAL_TEST` uncommented but `CAL_FIX` uncommented.

Note that GUIslice already incorporates a few workarounds for Adafruit_Touchscreen in various "Arduino" architectures (see `FIX_4WIRE_PIN_STATE`) in an attempt to avoid this incompatibility, but the same workaround isn't currently written for the Arduino Due.

thanks!

hellzakk, 2020-05-29:
My settings
```
#define YP     A3
#define XM     A2
#define YM     9
#define XP     8 
```

with CAL_TEST uncommented : white screen
with CAL_FIX uncommented : ok

ImpulseAdventure, 2020-05-29:
Thanks for trying out the test. Your observation confirms that the issue is outside of GUIslice (ie. it is caused by a pin-sharing conflict between the Adafruit touch library and the display driver).

Could you confirm what version of GUIsice you are running? It should be reported in the serial monitor window and also in the `/src/GUIslice_version.h` file.

If you are using the version from the Arduino IDE Library Manager (ie. `0.14.0`), please download and install the [latest from GitHub](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source) (eg. `0.14.0.19`) and let me know if that makes any difference for your run of example ex02 (FYI, some additional workarounds for Due were implemented in `0.14.0.3`).

thx

hellzakk, 2020-05-31:
Thanks to you for your work!

I've downloaded from github the (latest) following libraries:

- Guislice
- Adafruit GFX Library
- Adafruit ILI9341
- Adafruit TouchScreen
- MCUFRIEND_kbv

using **ard-shld-elegoo_28_res.h** and **DRV_TOUCH_ADA_SIMPLE** i confirm its working

ImpulseAdventure, 2020-05-31:
Excellent! It sounds like you might have been running an older version of GUIslice that predated my implementation of the pin-sharing workarounds for Due.

Thank you for the confirmation!

closed 2020-05-31 by ImpulseAdventure