ImpulseAdventure/GUIslice #236: Cannot solve 4-wire touchscreen "Seeedstudio 2,8" tft touch shield v2.1" on ESP32

AdmDeckerGH, 2020-06-01:
I cannot sove 4-wire Touchscreen for the shield "Seeedstudio 2,8" tft touch Shield v2.1" on ESP32.
I previously installed TFT_eSPI of Bodmer and I got the Display running.
The next target is to get the touchscreen working.

In User_Setup.h of TFT_eSPI I comment touch_cs out:
//#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen
And I activated support_transactions:
#define SUPPORT_TRANSACTIONS

In  "GUIslice_Config.h" I selected:
 #include "../configs/esp-tftespi-default-simple.h"

In "esp-tftespi-default-simple.h" I modified the following values:
#define ADATOUCH_PIN_YP     13
#define ADATOUCH_PIN_XM     12
#define ADATOUCH_PIN_YM     36
#define ADATOUCH_PIN_XP     34

I assumed the 4-wire pinout is:
A0 = YM
A1 = XM
A2 = YP
A3 = XP

What did I wrong ?

ImpulseAdventure, 2020-06-02:
Hi @AdmDeckerGH --

The first step after getting your display going is to ensure that we identify a suitable touch library for your TFT. According to the [seeedstudio wiki](https://wiki.seeedstudio.com/2.8inch_TFT_Touch_Shield_v2.0/), it does look like it uses a 4-wire resistive overlay, matching what you are selecting.

Before testing out with GUIslice, could you confirm that the touchscreen works correctly with the **Adafruit_Touchscreen** library example: `touchscreendemo.ino`? You will need to modify the XP/YP/XM/YM accordingly.

thanks!

AdmDeckerGH, 2020-06-02:
Hi **@ImpulseAdventure** 

I tried to run `touchscreendemo.ino`. It compiled and uploaded well but this didn't run (no blue led after booting). Also there was no serial output, tho...

I solved the 4-wire assignment. The Touchscreen works now perfectly on ESP32.

First of all, the pinout of the 4 wires of "_Seeedstudio tft touch v2.1_" shield are indeed as assumed:
A0 = YM
A1 = XM
A2 = YP
A3 = XP

To run the touchscreen I defined the Pins (in `esp-tftespi-default-simple.h`) for ESP32 as follows:
` #define ADATOUCH_PIN_YP     34` 
` #define ADATOUCH_PIN_XM     39`
` #define ADATOUCH_PIN_YM     36`
` #define ADATOUCH_PIN_XP     35`

Side note:
I don't know for sure if the following settings in `User_Setup.h` of **TFT_eSPI** have any effect, but I left TOUCH_CS as well as SUPPORT_TRANSACTIONS  commented out.
`//#define TOUCH_CS 21 // Chip select pin (T_CS) of touch screen`
...
`//#define SUPPORT_TRANSACTIONS`

Thanks.

ImpulseAdventure, 2020-06-03:
Hi @AdmDeckerGH -- thank you for testing out Adafruit's **Adafruit_Touchscreen** example sketch and indicating the pinout.

Just so that I understand your observations, can you confirm:
1) When you said "*I solved the 4-wire assignment. The Touchscreen works now perfectly on ESP32.*", I presume you mean the **Adafruit_Touchscreen** example (`touchscreendemo.ino`) now works for you. Did you also mean that GUIslice (with updated config) also works?
2) If GUIslice doesn't work, can you confirm what happens when you run the `diag_ard_touch_calib` sketch, ie:
2a) The display turns white after enabling touch (ie. can't run sketch)
2b) The display is OK, but touch doesn't appear in the correct locations
2c) The display is OK, but there is no reaction after a touch

thanks!

AdmDeckerGH, 2020-06-04:
Hi @ImpulseAdventure -- I just cleaned up my develop environment to garantee definitive results.

>     1. When you said "_I solved the 4-wire assignment. The Touchscreen works now perfectly on ESP32._", I presume you mean the **Adafruit_Touchscreen** example (`touchscreendemo.ino`) now works for you. Did you also mean that GUIslice (with updated config) also works?

Yes, the 4-wired touchscreen now works perfectly on ESP32 with **GUIslice** example (`diag_ard_touch_test.ino`).  
Unfortunately it does **not** work with **Adafruit_Touchscreen** example (`touchscreendemo.ino`).
Also it does **not** work with **TFT_eSPI**, because it does not support 4-wired touchscreens.

> 2. If GUIslice doesn't work, can you confirm what happens when you run the `diag_ard_touch_calib` sketch, ie:
>        2a) The display turns white after enabling touch (ie. can't run sketch) [...]

GUIslice is working. The sketch `diag_ard_touch_calib` can be executed successfuly.

> As for **TFT_eSPI**'s `User_Setup.h`: commenting out `TOUCH_CS` is likely important as it ensures that the TFT_eSPI library doesn't attempt to issue SPI transactions to a non-existent XPT2046 touch driver (the built-in touch driver assumed by TFT_eSPI).

Thanks for this information.

Thanks.

ImpulseAdventure, 2020-06-04:
Good to hear that GUIslice is working with your ESP32 and the calibration example.

Before closing this issue, can you also confirm that example `ex02_ard_btn_txt` works with your display (after updating your GUIslice config with the calibration details from `diag_ard_touch_calib`)?

AdmDeckerGH, 2020-06-05:
The details from `diag_ard_touch_calib` are:
`Status: Calibrate Done`
`ADATOUCH_X_MIN: 949`
`ADATOUCH_Y_MIN: 978`
`ADATOUCH_X_MAX: 129`
`ADATOUCH_Y_MAX: 129`
`Rotation detected OK`
The red dots matching in all 4 cases of screen rotation the touches.
I'm slightly confused by the min / max values. In my understanding the less values should assigned to the min ?

The example `ex02_ard_btn_txt` shows a centered button **Quit** in a landscape screen layout.
This blue button is surrounded by a black filled rectangle with a white frame.
Touching this button changes the background color to light blue.
Releasing this button changes the background color to the previous color.

ImpulseAdventure, 2020-06-05:
Thanks @AdmDeckerGH -- your results look good.

Regarding the min/max, the naming is based on the conventions that Adafruit originally used when defining calibration for their resistive displays (eg. STMPE610). But in reality the convention (MIN/MAX) is a little confusing since the MIN/MAX values might be swapped depending on the orientation of the display or touch overlay (eg. XMIN could be larger than XMAX). David Prentice (in his **mcufriend_kbv** library) has adopted a nice alternate convention by naming them TOP, LEFT, BOTTOM, RIGHT, etc.

In conclusion, it is perfectly valid to have X_MIN > X_MAX, or Y_MIN > Y_MAX, etc. -- it just depends on touch overlay wiring and orientation.

I'll close this now as everything appears to be working as expected. Thanks!

closed 2020-06-05 by ImpulseAdventure