ImpulseAdventure/GUIslice #237: Error while trying to use GUIslice with IL9486

jmmacip, 2020-06-01:
**Describe the bug**
Every time I try to compile any of the examples I get this error:

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp: In function 'bool gslc_DrvGetTxtSize(gslc_tsGui*, gslc_tsFont*, const char*, gslc_teTxtFlags, int16_t*, int16_t*, uint16_t*, uint16_t*)':

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1011:21: error: 'sFONT' was not declared in this scope

     m_disp.setFont((sFONT*)(pFont->pvFont));

                     ^~~~~

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1011:27: error: expected primary-expression before ')' token

     m_disp.setFont((sFONT*)(pFont->pvFont));

                           ^

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp: In function 'bool gslc_DrvDrawTxt(gslc_tsGui*, int16_t, int16_t, gslc_tsFont*, const char*, gslc_teTxtFlags, gslc_tsColor, gslc_tsColor)':

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1104:19: error: 'sFONT' was not declared in this scope

   m_disp.setFont((sFONT*)(pFont->pvFont));

                   ^~~~~

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:1104:25: error: expected primary-expression before ')' token

   m_disp.setFont((sFONT*)(pFont->pvFont));

                         ^

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

[config_files.zip](https://github.com/ImpulseAdventure/GUIslice-Builder/files/4710490/config_files.zip)

**To Reproduce**
TRy to compile any of the examples using the attached config files

**Expected behavior**
A clear and concise description of what you expected to happen.
Compiler runs clear of erros

**Crash log / Project files**
If you experienced a crash, please attach the following to help us diagnose the cause:
- Attach a ZIP file containing the project folder including the `*.prj` and `*.ino` or `*.c` and `*.ino.bak`.

**Version**
- Builder version: 0.14b004
- OS:  Windows 10

**Additional context**
Add any other context about the problem here.

ImpulseAdventure, 2020-06-01:
Hi @jmmacip --
It appears that you are using a more recent version of the Waveshare_ILI9486 library, but an older version of GUIslice. Please update GUIslice to the [latest on GitHub](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source) and let me know if that works better for you. thx

jmmacip, 2020-06-01:
HI,
Unluckily same result, I was running 0.14 already but tried the update.

ImpulseAdventure, 2020-06-02:
Hi @jmmacip -- that particular code was fixed in version v0.14.0.2 (https://github.com/ImpulseAdventure/GUIslice/commit/12810551ca770c05bf3ee507f591a3f79dc49559).

Please note that I was recommending that you update to the *latest version on GitHub* (see instructions [here](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source)). The latest version on GitHub (eg 0.14.0.19) is newer and has some additional fixes that are not yet available via the Arduino IDE release (0.14.0). Soon, a new release will be made into the Arduino IDE (ie. 0.14.1) that will contain the roll-up of everything in the current development branch.

If you can re-check that you are indeed using the latest from GitHub, that would be great.

thanks!

jmmacip, 2020-06-02:
Hi,
This is exactly what I did but I noticed that the library.propierties in the download still shows "version": "0.14.0"  but if I check GUIslice_version.h I have #define GUISLICE_VER "0.14.0.19" meaning that I'm running the latest version. I'm going to scap my config files and redo them from scratch.

ImpulseAdventure, 2020-06-02:
Thanks for checking on the version -- yes, you are correct, the `GUISLICE_VER` in `GUIslice_version.h` indicates that you indeed have installed the latest from GitHub (the `library.properties` is only updated when IDE releases are made).

What this suggests to me is that there might be multiple versions of the library on your computer: perhaps an old version installed via the Arduino IDE Library Manager as well as the newer version downloaded from GitHub.

To rule this out, let's check the following:
In the Arduino IDE, select File -> Preferences and ensure "Show verbose output during compilation" is enabled (checked).

When you compile your sketch, have a look to see if there are any lines near the bottom that indicate you have multiple libraries present. For example:
```
Multiple libraries were found for "GUIslice.h"
 Used: E:\My Documents\Develop\Electronics\Arduino\libraries\GUIslice
 Not used: E:\My Documents\Develop\Electronics\Arduino\libraries\GUIslice-master
```
In the above, the "GUIslice" folder was installed by the Arduino IDE library manager, whereas the "GUIslice-master" was the folder downloaded direct from GitHub. If we don't remove the old version, then the IDE could potentially be using the older version (as above).

If you could look in your Arduino /libraries folder to see if there might be any older versions present as well, that would be useful.

thanks

jmmacip, 2020-06-02:
Hi Calvin.
Compiled with verbose and I didn't got any duplicate library messages.
The compilation errors make reference to som error related to fonts, I recall seeing a coment in your code about IL9486 fonts could be this related?
C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx (2).cpp: In function 'bool gslc_DrvGetTxtSize(gslc_tsGui*, gslc_tsFont*, const char*, gslc_teTxtFlags, int16_t*, int16_t*, uint16_t*, uint16_t*)':

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx (2).cpp:1011:21: error: 'sFONT' was not declared in this scope

     m_disp.setFont((sFONT*)(pFont->pvFont));

                     ^~~~~

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx (2).cpp:1011:27: error: expected primary-expression before ')' token

     m_disp.setFont((sFONT*)(pFont->pvFont));

                           ^

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx (2).cpp: In function 'bool gslc_DrvDrawTxt(gslc_tsGui*, int16_t, int16_t, gslc_tsFont*, const char*, gslc_teTxtFlags, gslc_tsColor, gslc_tsColor)':

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx (2).cpp:1104:19: error: 'sFONT' was not declared in this scope

   m_disp.setFont((sFONT*)(pFont->pvFont));

                   ^~~~~

C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx (2).cpp:1104:25: error: expected primary-expression before ')' token

   m_disp.setFont((sFONT*)(pFont->pvFont));

                         ^

Using library GUIslice at version 0.14.0 in folder: C:\Users\admin\Documents\Arduino\libraries\GUIslice 
Using library Adafruit_GFX_Library at version 1.8.4 in folder: C:\Users\admin\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Waveshare_ILI9486 at version 2.0.0 in folder: C:\Users\admin\Documents\Arduino\libraries\Waveshare_ILI9486 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library XPT2046_Touchscreen at version 1.3 in folder: C:\Users\admin\Documents\Arduino\libraries\XPT2046_Touchscreen 
Using library Adafruit_BusIO at version 1.3.2 in folder: C:\Users\admin\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

ImpulseAdventure, 2020-06-03:
Thanks for including the compiler log.
That is strange -- something doesn't seem to line up...

On the basis of your compile log, it shows that you are encountering an error on GUIslice_drv_adagfx.cpp line 1011 (re: `sFont`). That code has not existed from version **0.14.0.2** onward. Yet, you have indicated that you are running **0.14.0.19**.

1) If you open up `C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_version.h`, what `GUISLICE_VER` do you see? I understood from your earlier report that you are seeing **0.14.0.19**.
2) If you open up `C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp` and scroll to line 1011, do you see the following?
`m_disp.setFont((sFONT*)(pFont->pvFont));`

If the answers to the above (1) and (2) are **0.14.0.19** and **Yes**, then it sounds like your download / unzip was corrupt. Please delete the `Arduino\libraries\GUIslice` folder and unzip/install it again from GitHub.

Hopefully that will help narrow it down :)

Pconti31, 2020-06-03:
I took a quick look at the current repository and sFont is nowhere to be found.
line 1011 is simply a blank line.

Looks like sFont was removed during a later checkin.
@@ -1006,10 +1012,7 @@ bool gslc_DrvGetTxtSize(gslc_tsGui* pGui,gslc_tsFont* pFont,const char* pStr,gsl
```
1008 1014  // Support library-specific font exceptions
1009 -  #if defined(DRV_DISP_WAVESHARE_ILI9486)
1010 -    // Waveshare_ILI9486 uses a different font structure
1011 -    m_disp.setFont((sFONT*)(pFont->pvFont));
1012 -  #elif defined(DRV_DISP_ADAGFX_RA8876_GV)
1015 +  #if defined(DRV_DISP_ADAGFX_RA8876_GV)
```
Paul--

jmmacip, 2020-06-04:
HI,
As Pconti31 pointed out there is nothing in line 1011. Looks like the code has been removed from the current release.

ImpulseAdventure, 2020-06-05:
> there is nothing in line 1011. Looks like the code has been removed from the current release

@jmmacip -- as mentioned in (https://github.com/ImpulseAdventure/GUIslice/issues/237#issuecomment-637915156), I suspect your download was corrupted. Please try re-downloading the latest from GitHub, unzip and let us know if the error goes away.

thanks!

jmmacip, 2020-06-05:
> 
> 
> > there is nothing in line 1011. Looks like the code has been removed from the current release
> 
> @jmmacip -- as mentioned in ([#237 (comment)](https://github.com/ImpulseAdventure/GUIslice/issues/237#issuecomment-637915156)), I suspect your download was corrupted. Please try re-downloading the latest from GitHub, unzip and let us know if the error goes away.
> 
> thanks!

Already tried downloading twice and line 1011 is empty, it is also empty in your current release code:
https://github.com/ImpulseAdventure/GUIslice/blob/5c71a9a616ddcce8bff2f2edaee9a4473868278f/src/GUIslice_drv_adagfx.cpp#L1011

  nTxtScale = pFont->nSize;
  m_disp.setTextSize(nTxtScale);

  // Fetch the font sizing
  m_disp.getTextBounds((char*)pStr,0,0,pnTxtX,pnTxtY,pnTxtSzW,pnTxtSzH);

Pconti31, 2020-06-05:
@jmmacip
@ImpulseAdventure 
I rolled back some of the changes to what I think should work for your device.
Let Calvin know if it works for you. If so it should be a simple cut and paste to get it back in the repository.  If not, then more serious issues are involved.
Paul--

[GUIslice_drv_adagfx.zip](https://github.com/ImpulseAdventure/GUIslice/files/4737730/GUIslice_drv_adagfx.zip)

Pconti31, 2020-06-05:
@jmmacip 
@ImpulseAdventure 
This earlier version has line 1011     m_disp.setFont((sFONT*)(pFont->pvFont));
So maybe that helps you out?
Paul--
[GUIslice_drv_adagfx.zip](https://github.com/ImpulseAdventure/GUIslice/files/4737774/GUIslice_drv_adagfx.zip)

ImpulseAdventure, 2020-06-06:
@Pconti31 -- Yes, that's correct. I mentioned the corruption as it seemed he was still observing compilation errors on line 1011 even though his file (line 1011) was blank.

@jmmacip -- I must admit that I'm a bit confused / surprised by your observations, so I'd like to rule out any misunderstanding and ensure that I have captured your observations correctly:
- C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_version.h indicates "0.14.0.19" (per https://github.com/ImpulseAdventure/GUIslice/issues/237#issuecomment-637337152)
- C:\Users\admin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp line 1011 is blank (per https://github.com/ImpulseAdventure/GUIslice/issues/237#issuecomment-638765185).
- In your first entry (https://github.com/ImpulseAdventure/GUIslice/issues/237#issue-628586875), you attached GUIslice_drv_adagfx.cpp which actually showed sFONT on line 1011. This suggests you must have originally been using the older version & later updated.
- Arduino IDE reported compilation errors (re: sFONT) on line 1011 (per https://github.com/ImpulseAdventure/GUIslice/issues/237#issuecomment-637709264), and is still currently reporting these errors.
- The Arduino IDE compile log indicates the library path is C:\Users\admin\Documents\Arduino\libraries\GUIslice, which matches the path to the sFONT error message above

If the above are all correct, the result doesn't make much sense to me as it suggests the compiler is returning an error on code that doesn't exist! Maybe I'm overlooking something obvious :)

I did note that you had put the following line in your GUIslice_config.h:
`#define GSLC_FEATURE_COMPOUND       1`
- This setting should be in your custom/user config file. GUIslice_config is only used to select between various user configs.
- Please delete this line and set the same flag in Section 6 of your custom config file, ie. configs/ard-shld-waveshare_40_xpt2046.h.

Assuming that doesn't help, the only thing I can imagine is that the Arduino IDE is somehow caching error messages / build...

Out of curiosity, if you do the following, does anything change?
- Ensure you are using Arduino IDE 1.8.12 (or later)
- Arduino IDE: select a different board (eg. Arduino Mega ADK)
- a) Recompile ex02_ard_btn_txt
- Arduino IDE: select your original board (eg. Arduino Mega 2560)
- b) Recompile ex02_ard_btn_txt

Do both a) and b) give you the same sFONT compiler error?

Thanks again

jmmacip, 2020-06-12:
Hi,
My arduino IDE is 1.8.12. That example compiles and runs correctly in both cases with mega ADK and 2560, when I check with Ex26_ard_calc, this was the example I've always tried to compile, it fails with the sFONT error. If I set the #define GSLC_FEATURE_COMPOUND 1 in /ard-shld-waveshare_40_xpt2046.h it is not detected and the compiler throws an error "Config: GSLC_FEATURE_COMPOUND required for this example but not enabled."

Pconti31, 2020-06-13:
@jmmacip  The GSLC_FEATURE_COMPOUNG being ignored sounds like you forgot to edit
GUIslice/src/GUIslice_config.h and comment out whatever config you were using for the mega while uncommenting your:
```
  //#include "../configs/ard-shld-waveshare_40_xpt2046.h"
```
As Calvin has pointed out above the  ex02_ard_btn_txt.ino is a good simple test for font. 
Paul--

ImpulseAdventure, 2020-07-04:
@jmmacip -- I am going to mark this issue as "pending recreate" as much of the evidence has been pointing towards some corruption in your Arduino IDE installation.

The error messages you were reporting (ie. `sFONT`) refer to code that hasn't existed in the repository since mid-April.

In case you have encountered difficulties during attempts to install the library from GitHub source, I plan on making an official release to Arduino IDE shortly (eg. **0.14.1**) that will enable an easy update directly from the Arduino IDE Library Manager.

Please feel free to open a new issue if you manage to install the new version and have any additional questions.

Thanks!

closed 2020-07-04 by ImpulseAdventure