ImpulseAdventure/GUIslice #239: Is it possible to use GUIslice with multiple Displays?

SunboX, 2020-07-05:
### Feature Request or Enhancement

We have two 2.4" Touch Displays added to an ESP32. Is it possible to use GUIslice to handle both isplays at the "same" time?

ImpulseAdventure, 2020-07-05:
Hi André, great question! I have not yet attempted to drive dual displays. That said, it *might* be possible to achieve this with two instances of the GUIslice GUI struct and then alternate calls to gslc_Update(). I must caution that I have not taken a look to see if there may be any gotchas (Eg. static vars, etc.)

Outside of the GUI itself, one would also need to confirm that the 2.4” display driver was capable of being adapted for dual displays. Assuming you are using TFT_eSPI, it looks like one can disable the driver’s TFT CS control and then wrap the TFT APIs with the appropriate CS. In GUIslice, this might mean something like the following in your main loop: (untested!)

```c
digitalWrite(TFT1_CS, LOW);
digitalWrite(TFT2_CS, HIGH);
gslc_Update(&m_gui1);
digitalWrite(TFT1_CS, HIGH);
digitalWrite(TFT2_CS, LOW);
gslc_Update(&m_gui2);
digitalWrite(TFT1_CS, HIGH);
digitalWrite(TFT2_CS, HIGH);
```

Lastly, if using the Builder to generate code, note that it only has awareness of a single GUI instance.

SunboX, 2020-07-06:
Hey, thanks for the quick response! Than I will try it with slightly modifying the auto generated code. We are using TFT_eSPI and Bodmer already modified it a little bit, so that we can easily use multiple displays like you proposed (changing CS): https://github.com/Bodmer/TFT_eSPI/issues/663

I will try, and let you know if I ran into issues. But it will take a while, have to finish another thing first.

But in general, this would be a great feature of GUIslice. As far as I know, Adafruit already supports multiple displays, and I tried it with changing CS like you wrote also with TFT_eSPI and it also works!

ImpulseAdventure, 2020-07-15:
Hi @SunboX -- I'm not sure if you have managed to try out the dual screen, but I am attaching a sample sketch that instantiates dual GUIs. I have not tried to wire it up to a second display, but I would be interested in hearing any results if you do have an opportunity to test it.

The attached sketch shares the elements on both GUIs -- this is not how we would normally implement it, but it should enable one to determine if the basic dual driving works or not. Note that the TFT0_CS and TFT1_CS would need to be altered according to your pinout.
[ex02_ard_btn_txt2.zip](https://github.com/ImpulseAdventure/GUIslice/files/4926232/ex02_ard_btn_txt2.zip)

SunboX, 2020-07-16:
Hey,

I didn't try it yet, because I'm currently trying to get TFT_eSPI working when doing CS by my own. If I do my own CS for the TFT only, it works fine. But as soon as I also try to do CS for one TFT and also the Touch Controller on this TFT, it does not render corectly anymore. So I currently try to figure out why. It's the simplest use case - one TFT with only the TFT_eSPI lib and doing CS for TFT and Touch by my own

SunboX, 2020-07-16:
Ah, and both are connected to the same SPI

ImpulseAdventure, 2020-07-17:
Hi -- can you confirm for me the following:
- When you were trying your experiments with TFT_eSPI and touch using your own CS toggle, was this without GUIslice?
- What touch library are you using? (eg. TFT_eSPI's built-in XPT2046 driver?)
- What TFT are you using? eg. 2.4" MCUFRIEND
- What TFT_eSPI User_Setup are you selecting?

thanks

SunboX, 2020-07-20:
Hi

> * When you were trying your experiments with TFT_eSPI and touch using your own CS toggle, was this without GUIslice?
It was without GUIslice, because I wanted to avoid unnecessary complexity for first try.

> * What touch library are you using? (eg. TFT_eSPI's built-in XPT2046 driver?)
FT_eSPI's built-in XPT2046 driver

> * What TFT are you using? eg. 2.4" MCUFRIEND
Such a "generic" 2.4" touch display:
[image]

I have three of them from different sources and for different prices. Also one from a German seller for "a lot of" money. They behave all the same.

> * What TFT_eSPI User_Setup are you selecting?

```
#define MULTI_TFT_SUPPORT

#define screenCS 23
#define touchCS 17
```
attached you will find my sample code:
[DisplayTest.zip](https://github.com/ImpulseAdventure/GUIslice/files/4946643/DisplayTest.zip)

ImpulseAdventure, 2020-07-26:
Hi @SunboX -- thanks for providing all of the extra details; that's helpful!

Given that you were observing the issues when just using TFT_eSPI directly (ie. without GUIslice), I'm thinking that bodmer might have the most direct means of debugging the issue between the display & touch controller using the externally-driven chip select. We have seen issues in the past whereby the display freezes once touch is enabled as a result of a display (typically 8080 parallel "mcufriend" style) that pin-shares with an analog 4-wire touch, but in your case these are both SPI slaves, so it is likely a case of either the display or touch driver leaving pins in the wrong state.

I suggest posting the question in the TFT_eSPI repository; hopefully bodmer can identify the cause. Once we have that working, we can proceed with exploring the dual display driver in GUIslice (though we would want to confirm a dual TFT_eSPI works first).

SunboX, 2020-07-28:
Hey, thanks for your help! I will come back as soon as I got it working basically.

ImpulseAdventure, 2020-08-16:
Closing for the moment pending further test results. If you find an opportunity to explore the multi-display example, let me know and I will gladly reopen. Thx

closed 2020-08-16 by ImpulseAdventure

SunboX, 2020-08-16:
Hey, no problem. I will, soon. 😉