ImpulseAdventure/GUIslice #240: The 2.4-inch ILI9341 screen displays nothing, only gray

thisisausername-rgb, 2020-07-11:
### Describe the bug
*The 2.4-inch ILI9341 screen displays nothing, only gray*
[image]

### Device hardware
- *MCU*: ESP32.
- *Display with link*: ILI9341 2.4
- *GUIslice config selected*:  `esp-tftespi-default-xpt2046`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* ???
TFT_eSPI graphicstest work 
[image]

- *Touch works correctly with standalone touch library examples without GUIslice:* ???
TFI_eSPI+XPT2046_Touchscreen.h work
### Expected behavior
*A description of what you expected to happen*
Screen display GUI
### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*
```
85$&P`N⸮⸮⸮Б.u⸮GUIslice version [0.14.0]:
- Init display handler [TFT_eSPI] OK
- Init touch handler [XPT2046_PS(SPI-HW)] OK
```
### Additional info
*Add any other context about the problem here.*

Pconti31, 2020-07-11:
It might be helpful if you attached a zip file to this issue containing your config/esp-tftespi-default-xpt2046.h, src/GUIslice_config.h, src/GUIslice_version.h, and TFT_eSPI/User_Setup.h and your <whatever you compiled>.ino even if its a sample.
Paul--

thisisausername-rgb, 2020-07-11:
[myfile.zip](https://github.com/ImpulseAdventure/GUIslice/files/4906809/myfile.zip)

Pconti31, 2020-07-12:
While I don't think its your problem (we need to wait for Calvin's input for that) one thing I did notice is that your User_Setup.h is configured to use TFT_eSPI's internal touch driver.  In this case you should be using esp-tftespi-default-xpt2046_int.h as your config file.
Paul--

thisisausername-rgb, 2020-07-12:
thanks a lot，after using esp-tftespi-default-xpt2046_int.h instead of esp-tftespi-default-xpt2046.h，My problem has been solved.
[image]

closed 2020-07-12 by thisisausername-rgb

SunboX, 2020-09-21:
hey @thisisausername-rgb I'm currently triying the exact same thing. But I don't get it to be working. Can you send me you setup / project as a starting point please? I don't know what I'm doing wront. Simple eSPI test runs fine, but I don't get it working using GUIslice.

thisisausername-rgb, 2020-09-21:
> hey @thisisausername-rgb I'm currently triying the exact same thing. But I don't get it to be working. Can you send me you setup / project as a starting point please? I don't know what I'm doing wront. Simple eSPI test runs fine, but I don't get it working using GUIslice.

[TFT_eSPI.zip](https://github.com/ImpulseAdventure/GUIslice/files/5256053/TFT_eSPI.zip)

SunboX, 2020-09-21:
Hm, that's the eSPI library. I know that one. ;) And I got it working with it. ;) But for some reason I can't get the GUIslice working with it. A small example would be nice, I guess I'm doing something wrong. But I don't know what.

thisisausername-rgb, 2020-09-21:
> 嗯，这就是eSPI库。我知道那个。;）并且我可以使用它。;）但是由于某种原因，我无法让GUIslice使用它。一个小例子会很好，我想我做错了。但是我不知道。

[GUIslice.zip](https://github.com/ImpulseAdventure/GUIslice/files/5256755/GUIslice.zip)

here is a example built by guisliceBuilder
[SimpCount.zip](https://github.com/ImpulseAdventure/GUIslice/files/5256765/SimpCount.zip)

Pconti31, 2020-09-21:
@SunBoX Did you comment out all references to TOUCH_CS inside TTFT_eSPI's User_Setup.h
and use GUIslice/config/ esp-tftespi-default-xpt2046_int.h inside GUIslice/src/GUIslice_Config.h?
Paul--

SunboX, 2020-09-21:
I'm one step further ... now I'm getting those Exceptions:
```
.pio\libdeps\esp32dev\GUIslice\src\GUIslice_drv_tft_espi.cpp:1371:17: error: 'struct gslc_tsGui' has no member named 'nSwapXY'
       if (pGui->nSwapXY) {
                 ^
.pio\libdeps\esp32dev\GUIslice\src\GUIslice_drv_tft_espi.cpp:1377:19: error: 'struct gslc_tsGui' has no member named 'nFlipX'
         if (pGui->nFlipX) {
                   ^
.pio\libdeps\esp32dev\GUIslice\src\GUIslice_drv_tft_espi.cpp:1380:19: error: 'struct gslc_tsGui' has no member named 'nFlipY'
         if (pGui->nFlipY) {
                   ^
```

I'm using PlatformIO to run the code. Here's my project:
[Uploading DisplayTest (2).zip…]()

SunboX, 2020-09-21:
Fixed it with https://github.com/ImpulseAdventure/GUIslice/issues/269