ImpulseAdventure/GUIslice #241: Useage of XGraph is very obscure

rotus8, 2020-07-14:
### Feature Request or Enhancement
XGraph appears a useful element and I have an application where I would like to use it, but there is little documentation on how to use it. Creation is straightforward. Scrolling (ScrollSet) is not obvious how to use, I don't see how to actually add points to the graph, and how to set the color of the points. There are different styles defined, but no description of what they are and how to implement the data.

It would be nice if there was an example; there is an example in the builder release but it is not very helpful. 

Maybe there is some documentation I missed. Any help would be most appreciated.

### Additional context
*Add any other context or screenshots about the feature request here.*

Pconti31, 2020-07-15:
I agree this control needs some major work to be useful, like being able to input or scale the X position.
Having labels and axis, an array of array of points to show multiple line graphs and so on...
Anyway, attached is a simpler example showing a sin wave.  I hope it helps.
Paul--
[ex_bld_graph_sin.zip](https://github.com/ImpulseAdventure/GUIslice/files/4927664/ex_bld_graph_sin.zip)

rotus8, 2020-07-15:
Thank you, this helps some. Unfortunately I
        cannot test for about a week due to other things. Where is the
        color of the points set? I assume it is colGraph in the
        ElemXGraphCreate. Can the color of the points be changed after
        creation? Does one of the parameters of ElemSetCol set the color
        of the points?
    Thanks for the support.
    Ted
    
    On 7/15/2020 1:12 PM, Paul Conti wrote:
    
    
      
      I agree this control needs some major work to be useful, like
        being able to input or scale the X position.
        Having labels and axis, an array of array of points to show
        multiple line graphs and so on...
        Anyway, attached is a simpler example showing a sin wave. I hope
        it helps.
        Paul--
        ex_bld_graph_sin.zip
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-658984235",
"url": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-658984235",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Pconti31, 2020-07-16:
Yes, colGraph sets the dot color (white in the example I posted).
The example I posted is meant to show using the `map` function to scale floating point  values into something XGraph can draw.
And No there is no API call to modify the dot color at runtime and even if there was it
wouldn't do what you wanted. It would change all of the dots on screen at once to the new color when I image you would really want only new values to have a different color.  
Paul--

rotus8, 2020-07-16:
If I have, for example, four graphs superimposed but with different colGraph values will the dots from each be visible? 

⁣Ted​

Pconti31, 2020-07-16:
@rotus8 
Sorry, as I said above it would be an excellent enhancement to support an array of array of points to show multiple line graphs superimposed on the one graph but currently this object only supports one array of data for one dot graph and one color for dots.  
@ImpulseAdventure I'm pretty sure Calvin will mark this topic as an enhancement request.
Paul--

ImpulseAdventure, 2020-07-16:
Hi guys -- yes, the XGraph could definitely do with some rework. It was only meant as a simple example of what could be done, but it is missing many features that could make it more useful, such as proper incremental redraw (would greatly speed it up and reduce flicker), multiple points, connected lines, etc.

Also, it looks like the example to demonstrate the control (ex11) wasn't included in the arduino example set (but it did appear in the LINUX examples). We'll have to add that later. Lastly, I agree that some proper documentation is necessary for the control -- again, this wasn't intended as a productized control, rather it was an example of showing how to create a different control type.

Anyway, for the purposes of your request, Ted, I have put together a very quick modification to the existing XGraph (tentatively called XGraph4) that should enable you to display 4 concurrent graphs. On top of this, it also supports setting a different color for each graph.

You can modify the number of graphs by changing GSLC_XGRAPH4_NUM_GRAPHS in XGraph4.c.

You can change the color of a specific graph with gslc_ElemXGraph4SetCol().
When you add points to this multi-graph, you need to pass in an array of int16_t (eg. int16_t[4]) as the points will all be added & drawn concurrently with the current implementation.

Please see example ex_bld_graph4_sin. The only modifications I made from Paul's example was:
- Search & replace XGraph with XGraph4
- Add XGraph4SetCol() to customize the individual graph colors
- Updated the ElemXGraph4Add() call to pass in an array of values from 4 sinusoids

Hope that helps!
Cal
[XGraph4_patch1.zip](https://github.com/ImpulseAdventure/GUIslice/files/4932275/XGraph4_patch1.zip)

rotus8, 2020-07-16:
Awesome! Thank you, I will test next week.
    Ted
    
    On 7/16/2020 8:03 AM, Calvin Hass
      wrote:
    
    
      
      Hi guys -- yes, the XGraph could definitely do with some
        rework. It was only meant as a simple example of what could be
        done, but it is missing many features that could make it more
        useful, such as proper incremental redraw (would greatly speed
        it up and reduce flicker), multiple points, connected lines,
        etc.
      Also, it looks like the example to demonstrate the control
        (ex11) wasn't included in the arduino example set (but it did
        appear in the LINUX examples). We'll have to add that later.
        Lastly, I agree that some proper documentation is necessary for
        the control -- again, this wasn't intended as a productized
        control, rather it was an example of showing how to create a
        different control type.
      Anyway, for the purposes of your request, Ted, I have put
        together a very quick modification to the existing XGraph
        (tentatively called XGraph4) that should enable you to display 4
        concurrent graphs. On top of this, it also supports setting a
        different color for each graph.
      You can modify the number of graphs by changing
        GSLC_XGRAPH4_NUM_GRAPHS in XGraph4.c.
      You can change the color of a specific graph with
        gslc_ElemXGraph4SetCol().
        When you add points to this multi-graph, you need to pass in an
        array of int16_t (eg. int16_t[4]) as the points will all be
        added & drawn concurrently with the current implementation.
      Please see example ex_bld_graph4_sin. The only modifications I
        made from Paul's example was:
      
        Search & replace XGraph with XGraph4
        Add XGraph4SetCol() to customize the individual graph colors
        Updated the ElemXGraph4Add() call to pass in an array of
          values from 4 sinusoids
      
      Hope that helps!
        Cal
        XGraph4_patch1.zip
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-659471545",
"url": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-659471545",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

ImpulseAdventure, 2020-08-02:
Closing for now, pending further updates. If you have an opportunity to test the multi-trace graph patch and notice any concerns, please feel free to post to this issue and I will reopen it.

I still have an internal TODO to create a proper "plot" or graph control that has more flexibility, and also upload an ex11_ard_graph to the repo.

thx

closed 2020-08-02 by ImpulseAdventure

rotus8, 2020-08-09:
Thanks for the work to enhance the graphX
        function for me. I finally got around to working it into my
        code. 
      
    Strangely, on compiling the ex_bld_graph4_sin for
        esp32, I get the following error:
         ex_bld_graph4_sin:251:46: error: invalid
        conversion from 'int16_t {aka short int}' to 'int16_t* {aka
        short int*}' [-fpermissive]
             gslc_ElemXGraph4Add(&m_gui,m_pElemGraph1,nY);
    This is in the routine graph_sin() which isn't
        even called in the code. I commented out line 251 and it
        compiles and runs with the expected display
      
    However, when I simply change my ElemXGraphCreate
        to ElemXGraph4Create along with associated initialization, 
        while the sketch does compile, it panics just creating the
        graph, no data point additions implemented yet. Commenting out the Graph4Create,
          Graph4set Style,  and Graph4SetCol lines, the sketch compiles
          and runs normally.
         GUIslice version [0.14.0.18]:
             - Init display handler [TFT_eSPI] OK
             - Init touch handler [STMPE610(SPI-HW)] OK
            Guru Meditation Error: Core  1 panic'ed (LoadProhibited).
        Exception was unhandled.
            Core 1 register dump:
            PC      : 0x400e4a4f  PS      : 0x00060130  A0      :
        0x800dbece  A1      : 0x3ffb13f0  
            A2      : 0x3ffd0c74  A3      : 0x00000000  A4      :
        0x00000000  A5      : 0x00000005  
            A6      : 0x00000000  A7      : 0x00000000  A8      :
        0x800e4a4f  A9      : 0x3ffb13d0  
            A10     : 0x00000000  A11     : 0x00000000  A12     :
        0x3ffb1380  A13     : 0x00000081  
            A14     : 0x00000000  A15     : 0x00000b40  SAR     :
        0x0000001f  EXCCAUSE: 0x0000001c  
            EXCVADDR: 0x0000005c  LBEG    : 0x4000c46c  LEND    :
        0x4000c477  LCOUNT  : 0x00000000  
      
    Is it possible that there is an incompatibility
        with the ESP32 libraries?
    I have attached my source files. It is set to run
        on an ESP32 Feather with 3.5" TFT FeatherWing display.
      
    Thank you for your work.
      
    Ted
      
    On 7/16/2020 8:03 AM, Calvin Hass
      wrote:
    
    
      
      Hi guys -- yes, the XGraph could definitely do with some
        rework. It was only meant as a simple example of what could be
        done, but it is missing many features that could make it more
        useful, such as proper incremental redraw (would greatly speed
        it up and reduce flicker), multiple points, connected lines,
        etc.
      Also, it looks like the example to demonstrate the control
        (ex11) wasn't included in the arduino example set (but it did
        appear in the LINUX examples). We'll have to add that later.
        Lastly, I agree that some proper documentation is necessary for
        the control -- again, this wasn't intended as a productized
        control, rather it was an example of showing how to create a
        different control type.
      Anyway, for the purposes of your request, Ted, I have put
        together a very quick modification to the existing XGraph
        (tentatively called XGraph4) that should enable you to display 4
        concurrent graphs. On top of this, it also supports setting a
        different color for each graph.
      You can modify the number of graphs by changing
        GSLC_XGRAPH4_NUM_GRAPHS in XGraph4.c.
      You can change the color of a specific graph with
        gslc_ElemXGraph4SetCol().
        When you add points to this multi-graph, you need to pass in an
        array of int16_t (eg. int16_t[4]) as the points will all be
        added & drawn concurrently with the current implementation.
      Please see example ex_bld_graph4_sin. The only modifications I
        made from Paul's example was:
      
        Search & replace XGraph with XGraph4
        Add XGraph4SetCol() to customize the individual graph colors
        Updated the ElemXGraph4Add() call to pass in an array of
          values from 4 sinusoids
      
      Hope that helps!
        Cal
        XGraph4_patch1.zip
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-659471545",
"url": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-659471545",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Pconti31, 2020-08-09:
@rotus8 Well, the sample runs on the ESP32 Feather with Display you gave me so I don't think its a ESP32 problem.   It looks like a memory read pass allocation to me.
For one graph you set something like:
```
int16_t                         m_anGraphBuf1[300]; // NRows=300
```
While for 4 you set:
```
int16_t                         m_anGraphBuf1[4*300]; // NRows=300
```
`gslc_ElemXGraph4Add()` requires an array of points, not a single one.
everything else stays the same

If m_anGraphBuf1 size isn't your problem and
 If your program isn't proprietary or massive in size zip it and attach
an we can take a look. 
Paul--

rotus8, 2020-08-09:
As always, thanks for the instant reply. The
        example is running on my setup, less the one statement that is
        unused anyway. 
       My previous email attached my
          code, although it is massive.
    My graph buffer is set quite large (5760),
        experimenting with it did not make a difference.
    Ted
    
    [
{
"@context": "http://schema.org",
"@type": "EmailMessage",
"potentialAction": {
"@type": "ViewAction",
"target": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-671098692",
"url": "https://github.com/ImpulseAdventure/GUIslice/issues/241#issuecomment-671098692",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Pconti31, 2020-08-09:
@rotus8 Sorry, my eye sight isn't what it used to be but I do not see any zip files attached but for calvins and mine.  So please repost the zip of the project.  
Paul--

rotus8, 2020-08-09:
I guess it was stripped off my simple email reply. Using github upload function...
[display.zip](https://github.com/ImpulseAdventure/GUIslice/files/5048502/display.zip)

Pconti31, 2020-08-09:
@rotus8 I think your problem is this:

  // Create graph E_ELEM_GRAPH_HISTORY
  pElemRef = gslc_ElemXGraph4Create(&m_gui,E_ELEM_GRAPH_HISTORY,E_PG_HISTORY,
    &m_sGraph1,(gslc_tsRect){10,48,300,400},E_FONT_TXT5,(int16_t*)&m_anGraphBuf1,
        **_5760_**,((gslc_tsColor){255,200,0}));

I believe it should be:
  // Create graph E_ELEM_GRAPH_HISTORY
  pElemRef = gslc_ElemXGraph4Create(&m_gui,E_ELEM_GRAPH_HISTORY,E_PG_HISTORY,
    &m_sGraph1,(gslc_tsRect){10,48,300,400},E_FONT_TXT5,(int16_t*)&m_anGraphBuf1,
        **_1440_**,((gslc_tsColor){255,200,0}));

Remember this routine is just a quick hack of the original and it expects 4 point not one
so you could alloc memory as m_anGraphBuf1[1440*4] or the [5760] but the creation wants 5760/4 in that case because this number is for the number of point sets of 4!

Paul--

rotus8, 2020-08-09:
Excellent, thank you! I'll make that a define so I don't forget.