ImpulseAdventure/GUIslice #243: gslc_ElemSetTxtStr does not update element correctly

nickjrz, 2020-07-21:
Hi all,
 
I have a long press button functionality but it only works for a certain amount of times and then the long press is disabled. Would you guys know a solution for this? Attached to this is my code 

`// Button callbacks
// - This function gets called when the button is pressed
bool CbBtnPower(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  if (eTouch == GSLC_TOUCH_DOWN_IN) {
    m_nTimePressStart = millis();
  } else if (eTouch == GSLC_TOUCH_UP_IN) {
    if ((millis() - m_nTimePressStart) >= 500)
    {  
      if (powerUp_triggered ==false){
        // Set a variable flag that we can use elsewhere
        powerUp_triggered = true;
        powerDown_triggered = false;
        static char userStatus[28] = "STATUS: GETTING READY";
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128});        
        Serial.println(true);   
      }
      else{
        powerUp_triggered = false;
        powerDown_triggered = true;
        static char userStatus[28] = "STATUS: SHUTTING DOWN";
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255});
        Serial.println(false);
      }  
    }else{
      static char userStatus[28] = "Hold button down for 5 sec";
      gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
        
      }
  }
  return true;
  
} `

Pconti31, 2020-07-22:
Sorry, but its really hard to hunt down the problem without seeing more of a program.
If you can't post the whole set of files (as a zip file) can you cut the program down to a smaller sample that uses this routine?
My only thought here is that you have failed to check 
```
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui,pElemRef);

if (pElem->nId== your button enum) {

}
```
so if you have more than this one button defined it wouldn't work as expected.  
By the way, 5 seconds is 5000 milliseconds not 500.  Milli stands for thousands so 1 sec is 1000.
Paul--

nickjrz, 2020-07-22:
[arduinoControlGUI.txt](https://github.com/ImpulseAdventure/GUIslice/files/4960284/arduinoControlGUI.txt)

Attached is the whole code and this is the only button defined on the code. Thanks for the quick reply.

Pconti31, 2020-07-22:
Ok, I tested the code and found some issues.
First you should check your Serial.read to make sure you do not overwrite your input buffer.
Second, the GSLC_TOUCH_UP_IN only occurs when the user touches than releases the button.
Which means that you are not checking each cycle.  So if the user after the first time while the power button is RED
fails to hold for the full three seconds your code stops working.  
You should check GSLC_TOUCH_MOVE_IN each cycle to see if the timer expired and give some feed back to the user
about time remaining say with a progress bar.  Also, when modifying an element other than the button you need to call:
```
    gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
```
for proper refresh of the text field.

In tracing I found
```
    }else{
      static char userStatus[28] = "Hold button down for 5 sec";
      gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
      if ((millis() - m_nTimePressStart) < 3000)
        Serial.println("DID NOT HOLD 3 seconds");
      Serial.print("powerUp_triggered: "); Serial.println(powerUp_triggered);
      Serial.print("powerDown_triggered: "); Serial.println(powerDown_triggered);
      Serial.print("m_nTimePressStart: "); Serial.println(m_nTimePressStart);
    }
```
I found that in the case where user fails to wait full three seconds (after the first successful wait) the variables are set as follows:
```
powerUp_triggered: 1
powerDown_triggered: 0
```
Your code seems to only works when:
```
powerUp_triggered: 0
powerDown_triggered: 0
```
My solution would be to create a finite state machine that describes what you want this power button to do,
and monitor GSLC_TOUCH_MOVE_IN  giving feedback to the user (progress bar?) so they know when to release or come up with a different UI that easier to understand...
Paul--

nickjrz, 2020-07-22:
Thanks for the help Paul, I will address those issues. 

The button is basically a one button start-up for a system, when the user presses the power button the system would boot up and vice versa. 

So for the button is there a way to have a button that would change to the other state once the user holds for 3 seconds (not a hold and release button) , the issue is that I am using an arduino uno so my storage space is very limited and using a dynamic element might take more space.

Pconti31, 2020-07-22:
Well, usually we do that in hardware since if we don't have power we don't have an active tft screen.  Think the typical android phone or tablet.
See [https://www.pololu.com/category/121/pololu-power-switches](url)
as an example.  
If you must have a toggle button with timer I don't think you are far off in your implementation. 
Let me know if you can't fix it and I'll see what I can come up with and post my solution for you.
Paul--

nickjrz, 2020-07-22:
The USB port where the device is connected will always have power. I want to have a hold button because the system has multiple sensors and processes going on at the same time and don't want the user to press the button by accident and turn off the whole system, so holding the button ensures that they are really trying to shut it off.

and  in the case where user fails to wait full three seconds (after the first successful wait) the variables are set as follows, because the code detects the short button press as a good long button press, that is why it sets those variables to those states:

`powerUp_triggered: 1
powerDown_triggered: 0`

it works now but the thing is that after multiple long button presses the functionality of the timer for button is disabled.
also what does this GSLC_TOUCH_MOVE_IN  represent in the context and how can I monitor it?

Pconti31, 2020-07-23:
I think I found your problem.  You defined:
```
int m_nTimePressStart = 0;
```
which on an UNO is a signed 16 bit number varying from -32768 to +32767
while millis() returns uint32_t an unsigned 32 bit number that varys from 0 to 4,294,967,295
so there were times when you did millis() - m_nTimePressStart and you got negative numbers
which would never be >= 3000
so change to:
```
unit32_t m_nTimePressStart;
```
should solve you problem.
Now checking GSLC_TOUCH_MOVE_IN  can allow you to give feedback to your users in your callback code.
a crude example say you add another text field E_ELEM_COUNT then inside your button callback:
```
  } else if (eTouch == GSLC_TOUCH_MOVE_IN) {
    uint32_t nTimeLeft = millis() - m_nTimePressStart;
    Serial.print("Test Timer: ");  Serial.println(nTimeLeft);
    if (nTimeLeft < 3000) {
      char acTxt[28];
      sprintf(acTxt,"%ldms",nTimeLeft);
      gslc_ElemSetTxtStr(&m_gui,m_pElemCount ,(char*)acTxt); 
      gslc_ElemSetRedraw(&m_gui, m_pElemCount, GSLC_REDRAW_INC);
    } else {
      gslc_ElemSetTxtStr(&m_gui,m_pElemCount ,(char*)" "); 
      gslc_ElemSetRedraw(&m_gui, m_pElemCount, GSLC_REDRAW_INC);
    }
  } else if (eTouch == GSLC_TOUCH_UP_IN) {
```
you would give your users a count of the passing milliseconds and blank the field when it hit.
You could also print a message release button at that point.
Paul--

Pconti31, 2020-07-23:
@nickjrz Sorry, but after a good night sleep I realized two things.
  One that I didn't explain GSLC_TOUCH_MOVE_IN.   This eTouch event is triggered during the user pressing the button.
So you are getting hundreds of these during this time, but its useful in this case so you can monitor the time lapse to determine if the three seconds have expired,  Once you detect that you can give feedback to the user so they know they can release the button.  Again, the GSLC_TOUCH_UP_IN only occurs after the user has released the button so in your existing code the user has to count 1001,1002,1003 to guess if its safe to release.
  The second thing is that the easiest way to inform the user is to simply do the same as you are doing in the GSLC_TOUCH_UP_IN and change the color of the button.  I tested this with your code and it works great.  No other changes to your code is required except to use uint32_t for m_nTimePressStart.
The full code is here
```
uint32_t m_nTimePressStart = 0;`
...
  } else if (eTouch == GSLC_TOUCH_MOVE_IN) {
    if (millis() - m_nTimePressStart >= 3000) {
      if (powerUp_triggered ==false){
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128});        
      }
      else {
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255});
      }  
    }
  } else if (eTouch == GSLC_TOUCH_UP_IN) {
```
Paul--

nickjrz, 2020-07-23:
Hey Paul,

Thanks for your input. So setting the variable m_nTimePressStart to uint32_t fixed the issue of the long press being disabled after a good successful long wait. 

I also tried the second option to inform the user when to release the button by changing the color of the button but I could not get that one to work correctly as the text keeps refreshing/ blinking while I hold the button down. This is what I what I have:

`uint32_t m_nTimePressStart = 0;

// Button callbacks
// - This function gets called when the button is pressed
bool CbBtnPower(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem* pElem = pElemRef->pElem;

  // Determine what type of event occurred on the button
  // - In this case we're just looking for the user releasing
  //   a touch over the button.
  if (eTouch == GSLC_TOUCH_DOWN_IN) {
    m_nTimePressStart = millis();
  } else if (eTouch == GSLC_TOUCH_MOVE_IN) {
    if ((millis() - m_nTimePressStart) >= 2000)
    {  
      if (powerUp_triggered ==false && xonarSafe_running == false){
        flushReceive();
        // Set a variable flag that we can use elsewhere
        powerUp_triggered = true;
        powerDown_triggered = false;
        static char userStatus[28] = "STATUS: GETTING READY";
        gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128});        
        Serial.println(true);   
        xonarSafe_running  = true;
      }
      else{
        if (xonarSafe_initialized == true && xonarSafe_running == true){
          flushReceive();
          powerUp_triggered = false;
          powerDown_triggered = true;
          static char userStatus[28] = "STATUS: SHUTTING DOWN";
          gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
          gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
          gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255});
          Serial.println(false);
          xonarSafe_initialized  = false;
          }
          else{
            gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
            static char userStatus[28] = "System is not initialized";
            gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
            }
      }  
    }else{
      gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
      static char userStatus[28] = "Hold button down for 2 sec";
      gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
      }
  }
  else if (eTouch == GSLC_TOUCH_UP_IN){}
  return true; 
} `

Pconti31, 2020-07-23:
Easy enough to get rid of the flicker just add another boolean.
```
bool CbBtnPower(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  static bool nTimerExpired = false;
  
  // Determine what type of event occurred on the button
  // - In this case we're just looking for the user releasing
  //   a touch over the button.
  if (eTouch == GSLC_TOUCH_DOWN_IN) {
    m_nTimePressStart = millis();
    nTimerExpired = false;
    Serial.print("Timer Started: ");  Serial.println(m_nTimePressStart);
  } else if (eTouch == GSLC_TOUCH_MOVE_IN && !nTimerExpired) {
    if (millis() - m_nTimePressStart >= 3000) {
      if (powerUp_triggered ==false){
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128});        
      }
      else {
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255});
      }  
      nTimerExpired = true;
    }
  } else if (eTouch == GSLC_TOUCH_UP_IN) {
    if (millis() - m_nTimePressStart >= 3000)
    {  
      if (powerUp_triggered ==false){
        // Set a variable flag that we can use elsewhere
        powerUp_triggered = true;
        powerDown_triggered = false;
        static char userStatus[28] = "STATUS: GETTING READY";
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
//        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128});        
        Serial.println(true);   
      }
      else {
        powerUp_triggered = false;
        powerDown_triggered = true;
        static char userStatus[28] = "STATUS: SHUTTING DOWN";
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
//        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255});
        Serial.println(false);
      }  
    }else{
      static char userStatus[28] = "Hold button down for 3 sec";
      gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
      gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
    }
  }
  return true;
  
} 
```
Paul--

nickjrz, 2020-07-23:
Thanks Paul, this works to get rid of the flickering but gave me a weird behavior with the button so I optimized to this:

`bool CbBtnPower(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  static bool nTimerExpired = false;
  // Determine what type of event occurred on the button
  // - In this case we're just looking for the user releasing
  //   a touch over the button.
  if (eTouch == GSLC_TOUCH_DOWN_IN) {
    m_nTimePressStart = millis();
    nTimerExpired = false;
  } else if (eTouch == GSLC_TOUCH_MOVE_IN  && !nTimerExpired) {
    if ((millis() - m_nTimePressStart) >= 3000){        
      if (powerUp_triggered ==false && xonarSafe_running == false){
         flushReceive();
        // Set a variable flag that we can use elsewhere
        powerUp_triggered = true;
        powerDown_triggered = false;
        static char userStatus[28] = "STATUS: GETTING READY";      
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128}); 
        gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
        Serial.println(true);   
        xonarSafe_running  = true;     
      }
      else {
        if (xonarSafe_initialized == true && xonarSafe_running == true){
          flushReceive();
          powerUp_triggered = false;
          powerDown_triggered = true;
          static char userStatus[28] = "STATUS: SHUTTING DOWN";
          gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
          gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255}); 
          gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
          Serial.println(false);
          xonarSafe_initialized  = false;
          }
          else{
            static char userStatus[28] = "System is not initialized";
            gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
            gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
            }
          }  
        nTimerExpired = true;
    }
  } else if (eTouch == GSLC_TOUCH_UP_IN) { 
    if (millis() - m_nTimePressStart < 3000)
    {
      static char userStatus[28] = "Hold button down for 2 sec";
      gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
      gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
      }
  }
  return true; 
} 
`

This only changes the state of the button when the user has held the button for 3 seconds or more. Thanks for your all of your help .

nickjrz, 2020-07-23:
Thanks Paul, this solves the flickering issue but gave me a weird behavior on the button so I optimized  it so that the button only changes states when the user presses the button for 3 seconds or more.

`bool CbBtnPower(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  static bool nTimerExpired = false;
  // Determine what type of event occurred on the button
  // - In this case we're just looking for the user releasing
  //   a touch over the button.
  if (eTouch == GSLC_TOUCH_DOWN_IN) {
    m_nTimePressStart = millis();
    nTimerExpired = false;
  } else if (eTouch == GSLC_TOUCH_MOVE_IN  && !nTimerExpired) {
    if ((millis() - m_nTimePressStart) >= 3000){        
      if (powerUp_triggered ==false && xonarSafe_running == false){
         flushReceive();
        // Set a variable flag that we can use elsewhere
        powerUp_triggered = true;
        powerDown_triggered = false;
        static char userStatus[28] = "STATUS: GETTING READY";      
        gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
        gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{255,0,0},{255,144,128}); 
        gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
        Serial.println(true);   
        xonarSafe_running  = true;     
      }
      else {
        if (xonarSafe_initialized == true && xonarSafe_running == true){
          flushReceive();
          powerUp_triggered = false;
          powerDown_triggered = true;
          static char userStatus[28] = "STATUS: SHUTTING DOWN";
          gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
          gslc_ElemSetCol(&m_gui, m_pElemTextBTN,{255,255,255},{0,0,255},{30,144,255}); 
          gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
          Serial.println(false);
          xonarSafe_initialized  = false;
          }
          else{
            static char userStatus[28] = "System is not initialized";
            gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus); 
            gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
            }
          }  
        nTimerExpired = true;
    }
  } else if (eTouch == GSLC_TOUCH_UP_IN) { 
    if (millis() - m_nTimePressStart < 3000)
    {
      static char userStatus[28] = "Hold button down for 2 sec";
      gslc_ElemSetTxtStr(&m_gui,m_pElemText ,(char*)userStatus);
      gslc_ElemSetRedraw(&m_gui, m_pElemText, GSLC_REDRAW_INC);
      }
  }
  return true; 
} 
`

Thanks for all of the help,
Nick

closed 2020-07-23 by nickjrz

ImpulseAdventure, 2020-07-27:
I realize this issue was already resolved thanks to Paul's feedback. However, I thought I would followup on one of the items below regarding redraw.

## Automatic Redraw
In general, I believe one shouldn't need to call `ElemSetRedraw()` manually as the element setter functions should take care of it for you. @Pconti31 -- do you recall if there were any specific APIs (eg. `ElemSetCol`, `ElemSetTxtStr`, etc.) or element types in which it appeared we still required an `ElemSetRedraw()` afterwards for redraw to occur?

## Element Color Update Flicker
On a related note, users probably wouldn't expect flicker to occur when repeatedly calling `ElemSetCol()` with the same colors already assigned to the element. I see that I have already implemented the checks to avoid redraw with superfluous calls to `ElemSetTxtCol()` and `ElemSetTxtStr()`, however, it looks like I'm missing the same color checks in `ElemSetCol()`.

Therefore, I plan to fix `ElemSetCol()` to only perform a redraw if the color values have actually been changed. This should enable the user to call `ElemSetCol()` repeatedly without causing any extra redraw flicker. This may also obviate the need for an extra boolean state.

Thoughts?

ImpulseAdventure, 2020-07-31:
For future reference, I have now pushed changes to the repository that will optimize the redraw for `ElemSetCol()` and `ElemSetGlowCol()`. You should be able to call those functions repeatedly with the same values without causing redraw flicker.

Thus I don't think there should be any need to call `ElemSetRedraw()` manually for the standard GUI