ImpulseAdventure/GUIslice #244: Check-box off return ID_NONE - why?

JohannesWilde, 2020-07-26:
Good day,

this is more a usage-question than an actual bug report.

I am wondering, why a check box does not return its own ID when it gets unselected.

GUIslice/src/elem/XCheckbox.c : appr. line 235
```
   // For checkbox:
   // - If not selected:  return ID_NONE     and state=false
   // - If     selected:  return current ID  and state=true

      if (bChecked) {
        nSelId = pElem->nId;
      } else {
        nSelId = GSLC_ID_NONE;
      }
```

Why not always return the current ID? This would allow one in the callback:

GUIslice/src/elem/XCheckbox.c : appr. line 258

```
    // Now send the callback notification
    (*pCheckbox->pfuncXToggle)((void*)(pGui), (void*)(pElemRef), nSelId, bChecked);
```

to use something like:
```

    switch (nSelId)
    {
    //<CheckBox Enums !Start!>
    case E_CHECK_XYZ:
    {
        if (bChecked)
        {
            // check box selected
        }
        else
        {
            // check box unselected
        }
        break;
    }
    }
```

On the other hand I can determine the current element using pElemRef - this however seems superfluous to me.

Is there a reason for this design-decision, that I currently can't see on my own?

I really woul appreciate any feedback. Thanks, and have a nice weekend,

Johannes

JohannesWilde, 2020-07-26:
I open a corresponding merge request -> #245 .

ImpulseAdventure, 2020-07-26:
Hi Johannes -- good question!

The checkbox logic shares much of the same implementation with the radio buttons. For radio buttons, it seemed to make sense to return `ID_NONE` if no buttons were selected (even though this may be an uncommon callback scenario). The same convention was used for checkboxes, but you make a very good point. It would seem more natural for the ID of the element triggering the event to always be returned (including a checkbox disable), and then rely on the `bChecked` parameter to differentiate the state.

I do think it would be a good idea to adjust the callback slightly as you have suggested. There is a chance that some users may already be depending on an `ID_NONE` for a checkbox deactivation event, but my expectation is that this could be relatively uncommon.

Since this could be a "breaking change" in the API for some, I am thinking that I will incorporate this into the upcoming 0.15.0 release (which should be very soon), unless I hear an opposing viewpoint on this change.

JohannesWilde, 2020-07-28:
Sounds good. Thanks for your feedback.