ImpulseAdventure/GUIslice #246: [Feature request] Support element move

Lucy2003, 2020-08-03:
### Feature Request or Enhancement
All elements are designed to have a X and Y,but can't be changed in CallBack.I can't find any function like "gslc_setElemXY" in API.
Is it possible to add this function for the elements not in FLASH? 

BTW:If the box element (and so on) can have a touch callback,that will be better!
In that case, I can read touch position and move the them like a window!

### Additional context
None.

Pconti31, 2020-08-03:
@Lucy2003 Callbacks for box are already supported.
Using the Builder just set the property "Touch Enabled?" = true.
I suspect you will also need to enable the draw callback to do your window.

It also may be possible to change x and y by directly modifying element ref structure
gslc_tsElem->rElem since its not protected.   Although it could screw up page refresh.
Note that if do try this you should also call either
```
    gslc_ElemSetRedraw(pGui,pElemRef,GSLC_REDRAW_INC);
or
    gslc_ElemSetRedraw(pGui,pElemRef,GSLC_REDRAW_FULL);
```
on any modified Element.  Supported API calls do this automatically but not if you directly
change data without an API call.

Paul--

Lucy2003, 2020-08-03:
I will try it.Thanks for reply.

ImpulseAdventure, 2020-08-03:
Hi @Lucy2003 --

I have implemented the basic implementation to support dynamic element movement and resizing. As Paul points out, you could modify an element's rect directly. However, with the following pull request #247, you should not need to call any redraw routines as it should be handled automatically.

Please try out Pull Request #247 

**Example code to resize an element:**
```c++
  gslc_tsRect rElemCur = (gslc_tsRect) { 100, 200, 80, 40 };
  gslc_ElemSetRect(&m_gui,pElemRefQuit,rElemCur);
```

**Example code to make an element move and "bounce" around a boxed region:**
```c++
void DoMove()
{
    // - Find the element from the enumeration ID
    gslc_tsElemRef* pElemRefQuit = gslc_PageFindElemById(&m_gui, E_PG_MAIN, E_ELEM_BTN_QUIT);
    gslc_tsRect rElemCur = gslc_ElemGetRect(&m_gui,pElemRefQuit);
    
    // - Move the element diagonally, but wrap at edges
    rElemCur.x = (rElemCur.x < 220)? rElemCur.x + 10 : 15;
    rElemCur.y = (rElemCur.y > 60)? rElemCur.y - 10 : 150;
    
    // - Update the element
    gslc_ElemSetRect(&m_gui,pElemRefQuit,rElemCur);  
}
```

If this works for you, I will merge the update.

Hope that helps!

Lucy2003, 2020-08-04:
Hi,ImpulseAdventure,

Element movement works well!

#### But,"Touch Enabled = True" seems not work.
It generated code,
  in `CbBtnCommon(...)` : `case E_BOX_TitleBar:`
  in  `InitGUIslice_gen()` : `gslc_ElemSetTouchFunc(&m_gui, pElemRef, &CbBtnCommon);`
#### But,when I touch the E_BOX_TitleBar,there is nothing output to the serial console (I add a `Serial.printf`).
Other button is normal.Maybe the `gslc_ElemSetTouchFunc` function cause this problem? 

There is my test file:
[ElemMove.zip](https://github.com/ImpulseAdventure/GUIslice/files/5019894/ElemMove.zip)

hardware:ESP32 Dev Board,with ILI9341 320x240 SPI TFT display.

Thanks.
--Lucy2003

ImpulseAdventure, 2020-08-04:
Good to hear the move is working! Thanks for providing all of the helpful details in your reply.

As for enabling the touch: by default, box and plain text elements don't capture touch events unless we enable it. It looks like we are missing the following next to your ElemSetTouchFunc() call: `gslc_ElemSetClickEn(&m_gui, pElemRef, true);`

@Pconti31 -- we might want to consider whether it makes sense for the Builder to insert this call whenever the user sets the "Touch Enabled" property to true (eg. on a box).

@Lucy2003 -- for now, you can add something like the following to your setup():
```c++
gslc_ElemSetClickEn(&m_gui, gslc_PageFindElemById(&m_gui, E_PG_MAIN, E_BOX_TitleBar), true);
```

I will proceed with merging the pull request for the dynamic move feature.

ImpulseAdventure, 2020-08-04:
GitHub auto-closed this issue too early (due to the PR merge). Reopening until we have confirmed everything is working OK (including the title bar click).

Lucy2003, 2020-08-04:
That's great! After add `gslc_ElemSetClickEn(&m_gui, gslc_PageFindElemById(&m_gui, E_PG_MAIN, E_BOX_TitleBar), true);`,it can capture touch events.

#### But,there is a new problem:when a part of the box out of the screen,some text will display by the screen border.
These characters should be unvisible.It seems the library didn't calculate the position of characters well.

There,my file,config is the same as before:
[ElemMove-1.zip](https://github.com/ImpulseAdventure/GUIslice/files/5020970/ElemMove-1.zip)
Please have a test,and you will find out what's the problem.

**Note: Please don't generate code again because some code are out of <!Begin!> and <!End!> ! If you do so,the code may be overwritten or builder will crash!**

Thank you.
--Lucy2003

ImpulseAdventure, 2020-08-04:
Nice example you have created!

The display drivers that GUIslice loads may optionally wrap text. It looks like both **Adafruit-GFX** and **TFT_eSPI** default to enabling a text wrap at the side of the screen which is not what we want.

Instead, you can call the native display driver APIs to disable the wrapping feature.

For example, I believe you can perform the following additional config in your setup:

## TFT_eSPI
I haven't tested this yet, but think it may work:
```c++
  TFT_eSPI* pDrvDisp = (TFT_eSPI*)gslc_GetDriverDisp(&m_gui);
  pDrvDisp->setTextWrap(false,false);
```

## Adafruit_ILI9341
I have tested the following on my ILI9341 display:
```c++
  Adafruit_ILI9341* pDrvDisp = (Adafruit_ILI9341*)gslc_GetDriverDisp(&m_gui);
  pDrvDisp->setTextWrap(false);
```

Does the above change the behavior you are observing?

Pconti31, 2020-08-04:
Well, I tried the ElemMove with my adruino mega and Adafruit 2.8 display with ILI9341 and like you it worked fine.
However, when I used my Adafruit ESP32 feather with 3.5 display and the TFT_wSPI driver and your code:
```
TFT_eSPI* pDrvDisp = (TFT_eSPI*)gslc_GetDriverDisp(&m_gui);
  pDrvDisp->setTextWrap(false,false);
```
Lets just say not so much.  Once the "Window Box" went past the end of the display
the box display looked ok but the Text Fields stayed outside the Box on the display canvas.
In a gray background.  It looked much neater than the artifacts that had been showing up.
For what that's worth.  Of course, it could just be my setup so hopefully its works for Lucy.
Paul--

Lucy2003, 2020-08-05:
@ImpulseAdventure I tried to add the lines you point out,but the result is the same as Pconti31.
@Pconti31 Thanks for pointing out how to improve my code!

Lucy2003, 2020-08-05:
#### I found this in TFT_eSPI.It can explain why the code doesn't work.

In TFT_eSPI.cpp:4608 (or near this line)
[image]

Pconti31, 2020-08-05:
@Lucy2003 Try adding this code:
```
          moveElem(btnYes);
          moveElem(btnCancel);
          gslc_tsRect rElemWinBox = m_asPage[1].rBounds;
          rElemWinBox = gslc_ElemGetRect(&m_gui, Window1); 
          m_asPage[1].rBounds = rElemBox;
          gslc_InvalidateRgnPage(&m_gui,&m_asPage[1]);
          return true;
        }
        break;
```
Paul--

Lucy2003, 2020-08-06:
That's very nice! I tested it,and it worked.

I use these code:
```C++
m_asPage[1].rBounds = gslc_ElemGetRect(&m_gui, Window1);
gslc_InvalidateRgnPage(&m_gui,&m_asPage[1]);
```
It will be shorter and doesn't need a new gslc_tsRect.Is there any bad influence to use these code?

From the API,
```C++
void gslc_InvalidateRgnPage  (  gslc_tsGui *  pGui, gslc_tsPage * pPage ) ;
//Include an entire page (eg.from a page stack) in the invalidation region
```

Does it set the new popup window region to be invalidation region and then GUI auto redraw?

BTW,how to calculate the page number in m_asPage[]?

Thanks.
--Lucy2003

Pconti31, 2020-08-06:
Yeah, yours is better. my code looked that way because I just cut and pasted my debug code to the screen and made a quick edit.
```
          gslc_tsRect rElemBox = m_asPage[1].rBounds;
  Serial.print("x:"); Serial.print(rElemBox.x); 
  Serial.print(" y:"); Serial.print(rElemBox.y); 
  Serial.print(" w:"); Serial.print(rElemBox.w); 
  Serial.print(" h:"); Serial.println(rElemBox.h);
          rElemBox = gslc_ElemGetRect(&m_gui, Window1); 
          m_asPage[1].rBounds = rElemBox;
          gslc_InvalidateRgnPage(&m_gui,&m_asPage[1]);
```
As to how did I know the array index.  Making use of the fact that pages are added one at a time to this array as  gslc_PageAdd() is called.  it's clear that E_PG_MAIN would be m_asPage[0] and
m_asPage[1] is E_PG_POPUP1.

You could of course use the same routine GUIslice API uses to find pages.
Start by creating a global to hold the address of the page, then finding its address in setup(),
then using the address inside your callback.  like so:
```
// Save some element references for direct access
gslc_tsPage*    m_asPopupPage     = NULL;
//<Save_References !Start!>

          m_asPopupPage->rBounds = gslc_ElemGetRect(&m_gui, Window1);
          gslc_InvalidateRgnPage(&m_gui,m_asPopupPage);
          return true;

void setup() {
...
  m_asPopupPage = gslc_PageFindById(&m_gui, E_PG_POPUP1);
  gslc_ElemSetClickEn(&m_gui, gslc_PageFindElemById(&m_gui, E_PG_POPUP1, E_BOX_TitleBar), true);
```
Now while we are discussing programmatically accessing elements what happens if you have more than one moving window and/or you are making changes to these windows during your development cycle?  You have to go in and edit the callbacks for any additions or subtractions to the moving windows when you should only need to track the UI element that you have click enabled on.  Well, it's possible to access the array of elements you have added to the page inside your callback like so:
```
      case E_BOX_TitleBar:
        if (Touching==1){
          if (lastX-nX==0) return true;
          if (lastY-nY==0) return true;
          gslc_tsRect rElemCur;
          gslc_tsElemRef* pElemRef;
          gslc_tsCollect* pCollect = &m_asPopupPage->sCollect;
          for (int i=0; i<pCollect->nElemCnt; i++) {
            pElemRef = &pCollect->asElemRef[i];
            moveElem(pElemRef);
          }
          m_asPopupPage->rBounds = gslc_ElemGetRect(&m_gui, Window1);
          gslc_InvalidateRgnPage(&m_gui,m_asPopupPage);
          return true;
        }
        break;
```
Now you can add or subtract elements to this window without modifying this callback.
You still need to make sure each UI element has been assigned an ElementRef name or this code won't work.

Paul--

Lucy2003, 2020-08-07:
The code is so good!

Now,with the new feature,

Lucy2003, 2020-08-07:
The code is so good!

Now,with the new feature, "canned" dialog in #81 is possible(when the RAM is as big as ESP8266 or so on,it can even support dialog movement).

closed 2020-08-07 by Lucy2003