ImpulseAdventure/GUIslice #248: i2c support with ESP32 and SDD1306?

X3msnake, 2020-08-25:
I'm trying to upload a test to a ESP32 with a SDD1306 that uses I2c instead of SPI

I find no documentation on how to configure this, and on the files it looks like ther is only support to SPI
How can I use GuiSlice with this hardware?

Pconti31, 2020-08-25:
@X3msnake I assume you really mean the SSD1306 OLED display driver?
In this case you first try and get  it working with Adafruit SSD1306 library and Adafruit GFX library by running the examples inside Adafruit SSD1306 library.  
Then use GUIslice config ard-adagfx-ssd1306-notouch.h for GUIslice support.
Paul--

ImpulseAdventure, 2020-08-25:
Hi @X3msnake — yes, I should be able to add support for I2C mode in the Adafruit_SSD1306 driver. As Paul mentions, we would usually start with confirming your test setup & initialization values needed to run the display with the Adafruit library directly.

Once confirmed, I will need to enhance GUIslice to add the I2C variant as an option. For this particular driver, I believe it should only involve adding a new m_disp instantiation and an updated call to m_disp.begin() in gslc_DrvInit(). I’ll take a look... in the meantime if you could provide me with a list of any mods you needed to the Adafruit library example to get it going, that would be great.

X3msnake, 2020-08-25:
What is needed is a way to initialize the i2c with custom pins.
Most examples i have tested don't work on the hardware i'm using without custom pins. I'm currently using a integrated ESP32 with integrate LION Battery and SSD1306 on pins 4 and 5

It is programable with the lolin D32 or D32 Pro

X3msnake, 2020-08-25:
i can't seem to find how to define the pins on the Adafruit library

X3msnake, 2020-08-25:
So i managed to hack the pins_arduino.h file for the lolin32 board and i can run the SSD1306 demo files directly from the examples without any need to declare pins

 https://github.com/lexus2k/ssd1306

Unfortunately the Adafruit Still returns a blank screen

X3msnake, 2020-08-26:
so after the pin map modification i am able to run the feather demo and it shows on screen but not the other i2c demos

X3msnake, 2020-08-26:
Plot thickens.
All 128x32 examples work (with missing lines as expected) but none of the 64px do

X3msnake, 2020-08-26:
And if i change the screen heigh to 64 it displays correctly the demo
So this works with the adafruit and even tho it does not have a way to set the SDA/SLC pins by modifying the pins_arduino.h file of the board i'm using.

I guess the correct way would be to find the propper pinout and make a controller for this board.

ImpulseAdventure, 2020-08-26:
@X3msnake -- I have now added support for Adafruit_SSD1306 in I2C mode (in branch https://github.com/ImpulseAdventure/GUIslice/compare/WIP248-SSD1306_Ada_I2C).

From your description above, I understand that you are now evaluating the [lexus2k/ssd1306](https://github.com/lexus2k/ssd1306) library.
GUIslice doesn't currently support lexus2k/ssd1306, however it does support the [lexus2k/lcdgfx](https://github.com/lexus2k/lcdgfx) which was supposed to be a replacement.

Could you try to see if you can run the **lcdgfx** library with your display (with the appropriate local mods you need for software I2C and display dimensions)?

thanks

X3msnake, 2020-08-26:
getting this error on compile time :|

ResolveLibrary(avr/pgmspace.h)
  -> candidates: []
compilation terminated.

X3msnake, 2020-08-26:
https://arduino-esp8266.readthedocs.io/en/latest/faq/a03-library-does-not-work.html ??

X3msnake, 2020-08-26:
so i changed every item found from avr/pgmspace.h to pgmspace
it compiled but no image

X3msnake, 2020-08-26:
i don't see how to use the lcdgfx library with guislice where do i set it up?

i can use lcdgfx demos directly

ImpulseAdventure, 2020-08-27:
Hi @X3msnake --

Thanks for confirming that the LCDGFX examples work when run standalone.
You're right... it looks like we might be missing an entry for LCDGFX in the `GUIslice_config.h` file... I'll fix that shortly.

To use LCDGFX, can you try the following:
- Edit `/src/GUIslice_config.h`:
- Add the following line under where it says "Arduino, ARM SAMD":
`#include "../configs/ard-lcdgfx-notouch.h"`
- Edit `/configs/ard-lcdgfx-notouch.h`:
- Uncomment the one line corresponding to your SSD1306 display config and comment out any others (eg. ILI9341). For example:
`#define DRV_DISP_LCDGFX_SSD1306_128x64_I2C`
- Launch Arduino IDE
- Run examples/arduino/ex01_ard_basic. Any output on the display or serial monitor?
- Run examples/arduino/ex40_ard_ctrls_small

thanks!

X3msnake, 2020-08-29:
Error!

GUIslice_drv.h:57:4: error: #error "Driver needs to be specified in GUIslice_config_*.h (DRV_DISP_*)"

   #error "Driver needs to be specified in GUIslice_config_*.h (DRV_DISP_*)"

X3msnake, 2020-08-29:
[image]

There is no LCDGFX driver

X3msnake, 2020-08-29:
Or any handling of that Library in the file
[image]

ImpulseAdventure, 2020-08-29:
@X3msnake — the LCDGFX handler should already exist in the library (it is defined by DRV_DISP_LCDGFX in GUIslice_drv_adagfx.cpp).

That error message is a bit surprising to see as it suggests that you might not have been using the example config file.

Can you confirm that you did each of the steps I listed in https://github.com/ImpulseAdventure/GUIslice/issues/248#issuecomment-681628862 ?

Thanks!

X3msnake, 2020-08-29:
> Can you confirm that you did each of the steps I listed in #248 (comment)

same error with the example file

X3msnake, 2020-08-29:
Arduino\libraries\GUIslice-master\src\GUIslice.c:59:48: fatal error: avr/pgmspace.h: No such file or directory

  -> candidates: []compilation terminated.

Following exactly the instructions with a clean library download i get this

X3msnake, 2020-08-29:
hacking the library by replacing avr/

[image]

I get this
```
In file included from C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_th.h:11:0,

                 from C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_th.cpp:8:

C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_th.cpp: In member function 'THPoint TouchHandler::scale(THPoint)':

C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:71:39: warning: comparison is always false due to limited range of data type [-Wtype-limits]

 #define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))

                                       ^

C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_th.cpp:65:14: note: in expansion of macro 'constrain'

     pOut.x = constrain(pOut.x,0,disp_xSize-1);

              ^

C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:71:39: warning: comparison is always false due to limited range of data type [-Wtype-limits]

 #define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))

                                       ^

C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_th.cpp:66:14: note: in expansion of macro 'constrain'

     pOut.y = constrain(pOut.y,0,disp_ySize-1);

              ^

C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:71:39: warning: comparison is always false due to limited range of data type [-Wtype-limits]

 #define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))

                                       ^

C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_th.cpp:67:14: note: in expansion of macro 'constrain'

     pOut.z = constrain(pOut.z,0,4095);

              ^

C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_drv_adagfx.cpp: In function 'bool gslc_DrvGetTxtSize(gslc_tsGui*, gslc_tsFont*, const char*, gslc_teTxtFlags, int16_t*, int16_t*, uint16_t*, uint16_t*)':

C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_drv_adagfx.cpp:952:57: error: no matching function for call to 'NanoFont::getTextSize(const char*&, uint16_t*&)'

   *pnTxtSzW = m_disp.getFont().getTextSize(pStr,pnTxtSzH);

                                                         ^

In file included from C:\Users\USERNAME\Documents\Arduino\libraries\lcdgfx\src/canvas/canvas.h:34:0,

                 from C:\Users\USERNAME\Documents\Arduino\libraries\lcdgfx\src/lcdgfx.h:34,

                 from C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_drv_adagfx.cpp:128:

C:\Users\USERNAME\Documents\Arduino\libraries\lcdgfx\src/canvas/font.h:122:15: note: candidate: lcduint_t NanoFont::getTextSize(const char*, lcduint_t*)

     lcduint_t getTextSize(const char *text, lcduint_t *height = nullptr);

               ^

C:\Users\USERNAME\Documents\Arduino\libraries\lcdgfx\src/canvas/font.h:122:15: note:   no known conversion for argument 2 from 'uint16_t* {aka short unsigned int*}' to 'lcduint_t* {aka unsigned int*}'

[... 31 lines cut ...]

                           ^

Foram encontradas múltiplas bibliotecas para «SPI.h»
Utilizado: C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Foram encontradas múltiplas bibliotecas para «GUIslice.h»
Utilizado: C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master
Foram encontradas múltiplas bibliotecas para «Adafruit_GFX.h»
Utilizado: C:\Users\USERNAME\Documents\Arduino\libraries\Adafruit_GFX_Library
Foram encontradas múltiplas bibliotecas para «lcdgfx.h»
Utilizado: C:\Users\USERNAME\Documents\Arduino\libraries\lcdgfx
Foram encontradas múltiplas bibliotecas para «Adafruit_I2CDevice.h»
Utilizado: C:\Users\USERNAME\Documents\Arduino\libraries\Adafruit_BusIO
Foram encontradas múltiplas bibliotecas para «Wire.h»
Utilizado: C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire
A utilizar a biblioteca GUIslice-master com a versão 0.15.0 na pasta: C:\Users\USERNAME\Documents\Arduino\libraries\GUIslice-master 
A utilizar a biblioteca Adafruit_GFX_Library com a versão 1.10.0 na pasta: C:\Users\USERNAME\Documents\Arduino\libraries\Adafruit_GFX_Library 
A utilizar a biblioteca lcdgfx com a versão 1.0.5 na pasta: C:\Users\USERNAME\Documents\Arduino\libraries\lcdgfx 
A utilizar a biblioteca Adafruit_BusIO com a versão 1.4.1 na pasta: C:\Users\USERNAME\Documents\Arduino\libraries\Adafruit_BusIO 
A utilizar a biblioteca Wire com a versão 1.0.1 na pasta: C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire 
A utilizar a biblioteca SPI com a versão 1.0 na pasta: C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 
exit status 1
Erro ao compilar para a placa LOLIN D32.

```

ImpulseAdventure, 2020-08-29:
Thanks @X3msnake — I think I have enough info now to make some updates to the library to support your I2C display fully with LCDGFX I2C, and address a few of the things you have encountered. Will take me a few days to get back to my development environment to test and will let you know when I have something ready for you to try out.

Thx

X3msnake, 2020-08-30:
awsome. thanks a bunch

ImpulseAdventure, 2020-09-04:
Hi @X3msnake -- I have now completed a number of fixes to address problems you reported here.
Could you try downloading and [installing the latest GUIslice from the GitHub repository](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source) and let me know if the compilation works better for you?

thanks

X3msnake, 2020-09-04:
Will do this weekend. Thanks :)

à(s) 08:21:

jesse-schein, 2020-09-05:
Ironically I was actually working on this tonight with an Lolin/Wemose ESP32 with built in OLED.

What I did was use ard-lcdgfx-notouch and made a copy for i2c.

Using the following:

#define DRV_DISP_LCDGFX           // lexus2k/lcdgfx
#define DRV_DISP_LCDGFX_SSD1306_128x64_I2C      // lexus2k/lcdgfx display: SSD1306 128x64 (I2C)

Then inside of my main.ino I just use Wire.begin(GPIO_NUM_5, GPIO_NUM_4); where my pins are the i2c pins.

So really their is already a working example of I2C SSD1306, its just inside of all of the configs and examples. This GUI slice library is really amazing, cant wait for my m5stack core2 to arrive so I can add touch support to it.

X3msnake, 2020-09-08:
What is the latest? Master branch?
Dowloading master and following the instructions you mentioned before gives me these errors on the attached log

[masterlog_200908.txt](https://github.com/ImpulseAdventure/GUIslice/files/5185409/masterlog_200908.txt)

X3msnake, 2020-09-08:
if however i comment out the ILI9341 that is set by default on ard-lcdgfx-notouch.h it compiles and uploads but there is no image still. even with the i2c hack on the esp pins file

X3msnake, 2020-09-08:
Was editing the wrong file.. duh

So i managed to build and it shows something on screen for the example 1 but this is not right i think

[image]

X3msnake, 2020-09-08:
There were some warning errors during compile, here is the compile full log:
[masterlog_ext04_ard_ctrls_200908.txt](https://github.com/ImpulseAdventure/GUIslice/files/5185596/masterlog_ext04_ard_ctrls_200908.txt)

X3msnake, 2020-09-10:
Any ideas why the partial pixel scramble?

ImpulseAdventure, 2020-09-12:
Hi @X3msnake -- thank you for posting some screenshots as those are helpful.

I assume your display resolution is 128x64. ex04_ard_ctrls was designed for much larger displays, so it is possible that some of the visual issues are due to wrapping in the display driver. ex40_ard_ctrls_small would be closer to what we want, but even that example targets 128x128.

LCDGFX does not currently support rounded rectangles, so that is why your example is not drawing correctly.
- Can you try modifying your "hello world" example but disabling the "corners rounded" option in the Builder?

Lastly, it looks like the monochrome SSD1306 driver might need a fix to change the white/black default. I'll look into that next.

thanks!

X3msnake, 2020-09-17:
Awsome.

Will try this tonight and let you know. thank you for the great help Master.

ImpulseAdventure, 2020-10-20:
Closing as assumed to be fixed. But if further problems remain, please feel free to open a new issue.

closed 2020-10-20 by ImpulseAdventure

X3msnake, 2020-10-20:
yes it is thanks