ImpulseAdventure/GUIslice #252: gslc_ElemSetTxtStr() not working

pablomaker113, 2020-08-29:
### Describe the bug
After updating to the new 0.15 version when I call gslc_ElemSetTxtStr() doesn't do anything. Am I missing something?

### Device hardware
- *MCU*:  ESP32.
- *GUIslice config selected*:  `esp-tftespi-default-xpt2046`

ImpulseAdventure, 2020-08-29:
Hi Pablo — can you confirm a couple things for me:

- Were you using the Builder? If so, did the text field have the “External Storage Size” greater than zero? (If the storage equals 0, then it will be read-only and the SetTxtStr will be ignored)
- Is this text field in FLASH? (Ie. Using ElemCreateTxt_P()?
- Most helpful would be if you can attach the sketch (& Builder project file if applicable)

Thank you

pablomaker113, 2020-08-29:
Hi, setting the “External Storage Size”  to my string size solved the issue.
Thank you for the support!!

closed 2020-08-29 by pablomaker113